MichaIng/DietPi #5845: Issues updating to 8.10 on Qauartz64 board that was running fine

Wilford888, 2022-10-29:
#### Details:
- Date           | Sat Oct 29 19:56:37 BST 2022
- DietPi version | v8.9.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Quartz64 (aarch64) (ID=49)
- Kernel version | `Linux DietPiTR 5.19.0-rc1 #1 SMP PREEMPT Fri Jul 15 09:39:18 UTC 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `getent hosts dns9.quad9.net`
- Exit code      | 2
- Software title | DietPi-Update
#### Steps to reproduce:

1. I've tried twice to update to 8.10.  Once from a running unit, and when that failed, I reflashed the eMMC with the latest image and rebooted.  On reboot, the unit tried to update automatically and again hung.  The update seems to finish but I cannot log in to the unit after the update.  This has happened twice in a row.
2. ...
#### Expected behaviour:

- Well, normally the updates work fine.
#### Actual behaviour:

- Unit seems to update, restarts, and then I can no longer access the board.
#### Extra details:

- I tried a complete reflash of the eMMC, after the auto-update ran, same result.  Updates seems to happen just fine.  Log out, cannot log back in.
#### Additional logs:

Wilford888, 2022-10-29:
One other strange thing:  this is a Rockport64 board.  And I've checked twice to make sure I downloaded and flashed the Rockport64 image to the eMMC.  But the log file reports this as a Quartz64; it is not.

Wilford888, 2022-10-29:
Tried the following:  I just received a QuartzA board, so I downloaded the image for that, flashed the eMMC. I was able to log in to that.  And that image is 8.10.  I think you have a problem with your current Rockport64 image on your site.

Joulinar, 2022-10-29:
I guess issue is more with Armbian kernel / U-boot update done in parallel to initial setup / update

canabino, 2022-10-29:
Date | Sat Oct 29 19:56:37 BST 2022
DietPi version | v8.9.2 (MichaIng/master)
Image creator | DietPi Core Team
Pre-image | Armbian
Hardware | RockPi 4b
Distro | bullseye (ID=6)
Software title | DietPi-Update

I am having the same issue but with my RockPi4.
I have recently successfully updated to Bullseye and had no problems.
Today I ran dietpi-update because it was recomending it to update to 8.10.2
After the update it wanted to restart but I couldn’t connect on SSH. I saw that the SBC wasn’t getting an IP address on my router. I then tried to view it from the HDMI connected to my TV but the TV got no signal from the HDMI port that my RockPi4 was connected to. 
I then installed a fresh image on the eMMC. It then boots up and I can access both SSH and on the HDMI. It then goes through all the steps and updates of a fresh install and asks to reboot. Then BAM, can’t get in from HDMI or SSH because no IP address gets assigned. It is connected via LAN cable.

Wilford888, 2022-10-29:
Joulinar, what do you mean?  Is there a solution or something else I might try?  Please note, I am not having this issue with a QuartzA board.  Only with the Rockport64.

Joulinar, 2022-10-29:
I don't have a solution yet. We would need to wait on the developer. But I see an Armbian kernel update on your logs that might have an influence. 

```
Get:2 http://armbian.tnahosting.net/apt bullseye/main arm64 linux-dtb-current-rockchip64 arm64 22.08.8 [380 kB]
Get:1 http://armbian.chi.auroradev.org/apt bullseye/main arm64 armbian-firmware all 22.08.6 [9409 kB]
Get:3 http://armbian.tnahosting.net/apt bullseye/main arm64 linux-image-current-rockchip64 arm64 22.08.8 [53.2 MB]
Get:4 http://armbian.tnahosting.net/apt bullseye/main arm64 linux-u-boot-rockpro64-current arm64 22.08.6 [436 kB]
```

Wilford888, 2022-10-29:
Having another issue with the Quartz64 board.  I can get past the boot but when I go to install Pi-hole, I get stuck in an endless loop because I already set the static IP address.  I keep getting asked to set a static IP.  If I hit cancel, installation of Pi-hole aborts.  If I hit okay, then I circle through do I want wifi removed, IP address has changed, please reboot.  Over and over, Pi-hole never gets installed.  I've been running Pi-hole for some months.  I really wish I could just downgrade to the prior release.

Wilford888, 2022-10-29:
I don't know if you have a QuartzA board handy.  If you do, install 8.10 and then try installing Pi-hole.  If you succeed, would really love to know what you did differently.

Wilford888, 2022-10-29:
To replicate my issue, set a static IP prior to trying to install Pi-Hole.

Wilford888, 2022-10-29:
The current Rockport64 image on your site is 8.9.2. Not sure if that is intentional or not.

Joulinar, 2022-10-29:
We don't update all the images on every release. Therefore it's fine to start on a older version as it will be updated on first boot automatically.

Pls don't mix the PiHole issue in. This has nothing to do with Tue original issue reported. I guess you are already inside the PiHole install script. This is an external script managed by PiHole guy's and we just load it. Inside the install process you need to confirm you have static IP. This is how PiHole guy's designed the process now. Or are you talking about another menu?

Wilford888, 2022-10-29:
Sorry for bringing up a different issue.  I will leave this with the original issue and check back from time to time.  Thank you.

Joulinar, 2022-10-29:
Pls open a new issue ff you have challenges on PiHole install process.

MichaIng, 2022-10-30:
You mean you use a "ROCKPro64", right? "Rockport64" is not an existing SBC 😉.

Is there an SD card attached to this board? Your bug report shows Quartz64 including the Quartz64 kernel, but the update log shows ROCKPro64 kernel/firmware upgraded. The only reason for such I can imagine is when bootloader/boot partition and root partition do not match, e.g. when a Quartz64 image is booted from SD card but then loading a ROCKPro64 rootfs from eMMC, or the other way round. Not sure how this is possible, though, without manually manipulating the kernel command-line parameters, since the rootfs should be always defined via UUID on our images 🤔.

Joulinar, 2022-10-30:
@MichaIng we have a similar report on our forum. https://dietpi.com/forum/t/cant-access-dietpi-after-latest-update/14781

It seems to happen after latest update. Maybe the kernel update has an influence.

MichaIng, 2022-10-30:
The issue with latest rockchip64 kernel has been recognised, taken down from the repo and fix is on the way: https://forum.armbian.com/topic/24196-rock-pi-4b-boot-failed-after-update/

MichaIng, 2022-11-01:
Issue shouldn't happen anymore, kernel v5.15.75 was the faulty one, now v5.15.76 (or later) is shipped which didn't causs the issue anymore.

Still doesn't explain the Quartz64 incidence, but that is unrelated to the kernel bug. So you can now reflash and won't be hit by the bug anymore.

Joulinar, 2022-11-01:
I guess we can close this issue, right? Because Armbian guys provided a new kernel version.

closed 2022-11-02 by MichaIng