MichaIng/DietPi #5848: Quartz64B Bullseye reboot loop (kernel starts)

robodrome, 2022-10-30:
### Quartz64B Bullseye reboot loop (kernel starts)

#### Required Information
- Image | [DietPi_Quartz64B-ARMv8-Bullseye](https://dietpi.com/downloads/images/DietPi_Quartz64B-ARMv8-Bullseye.7z)
- SBC model | [Pine Quartz64 model B v1.3](https://wiki.pine64.org/wiki/Quartz64)
- Power supply used | 5V 3A RAVpower
- SD card(s) used | SanDisk Extreme 32GB U3 (SDSQXAF-032G-GN6MN, A1/UHS-I/V30/U3/Class10), SanDisk Ultra 16GB (SDSQUAR-016G-GN6MA, A1/UHS-I/U1/Class10) and a brandless 8GB Class10

#### Additional Information (if applicable)
- Write software: balenaEtcher and/or Ubuntu image writer

#### Steps to reproduce

1. Download latest image from [here](https://dietpi.com/downloads/images/DietPi_Quartz64B-ARMv8-Bullseye.7z).
2. Write it to SDcard with [balenaEtcher](https://dietpi.com/docs/install/) or Ubuntu image writer (tried both).
3. Connect debug serial TTL-232R-3V3 (or simular) interface to Quartz64 connector (this works fine)
4. Start picocom according to [online manual](https://wiki.pine64.org/wiki/Quartz64#Connect_Debug_UART) on PC/laptop

NOTE: no eMMC or M.2 installed

#### Expected behaviour

- Linux running on Quartz64-b

#### Actual behaviour

- The system reboots over and over just after `systemd[1]: Set hostname to <DietPi>.`

#### Extra details

- Debug output:
```text
DDR Version V1.13 20220218
ln
ddrconfig:7
LPDDR4X, 324MHz
BW=32 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
tdqss: cs0 dqs0: 48ps, dqs1: -96ps, dqs2: -48ps, dqs3: -120ps, 
tdqss: cs1 dqs0: -24ps, dqs1: -48ps, dqs2: -48ps, dqs3: -144ps, 

change to: 324MHz
PHY drv:clk:36,ca:36,DQ:29,odt:0
vrefinner:24%, vrefout:41%
dram drv:40,odt:0
clk skew:0x61

change to: 528MHz
PHY drv:clk:36,ca:36,DQ:29,odt:0
vrefinner:24%, vrefout:41%
dram drv:40,odt:0
clk skew:0x58

change to: 780MHz
PHY drv:clk:36,ca:36,DQ:29,odt:0
vrefinner:24%, vrefout:41%
dram drv:40,odt:0
clk skew:0x58

change to: 1056MHz(final freq)
PHY drv:clk:36,ca:36,DQ:29,odt:60
vrefinner:16%, vrefout:22%
dram drv:40,odt:80
vref_ca:00000071
clk skew:0x2f
cs 0:
the read training result:
DQS0:0x3d, DQS1:0x40, DQS2:0x41, DQS3:0x39, 
min  : 0xb  0xb  0xd  0xa  0x1  0x2  0x6  0x2 , 0x9  0xa  0x4  0x2  0xc  0xb  0xe  0xa ,
      0x10 0x10  0xd  0xc  0x6  0x2  0x4  0x5 , 0xa  0x5  0x5  0x1  0xb  0xc  0x9  0xb ,
mid  :0x29 0x2a 0x2c 0x29 0x20 0x21 0x24 0x21 ,0x28 0x29 0x24 0x21 0x2b 0x2a 0x2c 0x29 ,
[... 840 lines cut ...]
[    1.898020] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    1.898886] ALSA device list:
[    1.898898]   No soundcards found.
[    1.900423] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    1.900435] platform regulatory.0: Falling back to sysfs fallback for: regulatory.db
[    1.918714] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Quota mode: none.
[    1.918881] VFS: Mounted root (ext4 filesystem) readonly on device 179:1.
[    1.930333] devtmpfs: mounted
[    1.936492] Freeing unused kernel memory: 6080K
[    1.942582] Run /sbin/init as init process
[    1.942588]   with arguments:
[    1.942590]     /sbin/init
[    1.942592]   with environment:
[    1.942594]     HOME=/
[    1.942595]     TERM=linux
[    2.790170] systemd[1]: System time before build time, advancing clock.
[    2.999495] NET: Registered PF_INET6 protocol family
[    3.001362] Segment Routing with IPv6
[    3.001406] In-situ OAM (IOAM) with IPv6
[    3.050224] systemd[1]: systemd 247.3-7+deb11u1 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +ZSTD +SECCOMP +BLKID +ELFUTILS +KMOD +IDN2 -IDN +PCRE2 default-hierarchy=unified)
[    3.051084] systemd[1]: Detected architecture arm64.
[    3.079671] systemd[1]: Set hostname to <DietPi>.
DDR Version V1.13 20220218
(...)
```

MichaIng, 2022-10-30:
Many thanks for reporting.

Could you test it with another Linux OS? I had boot issues with Quartz64 due to insufficient PSU and bad cable. At this stage without any further error message, I cannot imagine anything else being the issue 🤔.

robodrome, 2022-10-30:
Ok. Thanks. PSU is 5V 3A net adapter. I soldered a 3.5mm DC plug to it. I could try another one, but that doesn't make sense to me. Unless it is faulty in some weird way. Delivering 5V but enough amps. I also tried [Manjaro](https://github.com/manjaro-arm/quartz64-b-images/releases/download/22.10/Manjaro-ARM-kde-plasma-quartz64-b-22.10.img.xz). That also gets stuck in a boot loop.

```text
DDR Version V1.13 20220218
ln
ddrconfig:7
LPDDR4X, 324MHz
BW=32 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
tdqss: cs0 dqs0: 48ps, dqs1: -96ps, dqs2: -48ps, dqs3: -120ps, 
tdqss: cs1 dqs0: -24ps, dqs1: -72ps, dqs2: -48ps, dqs3: -144ps, 

change to: 324MHz
PHY drv:clk:36,ca:36,DQ:29,odt:0
vrefinner:24%, vrefout:41%
dram drv:40,odt:0
clk skew:0x61

change to: 528MHz
PHY drv:clk:36,ca:36,DQ:29,odt:0
vrefinner:24%, vrefout:41%
dram drv:40,odt:0
clk skew:0x58

change to: 780MHz
PHY drv:clk:36,ca:36,DQ:29,odt:0
vrefinner:24%, vrefout:41%
dram drv:40,odt:0
clk skew:0x58

change to: 1056MHz(final freq)
PHY drv:clk:36,ca:36,DQ:29,odt:60
vrefinner:16%, vrefout:22%
dram drv:40,odt:80
vref_ca:00000071
clk skew:0x2f
cs 0:
the read training result:
DQS0:0x3d, DQS1:0x40, DQS2:0x41, DQS3:0x39, 
min  : 0xb  0xb  0xd  0xa  0x0  0x2  0x6  0x1 , 0x9  0xa  0x4  0x2  0xb  0xb  0xe  0xa ,
      0x10 0x10  0xc  0xc  0x6  0x1  0x3  0x5 , 0xa  0x5  0x5  0x0  0xb  0xb  0x9  0xb ,
mid  :0x29 0x2a 0x2c 0x29 0x20 0x21 0x24 0x20 ,0x28 0x29 0x24 0x21 0x2a 0x2a 0x2c 0x29 ,
      0x2e 0x2e 0x2a 0x28 0x24 0x20 0x21 0x23 ,0x28 0x23 0x23 0x1e 0x2a 0x29 0x27 0x29 ,
max  :0x47 0x49 0x4c 0x49 0x40 0x41 0x43 0x40 ,0x48 0x48 0x44 0x41 0x4a 0x4a 0x4b 0x49 ,
      0x4c 0x4c 0x48 0x45 0x42 0x3f 0x3f 0x41 ,0x46 0x42 0x42 0x3d 0x49 0x48 0x45 0x47 ,
range:0x3c 0x3e 0x3f 0x3f 0x40 0x3f 0x3d 0x3f ,0x3f 0x3e 0x40 0x3f 0x3f 0x3f 0x3d 0x3f ,
      0x3c 0x3c 0x3c 0x39 0x3c 0x3e 0x3c 0x3c ,0x3c 0x3d 0x3d 0x3d 0x3e 0x3d 0x3c 0x3c ,
the write training result:
DQS0:0x35, DQS1:0x23, DQS2:0x29, DQS3:0x1f, 
min  :0x63 0x66 0x67 0x64 0x5a 0x5b 0x5d 0x5e 0x5e ,0x50 0x4f 0x4a 0x48 0x51 0x50 0x52 0x50 0x4c ,
      0x56 0x54 0x51 0x50 0x49 0x47 0x4a 0x4e 0x4f ,0x4d 0x4b 0x49 0x46 0x4e 0x4e 0x4b 0x50 0x4a ,
mid  :0x80 0x83 0x84 0x80 0x75 0x77 0x7a 0x79 0x79 ,0x6b 0x6b 0x65 0x63 0x6d 0x6b 0x6c 0x6b 0x68 ,
      0x72 0x71 0x6d 0x6c 0x65 0x60 0x63 0x69 0x69 ,0x6a 0x66 0x64 0x61 0x6a 0x6b 0x66 0x6c 0x63 ,
max  :0x9d 0xa0 0xa2 0x9c 0x91 0x93 0x97 0x94 0x94 ,0x87 0x88 0x80 0x7e 0x8a 0x86 0x86 0x87 0x85 ,
      0x8f 0x8f 0x89 0x88 0x82 0x7a 0x7d 0x84 0x84 ,0x87 0x82 0x80 0x7d 0x86 0x88 0x82 0x88 0x7d ,
range:0x3a 0x3a 0x3b 0x38 0x37 0x38 0x3a 0x36 0x36 ,0x37 0x39 0x36 0x36 0x39 0x36 0x34 0x37 0x39 ,
      0x39 0x3b 0x38 0x38 0x39 0x33 0x33 0x36 0x35 ,0x3a 0x37 0x37 0x37 0x38 0x3a 0x37 0x38 0x33 ,
cs 1:
the read training result:
DQS0:0x3a, DQS1:0x3e, DQS2:0x43, DQS3:0x3c, 
min  : 0xb  0xc  0xe  0xa  0x1  0x3  0x6  0x1 , 0x7  0x7  0x3  0x1  0xa  0x8  0xd  0x8 ,
       0xf  0xf  0xb  0xb  0x5  0x1  0x3  0x5 , 0xd  0x7  0x7  0x1  0xe  0xd  0xb  0xd ,
mid  :0x28 0x29 0x2b 0x28 0x1f 0x20 0x23 0x1f ,0x26 0x26 0x20 0x20 0x28 0x27 0x2a 0x27 ,
      0x2e 0x2e 0x2a 0x29 0x24 0x21 0x22 0x24 ,0x2b 0x26 0x26 0x20 0x2d 0x2c 0x29 0x2b ,
max  :0x45 0x47 0x49 0x46 0x3d 0x3e 0x41 0x3d ,0x45 0x45 0x3e 0x3f 0x47 0x46 0x48 0x46 ,
      0x4d 0x4d 0x49 0x48 0x43 0x41 0x41 0x44 ,0x49 0x46 0x45 0x3f 0x4c 0x4c 0x48 0x4a ,
range:0x3a 0x3b 0x3b 0x3c 0x3c 0x3b 0x3b 0x3c ,0x3e 0x3e 0x3b 0x3e 0x3d 0x3e 0x3b 0x3e ,
      0x3e 0x3e 0x3e 0x3d 0x3e 0x40 0x3e 0x3f ,0x3c 0x3f 0x3e 0x3e 0x3e 0x3f 0x3d 0x3d ,
the write training result:
DQS0:0x35, DQS1:0x23, DQS2:0x29, DQS3:0x1f, 
min  :0x56 0x59 0x5a 0x57 0x4d 0x50 0x52 0x52 0x50 ,0x4d 0x4e 0x48 0x46 0x50 0x4f 0x50 0x50 0x4b ,
[... 62 lines cut ...]
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
1:    Manjaro ARM
Retrieving file: /initramfs-linux.img
Retrieving file: /Image
append: initrd=/initramfs-linux.img console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=25837582-6919-4aaa-84db-270f9ae2febd rw rootwait audit=0 splash plymouth.ignore-serial-consoles
Retrieving file: /dtbs/rockchip/rk3566-quartz64-b.dtb
Moving Image from 0x2080000 to 0x2200000, end=3a90000
## Flattened Device Tree blob at 0a100000
   Booting using the fdt blob at 0xa100000
   Loading Ramdisk to ed6bc000, end edeecca5 ... OK
   Loading Device Tree to 00000000ed69e000, end 00000000ed6bb5ff ... OK

Starting kernel ...

:: running early hook [udev]
Starting version 251.6-2-arch
:: running early hook [plymouth]
(...)
```

Joulinar, 2022-10-30:
hmm might it be a hardware issue if it is failing on a complete different destro as well?

robodrome, 2022-10-30:
That's possible of course. I will contact Pine about this then. See what they have to say. If anything. The warranty on the Quartz64b is very limited if I recall correctly. Thanks for your help!

closed 2022-10-30 by robodrome

robodrome, 2022-11-01:
It seems this issue was related to power supply. Some power supplies need minimal load to operate correctly. It seems this was the case.