MichaIng/DietPi #5849: [bug] changing password to very long password (128 characters tested) locks user out of system if using dietpi-config

t3dium, 2022-10-30:
When using dietpi config, entering a very long password locks the user out of the system. I've reproduced this bug on two seperate systems running dietpi and when changing the password via dietpi config, i get locked out of the system and ssh with that exact password fails.

Trying to change the password via `passwd` to the long one, succeeds however.

When using shorter password, this issue isn't present. The long password was also purely numbers and letters, so the issue isn't related to symbols.

**SSH server tested:** default dropbear
**Dietpi version:** latest

Also pasted the password as opposed to typing obviously, so it couldn't be a spelling mistake.

MichaIng, 2022-10-30:
I checked whether whiptail password boxes themselves have a limitation, but 135 characters worked well. Can you test whether it works this way, i.e. when you enter your long password into the input boxes, whether the `echo` prints it to console correctly?
```sh
G_WHIP_PASSWORD
echo "$result"
unset -v result
```

t3dium, 2022-10-30:
> ```shell
> G_WHIP_PASSWORD
> echo "$result"
> unset -v result
> ```

How can i run this? i put it in a script and tried running with sh and bash, but it says command not found on the G_WHIP_PASSWORD line:

[image]

MichaIng, 2022-10-30:
Run it from console directly instead of within a script. It is a shell function, hence not available in scripts if no `dietpi-globals` are loaded explicitly.

t3dium, 2022-10-30:
tried a dummy 128 character password, and using that method it returns the password in the echo
[image]

MichaIng, 2022-10-30:
Then the only other possibility is that `chpasswd` handles it differently than `passwd`:
```sh
chpasswd <<< 'dietpi:yourlongpassword'
```
changes it for the `dietpi` user.

MichaIng, 2022-11-09:
Tried (150x`1`):
```sh
chpasswd <<< 'dietpi:111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111'
login
# dietpi
# 111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111
```
works well. Setting the same password via `dietpi-config` works fine as well, with both `root` and `dietpi` logins via local console/login command. However, with Dropbear login, indeed the maximum number of characters is 100: With 101x`1` it fails.

Switching to OpenSSH, login works, switching back to Dropbear, login fails. So clearly a Dropbear limitation.

MichaIng, 2022-11-09:
And here the related commit: https://github.com/mkj/dropbear/commit/8b4f60a
Ah, Dropbear even seems to log it, I just didn't have a look into the server logs, to be true 😅.

t3dium, 2022-11-11:
Perfect thanks

closed 2022-11-11 by t3dium