MichaIng/DietPi #5850: [bug] infinite "command not found errors" when pressing ok to changing the password on first install

t3dium, 2022-10-30:
[image]

I can't reproduce this bug after i re-flashed the usb with dietpi and reinstalled, it only happened once randomly.

MichaIng, 2022-10-30:
It looks like the filesystem is corrupted, missing basic system commands. Run `fsck` on it from a different Linux system and you'll find a lot of filesystem errors.

Is it a 2.5" drive and do you power it via dedicated PSU or through USB only? The letter won't work reliably on RPi USB ports and causes exactly such kind of errors. That the issue is triggered when attaching another USB device is another indicator to USB power issues.

t3dium, 2022-10-30:
> Is it a 2.5" drive and do you power it via dedicated PSU or through USB only? The letter won't work reliably on RPi USB ports and causes exactly such kind of errors. That the issue is triggered when attaching another USB device is another indicator to USB power issues (the max USB power is shared across all ports).

Just two flash drives, using the raspberry pi's power.

MichaIng, 2022-10-30:
Please check:
```sh
dmesg -l emerg,alert,crit,err
```
for kernel errors.

t3dium, 2022-10-30:
> Please check:
> 
> ```shell
> dmesg -l emerg,alert,crit,err
> ```
> 
> for kernel errors.

I'll check this soon when i boot into the usb again, but i dont think it'd work since no commands were previously working. They were all returning input/output error.

MichaIng, 2022-10-30:
Then you need to reflash the image, boot without the second USB device attached. Then attach/mount the second USB device, check for kernel errors, in case run some updates/do some I/O and check for kernel errors again. This would be to verify that it's a powering issue, or in case whether it's another specific issues with those drives or their combination somehow.

MichaIng, 2022-12-30:
Marking this as closed for now. Feel free to reopen if the issue persists and you find time to further debug.

closed 2022-12-30 by MichaIng