MichaIng/DietPi #5851: dietpi-* curses apps don't recognize TERM=xterm-kitty

JoshMock, 2022-10-31:
### Creating a bug report/issue

#### Required Information
- DietPi version |
  ```
  G_DIETPI_VERSION_CORE=8
  G_DIETPI_VERSION_SUB=10
  G_DIETPI_VERSION_RC=2
  G_GITBRANCH='master'
  G_GITOWNER='MichaIng'
  ```
- Distro version | `bullseye 0`
- Kernel version | `Linux nas 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | unsure
- SD card used | `Sandisk MicroSDHC V30 32GB Extreme Pro`

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? Yes.

#### Steps to reproduce
1. `ssh` into a machine running Dietpi using [Kitty](https://sw.kovidgoyal.net/kitty/) as your terminal emulator, by running `kitty +kitten ssh username@hostname`
2. `echo $TERM` returns `xterm-kitty`
3. all necessary info for `xterm-kitty` appears to be written correctly to `~/.terminfo` directory
4. load any `dietpi-*` TUI and see that it won't start, with the following error:
   ```
   tput: unknown terminal "xterm-kitty"
   /boot/dietpi/func/dietpi-globals: line 271: ( 42 + 5 ) /  : syntax error: operand expected (error token is "/  ")
   ```

#### Expected behaviour

- `dietpi-*` apps should behave correctly and/or fail more gracefully when using Kitty

#### Actual behaviour

- I can't run any `dietpi-*` app, unless I remember to set `TERM=xterm-color` before ssh-ing into a Dietpi box.

#### Extra details

- Kitty version: `0.26.4`
- Desktop operating system running Kitty: Arch Linux (`6.0.5-arch1-1`)
- Window manager: SwayWM 1.7

Joulinar, 2022-10-31:
Should be similar to this one https://github.com/MichaIng/DietPi/issues/5800

MichaIng, 2022-10-31:
Ignore the first part of the linked issue, I mixed up "kitty" (terminal emulator) with "KiTTY" (SSH client). However, the solution to install the terminfo system-wide is correct:
```sh
mkdir -p /etc/terminfo/x
mv ~/.terminfo/x/xterm-kitty /etc/terminfo/x/xterm-kitty
rmdir -p --ignore-fail-on-non-empty ~/.terminfo/x
```

MichaIng, 2022-10-31:
Btw, an alternative on Debian/Raspbian/DietPi/Ubuntu:
```sh
sudo apt install xterm-kitty
```
It's an older version but should work fine, and is installed system-wide, of course.

JoshMock, 2022-10-31:
I appreciate that. I hadn't associated it with a `sudo` thing, but it does occur to me that dietpi-* scripts are possibly the only TUIs I'm running via `sudo`.

closed 2022-10-31 by JoshMock