MichaIng/DietPi #5853: dietpi-backup to NFS terminated

jchhenderson, 2022-11-01:
#### Details:
- Date           | Tue Nov  1 12:26:45 GMT 2022
- DietPi version | v8.9.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `mkdir -p /mnt/nfs_client/dietpi-backup`
- Exit code      | 1
- Software title | DietPi-Backup
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:
I was backup up to a NFS volume
- ...
#### Actual behaviour:
`Unable to continue, DietPi-Backup will now terminate.`
- ...
#### Extra details:

- ...
#### Additional logs:
```
mkdir: cannot create directory ‘/mnt/nfs_client’: No such device
```

Joulinar, 2022-11-01:
Are you sure NFS is mounted correctly to `mnt/nfs_client`? Can you share

`df -h`

jchhenderson, 2022-11-01:
Hi,

I am not sure that will help as my Pi froze and I had to pull the power.   However it had been backing up for 10 mins before it terminated.

It also showed that the backup had been successful.  Very strange.

I have since also updated to 8.10.2

Here are the results anyway...

```
root@DietPi:~# df -h
Filesystem             Size  Used Avail Use% Mounted on
/dev/root              220G   41G  171G  20% /
devtmpfs               916M     0  916M   0% /dev
tmpfs                  950M     0  950M   0% /dev/shm
tmpfs                  380M  5.7M  375M   2% /run
tmpfs                  5.0M     0  5.0M   0% /run/lock
tmpfs                  1.0G     0  1.0G   0% /tmp
tmpfs                   50M     0   50M   0% /var/log
/dev/sda1              127M   35M   92M  28% /boot
overlay                220G   41G  171G  20% /mnt/dietpi_userdata/docker-data/overlay2/9a7393c6e9fe47237ff3c923d2b56c04271cc52bce4ad147609fab6e362f428f/merged
overlay                220G   41G  171G  20% /mnt/dietpi_userdata/docker-data/overlay2/61dc79a007bc6eb0d82a5c5d7fea8937e91ea7c7781b0a1399451aa26e8cbb73/merged
overlay                220G   41G  171G  20% /mnt/dietpi_userdata/docker-data/overlay2/592641855bbfceb704a3c180bc86ee3ddc7890bee6cbb00f68fa07a898e079ac/merged
overlay                220G   41G  171G  20% /mnt/dietpi_userdata/docker-data/overlay2/996d4fd52ccdd6f39f4f37cec1ca4c9e5a34f4d0279410ab7cb102bbf66d6368/merged
overlay                220G   41G  171G  20% /mnt/dietpi_userdata/docker-data/overlay2/01177c2dce6a0b21bd957f393d02b0d27eb2d75ac1a0b54510638af8bddf99b2/merged
192.168.1.63:/backups   21T   16T  5.2T  75% /mnt/nfs_client
```

MichaIng, 2022-11-01:
Does it work when you manually create the dir?
```sh
mkdir -p /mnt/nfs_client/dietpi-backup
```

jchhenderson, 2022-11-01:
yes, that seems to work fine.

jchhenderson, 2022-11-01:
maybe I had a network blip?

Joulinar, 2022-11-01:
Probably. Can you run the backup again?

jchhenderson, 2022-11-01:
sure.  will do it later.

jchhenderson, 2022-11-22:
yep, got this...
```
root@DietPi:~# dietpi-backup

 DietPi-Backup
─────────────────────────────────────────────────────
 Mode: Backup

[ INFO ] DietPi-Backup | Rotating backup archive
[  OK  ] DietPi-Backup | mv /mnt/nfs_client/dietpi-backup/data_7 /mnt/nfs_client/dietpi-backup/data_8
[  OK  ] DietPi-Backup | mv /mnt/nfs_client/dietpi-backup/data_6 /mnt/nfs_client/dietpi-backup/data_7
[  OK  ] DietPi-Backup | mv /mnt/nfs_client/dietpi-backup/data_5 /mnt/nfs_client/dietpi-backup/data_6
[  OK  ] DietPi-Backup | mv /mnt/nfs_client/dietpi-backup/data_4 /mnt/nfs_client/dietpi-backup/data_5
[  OK  ] DietPi-Backup | mv /mnt/nfs_client/dietpi-backup/data_3 /mnt/nfs_client/dietpi-backup/data_4
[  OK  ] DietPi-Backup | mv /mnt/nfs_client/dietpi-backup/data_2 /mnt/nfs_client/dietpi-backup/data_3
[  OK  ] DietPi-Backup | mv /mnt/nfs_client/dietpi-backup/data /mnt/nfs_client/dietpi-backup/data_2
[  OK  ] DietPi-Backup | mkdir -p /mnt/nfs_client/dietpi-backup/data
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : docker
[  OK  ] DietPi-Services | stop : grafana-server
[  OK  ] DietPi-Services | stop : mosquitto
[  OK  ] DietPi-Services | stop : tautulli
[ INFO ] DietPi-Backup | Checking for required APT packages: rsync
[ INFO ] DietPi-Backup | Checking for sufficient disk space via rsync dry-run, please wait...
[  OK  ] DietPi-Backup | Free space check: path=/mnt/nfs_client/dietpi-backup/data | available=4927332 MiB | required=1394966 MiB
[ INFO ] DietPi-Backup | Backup to /mnt/nfs_client/dietpi-backup in progress, please wait...
sending incremental file list
    634,027,197  96%    6.98MB/s    0:01:26 (xfr#13402, ir-chk=1430/19918)client_loop: send disconnect: Broken pipe
```

MichaIng, 2023-02-12:
Is this issue still happening with latest kernel and package versions?

JCHH2, 2023-02-12:
I'll do a back-up later and let you know - thanks for the reminder

MichaIng, 2023-02-14:
Okay that is good. Next time it will be faster since it will then copy only the diff to the backup no. 10 instead of creating a whole new copy.

I'll close this issue then. Feel free to reopen when required.

closed 2023-02-14 by MichaIng

JCHH2, 2023-02-14:
I had to kill the backup when I realised that it had already taken 7 hours to get to 54Gb and it needed to get to 85GB (I needed some services back and didn't realise it would take this long).

I assumer that backup is now corrupted and I will have to make ~11 hours available to do it again?

MichaIng, 2023-02-14:
No such a sync can be resumed just fine. The only thing I'm not sure about is that, AFAIK, `dietpi-backup` rotates it next time despite the aborted backup, which will then be move to `/mnt/nfs_client/dietpi-backup/data_2` as if it was successful.

You could ignore it since `data_10` will be used as basis for the next attempt, so its a diff sync only then. Or you could could rotate everything one turn back and `data` to `data_10` to fool our script, so it really continues the sync you just aborted 😄.

JCHH2, 2023-02-14:
Thanks.

I'd like to continue the sync I just aborted.  Exactly how do I do that (how do I turn `data` back to `data_10`)?

MichaIng, 2023-02-14:
You see the directory renaming (`mv`) that has been done in your output above. That needs to be reverted:
```sh
mv /mnt/nfs_client/dietpi-backup/data /mnt/nfs_client/dietpi-backup/data_11
mv /mnt/nfs_client/dietpi-backup/data_2 /mnt/nfs_client/dietpi-backup/data
mv /mnt/nfs_client/dietpi-backup/data_3 /mnt/nfs_client/dietpi-backup/data_2
mv /mnt/nfs_client/dietpi-backup/data_4 /mnt/nfs_client/dietpi-backup/data_3
mv /mnt/nfs_client/dietpi-backup/data_5 /mnt/nfs_client/dietpi-backup/data_4
mv /mnt/nfs_client/dietpi-backup/data_6 /mnt/nfs_client/dietpi-backup/data_5
mv /mnt/nfs_client/dietpi-backup/data_7 /mnt/nfs_client/dietpi-backup/data_6
mv /mnt/nfs_client/dietpi-backup/data_8 /mnt/nfs_client/dietpi-backup/data_7
mv /mnt/nfs_client/dietpi-backup/data_9 /mnt/nfs_client/dietpi-backup/data_8
mv /mnt/nfs_client/dietpi-backup/data_10 /mnt/nfs_client/dietpi-backup/data_9
mv /mnt/nfs_client/dietpi-backup/data_11 /mnt/nfs_client/dietpi-backup/data_10
```