MichaIng/DietPi #5855: Keep seeing proc: Bad value for 'hidepid' in logs on latest install.

m0jek, 2022-11-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | bullseye
- Kernel version | Linux Janeway 5.4.218-odroidxu4 #22.08.6 SMP PREEMPT Tue Oct 18 06:39:15 UTC 2022 armv7l GNU/Linux
- SBC model | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)
- Power supply used | 5V 6A CWT (supplied by Odroid UK)
- SD card used | eMMC 32G

#### Additional Information (if applicable)
- Software title | N/A base OS
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | 87ce833d-6743-4abb-adaa-48914cb359b1

#### Steps to reproduce
- Just install the latest DietPi OS.
- kernel logs will show;
  Nov 02 07:30:12 Janeway kernel: proc: Bad value for 'hidepid'

#### Expected behaviour

- shouldn't see anything.

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2022-11-02:
Expected on Linux <5.8 with recent systemd, you can ignore this message: https://github.com/systemd/systemd/issues/16896

m0jek, 2022-11-02:
Thank you for the update.
I have read teh link and will close this issue.

closed 2022-11-02 by m0jek