MichaIng/DietPi #5856: RoonBridge install fails

net-david, 2022-11-02:
[FAILED] DietPi-Software | curl -sSfL https://download.roonlabs.com/builds/RoonBridge_linuxarmv8.tar.bz2 -o RoonBridge_linuxarmv8.tar.bz2
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
aplay: device_list:274: no soundcards found...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: 5711bab6-a68d-4a10-82a3-f67cf2fb35dc

I believe the issue is the download site for Roon should be download.roonlabs.net, not .com

MichaIng, 2022-11-02:
Many thanks for your report.

It seems downloads are now hosted on a different domain: https://download.roonlabs.net/builds/RoonBridge_linuxarmv8.tar.bz2

Please try:
```sh
set -i 's/download.roonlabs.com/download.roonlabs.net/' /boot/dietpi/dietpi-software
```
Then repeat the install.

MichaIng, 2022-11-02:
Solved for next release: https://github.com/MichaIng/DietPi/commit/9b2a21b

Joulinar, 2022-11-02:
I guess it should be `sed` and not `set` 🤣 

```
sed -i 's/download.roonlabs.com/download.roonlabs.net/' /boot/dietpi/dietpi-software
```

net-david, 2022-11-02:
That did it! Thanks!

Joulinar, 2022-11-02:
Ok, I will go to close the issue. Feel free to reopen if needed.

closed 2022-11-02 by Joulinar