MichaIng/DietPi #5857: Home Assistant installation is missing libcblas as used by numpy

m0jek, 2022-11-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | bullseye
- Kernel version | Linux Janeway 5.4.218-odroidxu4 #22.08.6 SMP PREEMPT Tue Oct 18 06:39:15 UTC 2022 armv7l GNU/Linux
- SBC model | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)
- Power supply used | 5V 6A
- SD card used | eMMC

#### Additional Information (if applicable)
- Software title | Homeassistant
- Was the software title installed freshly or updated/migrated? fresh install
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | 87ce833d-6743-4abb-adaa-48914cb359b1

#### Steps to reproduce
- Install using "dietpi-software" tool.
- follow guide on https://dietpi.com/docs/software/home_automation/

#### Expected behaviour

-  onboarding works with no "undefined error"
-  logs are pretty clear, only warnings
```
Nov 02 13:10:53 Janeway systemd[1]: Started Home Assistant (DietPi).
Nov 02 13:10:57 Janeway Home Assistant[2573]: 2022-11-02 13:10:57.099 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
```

#### Actual behaviour

- When onboarding web page gives a undefined error.
- one can still login, but then there are errors in logs
- extract of logs;
```
Nov 02 12:46:16 Janeway systemd[1]: Started Home Assistant (DietPi).
Nov 02 12:46:20 Janeway Home Assistant[2070]: 2022-11-02 12:46:20.589 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Nov 02 12:46:26 Janeway Home Assistant[2070]: 2022-11-02 12:46:26.361 ERROR (MainThread) [homeassistant.setup] Setup failed for cloud: Unable to import component:
Nov 02 12:46:26 Janeway Home Assistant[2070]: IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!
Nov 02 12:46:26 Janeway Home Assistant[2070]: Importing the numpy C-extensions failed. This error can happen for
Nov 02 12:46:26 Janeway Home Assistant[2070]: many reasons, often due to issues with your setup or how NumPy was
Nov 02 12:46:26 Janeway Home Assistant[2070]: installed.
Nov 02 12:46:26 Janeway Home Assistant[2070]: We have compiled some common reasons and troubleshooting tips at:
Nov 02 12:46:26 Janeway Home Assistant[2070]:     https://numpy.org/devdocs/user/troubleshooting-importerror.html
Nov 02 12:46:26 Janeway Home Assistant[2070]: Please note and check the following:
Nov 02 12:46:26 Janeway Home Assistant[2070]:   * The Python version is: Python3.9 from "/home/homeassistant/.pyenv/versions/3.9.15/bin/python3.9"
Nov 02 12:46:26 Janeway Home Assistant[2070]:   * The NumPy version is: "1.23.2"
Nov 02 12:46:26 Janeway Home Assistant[2070]: and make sure that they are the versions you expect.
Nov 02 12:46:26 Janeway Home Assistant[2070]: Please carefully study the documentation linked above for further help.
Nov 02 12:46:26 Janeway Home Assistant[2070]: Original error was: libcblas.so.3: cannot open shared object file: No such file or directory
Nov 02 12:46:34 Janeway Home Assistant[2070]: 2022-11-02 12:46:34.718 ERROR (MainThread) [homeassistant.setup] Setup failed for mobile_app: Unable to import component:
Nov 02 12:46:34 Janeway Home Assistant[2070]: IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!
Nov 02 12:46:34 Janeway Home Assistant[2070]: Importing the numpy C-extensions failed. This error can happen for
Nov 02 12:46:34 Janeway Home Assistant[2070]: many reasons, often due to issues with your setup or how NumPy was
Nov 02 12:46:34 Janeway Home Assistant[2070]: installed.
Nov 02 12:46:34 Janeway Home Assistant[2070]: We have compiled some common reasons and troubleshooting tips at:
Nov 02 12:46:34 Janeway Home Assistant[2070]:     https://numpy.org/devdocs/user/troubleshooting-importerror.html
```

#### Extra details

- the fix is to install the following as suggested in https://numpy.org/devdocs/user/troubleshooting-importerror.html
- `apt install libatlas-base-dev`
- Restart homeassistant; `systemctl restart home-assistant`
- and the logs will be clear and no more module errors. 
```
Nov 02 13:10:53 Janeway systemd[1]: Started Home Assistant (DietPi).
Nov 02 13:10:57 Janeway Home Assistant[2573]: 2022-11-02 13:10:57.099 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
```

Joulinar, 2022-11-02:
Already known, a fix will become available on next release https://github.com/MichaIng/DietPi/commit/23910ee

Joulinar, 2022-11-02:
I'm going to close the issue if there are no further question. Feel free to reopen if needed.

closed 2022-11-02 by Joulinar