MichaIng/DietPi #5858: Orange Pi 4 LTS create own repository failed

AkaMrBill, 2022-11-03:
### Creating a bug report/issue

#### Details:
- Date           | Thu Nov  3 21:38:10 UTC 2022
- DietPi version | v8.10.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=)
- Kernel version | `Linux orangepi4-lts 5.10.43 #3.0.6 SMP Mon Aug 1 11:52:03 CST 2022 aarch64 GNU/Linux`
- Distro         |  (ID=)
- Command        | `curl --key  --connect-timeout 8 --retry 1 --retry-delay 4 -sSvT .7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/bugreport/`
- Exit code      | 79
- Software title | DietPi-BugReport
#### Steps to reproduce:

1. Burn Debian image to TF card and then copy to eMMC.
2. Run the conversion script to create DietPi image from the Debian install#### Expected behaviour:

- DietPi installation should be created.
#### Actual behaviour:

- Errors are reported for dependancies that should be, but won't be installed.
#### Extra details:

- ...
#### Additional logs:
```
*   Trying 185.101.93.93:29248...
* Connected to ssh.dietpi.com (185.101.93.93) port 29248 (#0)
* Found host ssh.dietpi.com in /root/.ssh/known_hosts
* Set "ssh-ed25519" as SSH hostkey type
* Closing connection 0
curl: (79) Error in the SSH layer
```

Joulinar, 2022-11-03:
The error you shared is about not being able to upload a bug report. Is there another issue you like to share next to the bug report?

AkaMrBill, 2022-11-03:
Crap, oops.
Yeah, I was trying to create a diet pie image from an installed Davion
image on an Orange pi 4 LTS.
The script errored out saying that there were several dependencies, and
provided a huge list, that needed to be installed but would not be
installed.

Let me try running the process again and see if I can get the actual list.

Bill K.

AkaMrBill, 2022-11-03:
I just read what I replied, what a mess. Speach to text SUCKS!

I was trying to create a custom distribution for an orange4 LTS. I was
trying to modify the Debian distribution, available on their website, to
dietpi. I am trying one more thing before I send you the results. I will
get back to you.

BK

Joulinar, 2022-11-03:
Best is to answer directly on GitHub. Answering by mail has some limitations 😉

AkaMrBill, 2022-11-03:
Got it.

I am trying an additional step to those provided on how to "Create your Own Distribution".
The instructions show to do a "apt update", but do not include "apt upgrade". So I am trying that now. I will report back on whether that resolved the issue.

Joulinar, 2022-11-03:
Try to share the full log once issue occurs.

AkaMrBill, 2022-11-03:
OK, so that failed.
I wold love to share the full log, but will need guidence on where to find it.
BTW, I attempted to perform the conversion to Dietpi via an SSH terminal due to the warning about not trying it from a "desktop".

Joulinar, 2022-11-03:
Simply try to scroll up on your SSH terminal and copy/past the text

AkaMrBill, 2022-11-03:
Yep, tried that. Putty is not providing the copy/paste functionality I would expect. Let me give it another shot.

Joulinar, 2022-11-03:
Usually just marking things in Putty will copy the text automatically.

AkaMrBill, 2022-11-03:
root@orangepi4-lts:~# bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/M                                                                         ichaIng/DietPi/master/.build/images/dietpi-installer')"
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Reading package lists... Done
[ INFO ] Selected Git branch: MichaIng/master
[ INFO ] DietPi-PREP | Detected distribution version: Bullseye (ID: 6)
[ INFO ] DietPi-PREP | Detected target CPU architecture: aarch64 (ID: 3)

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [0] Detecting existing DietPi system

[ INFO ] DietPi-PREP | DietPi system found, uninstalling old instance...
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : hostapd
[  OK  ] DietPi-Services | stop : cups
removed '/etc/systemd/system/dietpi-cloudshell.service'
removed '/etc/systemd/system/dietpi-firstboot.service'
removed '/etc/systemd/system/dietpi-fs_partition_resize.service'
removed '/etc/systemd/system/dietpi-kill_ssh.service'
removed '/etc/systemd/system/dietpi-postboot.service'
removed '/etc/systemd/system/dietpi-preboot.service'
removed '/etc/systemd/system/dietpi-ramlog.service'
removed '/etc/systemd/system/dietpi-vpn.service'
removed '/etc/systemd/system/dietpi-wifi-monitor.service'
removed '/boot/dietpi/dietpi-led_control'
removed '/boot/dietpi/dietpi-cpuinfo'
removed '/boot/dietpi/misc/start_kodi'
removed '/boot/dietpi/misc/dietpi-justboom'
removed '/boot/dietpi/misc/dietpi-arr_to_RAM'
removed directory '/boot/dietpi/misc'
removed '/boot/dietpi/dietpi-cleaner'
removed '/boot/dietpi/dietpi-vpn'
removed '/boot/dietpi/postboot'
removed '/boot/dietpi/dietpi-morsecode'
removed '/boot/dietpi/dietpi-letsencrypt'
removed '/boot/dietpi/dietpi-drive_manager'
removed '/boot/dietpi/dietpi-cron'
removed '/boot/dietpi/dietpi-launcher'
removed '/boot/dietpi/dietpi-bugreport'
removed '/boot/dietpi/dietpi-autostart'
removed '/boot/dietpi/.version'
removed '/boot/dietpi/preboot'
removed '/boot/dietpi/dietpi-software'
removed '/boot/dietpi/dietpi-update'
removed '/boot/dietpi/dietpi-survey'
removed '/boot/dietpi/dietpi-services'
removed '/boot/dietpi/dietpi-config'
removed '/boot/dietpi/func/dietpi-obtain_hw_model'
removed '/boot/dietpi/func/dietpi-benchmark'
removed '/boot/dietpi/func/dietpi-set_swapfile'
removed '/boot/dietpi/func/dietpi-globals'
removed '/boot/dietpi/func/dietpi-print_large'
removed '/boot/dietpi/func/dietpi-set_cpu'
removed '/boot/dietpi/func/change_hostname'
removed '/boot/dietpi/func/dietpi-set_software'
removed '/boot/dietpi/func/dietpi-logclear'
removed '/boot/dietpi/func/dietpi-optimal_mtu'
removed '/boot/dietpi/func/create_mysql_db'
removed '/boot/dietpi/func/dietpi-banner'
removed '/boot/dietpi/func/dietpi-ramlog'
removed '/boot/dietpi/func/dietpi-set_hardware'
removed '/boot/dietpi/func/dietpi-wifidb'
removed '/boot/dietpi/func/run_ntpd'
removed '/boot/dietpi/func/dietpi-set_userdata'
removed directory '/boot/dietpi/func'
removed '/boot/dietpi/dietpi-sync'
[... 286 lines cut ...]

 | IP: 192.168.1.133 | Use up/down buttons to scroll text───────────┤ DietPi-PREP ├────────────────────────────────────────────────────┐
                │ APT purge for: dbus dhcpcd5 mountall initscripts chrony *office* *xfce* *qt5* *xserver* *xorg* glib-networking
                │ libgtk-3-0 libsoup2.4-1 libglib2.0-0 lightdm xl2tpd x11-common
                │  - Command: apt-get -y --allow-change-held-packages autopurge dbus dhcpcd5 mountall initscripts chrony *office*
                │ *xfce* *qt5* *xserver* *xorg* glib-networking libgtk-3-0 libsoup2.4-1 libglib2.0-0 lightdm xl2tpd x11-common
                │  - Exit code: 100
                │  - DietPi version: v8.10.2 (MichaIng/master) | HW_MODEL: 24 | HW_ARCH: 3 | DISTRO: 6
                │  - Error log:
                │ Reading package lists...
                │ Building dependency tree...
                │ Reading state information...
                │ Some packages could not be installed. This may mean that you have
                │ requested an impossible situation or if you are using the unstable
                │ distribution that some required packages have not yet been created
                │ or been moved out of Incoming.
                │ The following information may help to resolve the situation:
                │
                │                        Retry          : Re-run the last command that failed                 ↑
                │                        Open subshell  : Open a subshell to investigate or solve the issue   ▮
                │
                │
                │                                  <Ok>                                      <Exit>
                │                                                                                                                      │
                └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

Joulinar, 2022-11-03:
Looks like our installer is not purging all packages, causing some conflict now. I guess we simply do not expect them. Probably they need to be purged manually. @MichaIng could you have a look.

AkaMrBill, 2022-11-04:
They offer four images, debian Buster in Server and Desktop as well as debian Bullseye in Server and Desktop.

https://drive.google.com/drive/folders/1W0Bm-GGvVqgiDeSdEy_PSGlA23ahMGC6

Would trying the "Server" version meet the need? I am more than willing to give it a shot if it might work.

Joulinar, 2022-11-04:
Yes try Bullseye server version.

MichaIng, 2022-11-04:
I suggest to use the "Bullseye CLI" image of Armbian, which will lead most likely to a much cleaner (and faster prepared) DietPi image: https://www.armbian.com/orange-pi-4-lts/

The error you see is because of APT resolver being nasty, not simply purging everything which depends on the packages that are to be purged 😄. I never understood why it does this, which is why the list of packages to be purged, to workaround the issue, has grown further and further. E.g. in your case, doing:
```sh
apt purge chromium-x11 ffmpeg
```
probably another one, should solve the issue, but adding more and more packages isn't a good solution. Probably we should trust again `apt autopurge` to really autopurge everything that is not marked to stay. For some reasons it didn't in the past, without any obvious reasons.

I'll run some tests when I'm back home.

AkaMrBill, 2022-11-04:
I am currently attempting the Server build. If that fails,  I will try the Armbian image.

AkaMrBill, 2022-11-04:
SUCCESS!!   Creating a custom dipository worked with the Server image.
So, now how exactly do I make sure to get a copy of the image to have for future use? I have not powered down yet.

MichaIng, 2022-11-04:
I also removed this active purging from the installer: https://github.com/MichaIng/DietPi/commit/08afa23
This shouldn't be required anymore on any supported Debian version and solves all these resolver errors.

AkaMrBill, 2022-11-04:
OK, two questions: @MichaIng 
1) What are the exact steps I need to follow to save the new "First DietPi boot" image that I created?
2) Would you like me to try the original process I had the error on to help validate your changes?

MichaIng, 2022-11-04:
1. If you have another DietPi system, e.g. a virtual machine or so, you can attach the SD card and run
    ```sh
    bash -c "$(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/dietpi-imager')"
    ```
    to create a shrunk image from it.
2. Only if you find time. A full package list to verify that everything got autoremoved as expected would be useful then:
    ```sh
    dpkg-query -Wf '${Package}\n'
    ```

Joulinar, 2022-11-04:
Every Linux box should be fine to run the imager. Isn't it?

MichaIng, 2022-11-04:
Ah right, every Debian and Ubuntu systems at least (since it pulls packages via APT).

AkaMrBill, 2022-11-08:
So, when retrying the original Orange Pi Debian Desktop, I still fail with the following log.

root@orangepi4-lts:~# bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-installer')"
removed '/etc/apt/apt.conf.d/71-orangepi-no-recommends'
Get:1 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye InRelease [116 kB]
Get:2 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye-updates InRelease [44.1 kB]
Get:3 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye-backports InRelease [49.0 kB]
Get:4 http://mirrors.tuna.tsinghua.edu.cn/debian-security bullseye-security InRelease [48.4 kB]
Get:5 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye/non-free arm64 Packages [88.5 kB]
Get:6 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye/contrib arm64 Packages [49.0 kB]
Get:7 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye/main armhf Packages [10.8 MB]
Get:8 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye/main arm64 Packages [10.9 MB]
Get:9 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye/contrib armhf Packages [47.8 kB]
Get:10 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye/non-free armhf Packages [67.2 kB]
Get:11 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye-updates/main armhf Packages [12.0 kB]
Get:12 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye-updates/main arm64 Packages [12.0 kB]
Get:13 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye-backports/main arm64 Packages [359 kB]
Get:14 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye-backports/main armhf Packages [356 kB]
Get:15 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye-backports/non-free armhf Packages [9064 B]
Get:16 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye-backports/non-free arm64 Packages [9868 B]
Get:17 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye-backports/contrib armhf Packages [3328 B]
Get:18 http://mirrors.tuna.tsinghua.edu.cn/debian bullseye-backports/contrib arm64 Packages [3840 B]
Get:19 http://mirrors.tuna.tsinghua.edu.cn/debian-security bullseye-security/main armhf Packages [242 kB]
Get:20 http://mirrors.tuna.tsinghua.edu.cn/debian-security bullseye-security/main arm64 Packages [242 kB]
Fetched 23.5 MB in 2min 26s (161 kB/s)
Reading package lists... Done
[ INFO ] Selected Git branch: MichaIng/master
[ INFO ] DietPi-PREP | Detected distribution version: Bullseye (ID: 6)
[ INFO ] DietPi-PREP | Detected target CPU architecture: aarch64 (ID: 3)

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [0] Detecting existing DietPi system

[ INFO ] DietPi-PREP | No DietPi system found, skipping old instance uninstall...

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [1] Target system inputs

[ INFO ] DietPi-PREP | Entered image creator: Bill Kendall
[ INFO ] DietPi-PREP | Entered pre-image info: https://drive.google.com/drive/folders/1W0Bm-GGvVqgiDeSdEy_PSGlA23ahMGC6    (Bullseye Desktop)
[ INFO ] DietPi-PREP | Selected hardware model ID: 24
[ INFO ] DietPi-PREP | Marking WiFi as required
[ INFO ] DietPi-PREP | Selected Debian version: bullseye (ID: 6)

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [2] Downloading and installing DietPi source code

[  OK  ] DietPi-PREP | Downloading source code
[  OK  ] DietPi-PREP | Unpacking source code
[  OK  ] DietPi-PREP | Removing unused files
[  OK  ] DietPi-PREP | Hardening source code mode
[ INFO ] DietPi-PREP | Moving kernel and boot configuration to /boot
[  OK  ] DietPi-PREP | mv DietPi-master/dietpi.txt /boot/
[  OK  ] DietPi-PREP | mv DietPi-master/README.md /boot/dietpi-README.md
[  OK  ] DietPi-PREP | mv DietPi-master/LICENSE /boot/dietpi-LICENSE.txt
[  OK  ] DietPi-PREP | Copy DietPi scripts to /boot/dietpi
[  OK  ] DietPi-PREP | Copy DietPi system files in place
[  OK  ] DietPi-PREP | Clean download location
[ INFO ] DietPi-PREP | Storing DietPi version info:
[  OK  ] DietPi-PREP | Desired setting in /boot/dietpi.txt was already set: DEV_GITBRANCH=master
[  OK  ] DietPi-PREP | Desired setting in /boot/dietpi.txt was already set: DEV_GITOWNER=MichaIng
[ INFO ] DietPi-PREP | Applying DietPi live patches to fix known bugs in this version
[  OK  ] DietPi-PREP | cp /boot/dietpi/.version /var/lib/dietpi/.dietpi_image_version
[  OK  ] DietPi-PREP | systemctl daemon-reload

 DietPi-PREP
─────────────────────────────────────────────────────
[... 262 lines cut ...]
          │  chromium-x11 : Depends: libatk-bridge2.0-0 (>= 2.5.3) but it is not going to be installed
          │                 Depends: libatk1.0-0 (>= 2.2.0) but it is not going to be installed
          │                 Depends: libatspi2.0-0 (>= 2.9.90) but it is not going to be installed
          │                 Depends: libgdk-pixbuf2.0-0 (>= 2.22.0) but it is not going to be installed
          │                 Depends: libglib2.0-0 (>= 2.39.4) but it is not going to be installed
          │                 Depends: libgtk-3-0 (>= 3.19.12) but it is not going to be installed
          │                 Depends: libharfbuzz0b (>= 2.2.0) but it is not going to be installed
          │                 Depends: libpango-1.0-0 (>= 1.14.0) but it is not going to be installed
          │                 Depends: libpangocairo-1.0-0 (>= 1.14.0) but it is not going to be installed
          │  ffmpeg : Depends: libsdl2-2.0-0 (>= 2.0.12) but it is not going to be installed
[... cut ...]
          │                        Open subshell  : Open a subshell to investigate or solve the issue   ▮
          │
          │
          │                                  <Ok>                                      <Exit>
          │                                                                                                                      │
          └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
So, somthing is still not right.

MichaIng, 2022-11-08:
You need to use the `dev` branch for pulling the script, until next DietPi release:
```sh
bash -c "$(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/dietpi-imager')"
```

MichaIng, 2022-11-12:
@AkaMrBill
Does it work with the version from the `dev` branch?

AkaMrBill, 2022-11-12:
Once I got it working based on the Armbian CLI build, I stopped testing.  I have now validated the work I was trying to complete.  So, if you still need me to try, I certainly can give it a shot once I return home next week.

MichaIng, 2022-11-18:
Works fine now with the Bullseye image from Orange Pi. Part of the issue might have been that the image ships with ffmpeg packages marked as hold. All packages will be unhold now: https://github.com/MichaIng/DietPi/commit/8bf78fa

closed 2022-11-18 by MichaIng