MichaIng/DietPi #5859: dietpi-update does not work

cronick2021, 2022-11-03:
#### Details:
- Date           | Thu Nov  3 23:16:02 CET 2022
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux Docker 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `curl -sSfLO https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches`
- Exit code      | 6
- Software title | DietPi-Update
#### Steps to reproduce:
I start the "dietpi-update" and it does not run cleanly.
See error message above.

#### Expected behaviour:
It should install the update cleanly.

#### Actual behaviour:
It cancels the update. See error message.

#### Extra details:
Docker is installed on the RaspPi with several containers (e.g. AdGuard).

#### Additional logs:
```
curl: (6) Could not resolve host: raw.githubusercontent.com
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

Can anyone help me?
Thanks

Joulinar, 2022-11-03:
Are you using any AdBlock software inside your network?

>curl: (6) Could not resolve host: raw.githubusercontent.com``

cronick2021, 2022-11-03:
Yes, I use AdGuard Home. That means I just need to disable it and then the update should work?

Joulinar, 2022-11-03:
No, check your network settings in `dietpi-config`. Set a Static IP if not already done and use a global upstream DNS.

closed 2022-11-04 by cronick2021

MichaIng, 2022-11-04:
I'd also add `raw.githubusercontent.com` to the whitelist, there is zero reason to block it. If you can even identify the blocklist which ships this entry, awesome would be to contact that list's maintainer and ask to remove the entry from the list.

Joulinar, 2022-11-04:
@MichaIng issue was not the blocklist. It was due to AGH container being offline as we stop Docker on updates.

MichaIng, 2022-11-04:
Ah sorry for not reading everything closely, AGH + `raw.githubusercontent.com` made me think about a repeating known issue 😄. Ignore my comment above then.