MichaIng/DietPi #5860: dietpi-software: home-assistant new reuirement for install: python3-pycountry

enviousjag, 2022-11-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `bullseye 0`
- Kernel version | `uname -a`
- SBC model | RPi 4 Model B (aarch64)

#### Steps to reproduce
1. install homeassistant via dietpi-software
2. follow the official instructions on restarting the home-assistant service
3. open the browser and acces home-assistant via IP:8123
4. create account 
5. personalize location & locale
6. Error message unable to save X
7. open terminal 
8. journalctl -u home-assistant
9. save output
https://bin.disroot.org/?2714255fda66d13b=#5Rwg1Wrpv2tWtUoMdq1LxKoagWPzFQT2kBqYiEPj8JFo

#### Expected behaviour
python3-pycountry should be installed a added dependency and error will disappear.

MichaIng, 2022-11-04:
Thanks for reporting.

I faced the same error on first login/setup, but was able to continue after reloading the page.

Reported here as well: https://github.com/home-assistant/core/issues/75155
Weird is that `pycountry` is neither contained in HA's `requirements.txt` (so not pulled in by `pip` on module install) nor in `requirements_all.txt`, so never automatically installed on service start, or when accessing certain UI elements. For any HA core dependency, this is what we (need to) count on.

Hence an upstream bug, it looks to me: `pycountry`, required by a component shipped with HA core, would need to be installed when the service starts, or when/before certain UI elements are accessed/show up.

fazzer4x, 2022-11-04:
got it, ty

MichaIng, 2022-11-04:
An issue in `redios`: https://github.com/frenck/python-radios/issues/140
`pycountry` is actually listed as dependency there: https://github.com/frenck/python-radios/blob/main/pyproject.toml#L36
Also the package's sources `setup.py` on PyPI has it defined as dependency. Not sure how it can be missing in the wheel then, or for which other reason it doesn't automatically pulled in as expected 🤔. Makes sense to test with `pip3 install radios`, whether this pulls in `pycountry` or not.

closed 2024-10-03 by enviousjag