MichaIng/DietPi #5861: Dietpi 8.10.2 Unable update Amiberry to 5.4 version

corasaniti, 2022-11-04:
I am trying to update Amiberry from version 5.3 to version 5.4 with the follow command

`dietpi-software reinstall 108`

but I get the following error. 
I am using the latest version of Dietpi 8.10.2
Thanks

[image]

Joulinar, 2022-11-04:
Would be good if you could have filled the bug report with your system details.

corasaniti, 2022-11-04:
> Would be good if you could have filled the bug report with your system details.

Rpi4 8GB
Dietpi 8.10.2 bullseye based 32bit
Thanks

MichaIng, 2022-11-04:
Thanks for reporting.

The 32-bit packages are for Raspbian and Debian at the same time, and hence need to be compiled on Raspbian to be compatible, which has `+rpiX` appended to the version string in some cases. The build script expects `+rpiX` to be at the end (usually it is), but in the case it's before the Debian build appendix 🤔. I'll rebuild the 32-bit RPi Armbian packages to remove everything after and including `+rpiX`.

Joulinar, 2022-11-04:
another report on our forum https://dietpi.com/forum/t/dietpi-amiberry-distri-broken/14841

MichaIng, 2022-11-05:
New packages are ready: https://dietpi.com/downloads/binaries/bullseye/testing/
```sh
cd /tmp
curl -LO 'https://dietpi.com/downloads/binaries/bullseye/testing/amiberry_rpi4.deb'
apt install ./amiberry_rpi4.deb
rm amiberry_rpi4.deb
```
Would be great if you could test with an actual game, then I move them into stable download place.

corasaniti, 2022-11-05:
@MichaIng 
Thanks in advance,
If i perform this workaround, when i then upgrade to dietpi 8.10.11 will i have any problems?

MichaIng, 2022-11-05:
> If i perform this workaround, when i then upgrade to dietpi 8.10.11 will i have any problems?

Nope, you won't cause any issue with the update.

corasaniti, 2022-11-05:
@MichaIng 
I updated Amiberry to version 5.4. Everything seems to be working fine. 
Thanks

MichaIng, 2022-11-05:
Great, thanks for testing 👍.

closed 2022-11-08 by MichaIng