MichaIng/DietPi #5862: Wordpress - database issue

aterhorst, 2022-11-05:
### Creating a bug report/issue

I followed the instruction at https://dietpi.com/docs/software/social/#wordpress (first-time connect). I get this error message:

The database server could be connected to (which means your username and password is okay) but the database could not be selected.

- Are you sure it exists?
- Does the user wordpress have permission to use the database?

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye 1
- Kernel version | `uname -a`
Linux DietPi 5.15.74-v7+ #1595 SMP Wed Oct 26 11:03:05 BST 2022 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 3 Model B+ (armv7l)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Wordpress
- Was the software title installed freshly or updated/migrated?
Fresh
- Can this issue be replicated on a fresh installation of DietPi?
Have yet to try.

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. http:/<IP address>/wordpress
2. Prompted to enter credentials. Entered as indicated https://dietpi.com/docs/software/social/#wordpress (first-time connect)

#### Expected behaviour

- I should be able to launch wordpress

#### Actual behaviour

- I get an error message

#### Extra details

- ...

MichaIng, 2022-11-05:
Many thanks for your report.

Can you check:
```sh
mysql -e 'show databases; show grants for wordpress@localhost;'
```

aterhorst, 2022-11-06:
I resolved the issue. The first-time login has "wordpress" in the database and user fields. I assumed these were default. However, I discovered one has to type "wordpress" out, even though it appears in the fields. Then everything works.

closed 2022-11-06 by aterhorst

MichaIng, 2022-11-06:
Ah okay, that is not too intuitive indeed 🙂.