MichaIng/DietPi #5863: Update failed. There is my report. Please help :)

Tazmanischerteufel, 2022-11-05:
#### Details:
- Date           | Sat Nov  5 13:21:42 GMT 2022
- Bug report     | 8818caa0-817c-4e3d-afad-18c2ffc7cec7
- DietPi version | v8.6.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Type 'curl' is not known on line 1 in source list /etc/apt/sources.list.d/plexmediaserver.list
E: The list of sources could not be read.
```

MichaIng, 2022-11-05:
Many thanks for reporting. Not sure if we had a bug in there? Can you please show:
```sh
cat /etc/apt/sources.list.d/plexmediaserver.list
```

MichaIng, 2023-02-12:
Marking as closed as of outstanding reply. Feel free to reopen if required.

closed 2023-02-12 by MichaIng