MichaIng/DietPi #5864: Can't update my dietpi.

InfoALR, 2022-11-05:
Dear sirs or madams,

I want do update my dietpi installation. over the "dietpi-launcher" menu.

I received the following message:

[ INFO ] DietPi-Update | APT update, please wait...
E: Conflicting values set for option Signed-By regarding source https://download.docker.com/linux/debian/ bullseye: /etc/apt/keyrings/docker.gpg !=
E: The list of sources could not be read.
[FAILED] DietPi-Update | APT update
 - Command: apt-get -y update
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
/boot/dietpi/dietpi-bugreport: line 138: aplay: command not found
/boot/dietpi/dietpi-bugreport: line 138: aplay: command not found
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: bad93ed0-7972-4af2-88cf-1d1cc686a285

Thanks a lot for your help.

MichaIng, 2022-11-06:
Your have two conflicting Docker repo entries, one in `/etc/apt/sources.list.d/docker.list` and one in `/etc/apt/sources.list.d/omvextras.list`. I suggest to remove the weaker entry from the 2nd file:
```sh
sed -i '/docker/d' /etc/apt/sources.list.d/omvextras.list
apt update
```
But it will be likely recreated on `openmediavault-omvextrasorg` package upgrade. Not sure how/why you installed Docker again separately, with the official installer, I guess? This is basically conflicting. Note that OMV is extremely intrusive, changing and expecting the system to be OMV only and quickly conflicting with other/custom setup, including many DietPi software and config options.

MichaIng, 2023-02-12:
Marking as closed as of outstanding reply. Feel free to reopen if required.

closed 2023-02-12 by MichaIng