MichaIng/DietPi #5865: Error when performing rpi-eeprom-update

Trunkzeh, 2022-11-06:
Running Dietpi v8.10.2 on RPI4. Performed an apt upgrade this morning and there was an update to the rpi-eeprom package; taking me from version 13.19-1 to 13.21-1. Upon running rpi-eeprom-update -a; the following is presented:

```
*** INSTALLING EEPROM UPDATES ***

BOOTLOADER: update available
   CURRENT: Tue Oct 18 10:57:44 UTC 2022 (1666090664)
    LATEST: Fri Nov  4 15:19:15 UTC 2022 (1667575155)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
   CURRENT: Tue Oct 18 10:57:44 UTC 2022 (1666090664)
    UPDATE: Fri Nov  4 15:19:15 UTC 2022 (1667575155)
    BOOTFS: /boot
xxd not found. Try installing the xxd package.
```
Issue was resolved by installing the xxd package; current version fetched by apt is 2.8.2

Trunkzeh, 2022-11-06:
Checking the rpi-eeprom github; it seems this is a known issue that was reported recently:

https://github.com/raspberrypi/rpi-eeprom/issues/450

Looks to be a regression; we can either wait for it to be fixed (again) by the rpi-eeprom maintainers or add the xxd package in the interim :)

MichaIng, 2022-11-06:
The reason it got explicitly removed from dependencies: https://github.com/raspberrypi/rpi-eeprom/issues/383
Need to check whether some recent commit now checks for it/requires it despite no explicit key signing used.

MichaIng, 2022-11-07:
Solved with next package upgrade. Making this as closed.

closed 2022-11-07 by MichaIng