MichaIng/DietPi #5868: Grafana repository changed its 'Origin' value

MichaIng, 2022-11-08:
If Grafana is installed, DietPi updates, software installs and other non-interactive `apt update`/`apt-get update` calls can currently fail with:
```
E: Repository 'https://packages.grafana.com/oss/deb stable InRelease' changed its 'Origin' value from 'grafana stable' to '. stable'
E: Repository 'https://packages.grafana.com/oss/deb stable InRelease' changed its 'Label' value from 'grafana stable' to '. stable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
```
These repo meta info changes must be accepted interactively once. For this do:
```sh
apt update
```
When being asked, type `y` and return to confirm the change:
```
Do you want to accept these changes and continue updating from this repository? [y/N] y
```

Joulinar, 2022-11-08:
How are we dealing with this in a DietPi update? Should we check beforehand and inform user to run `apt update` manually?

MichaIng, 2022-11-08:
We could even pass a flag in pre-patches, like we did when Bullseye was released and suite name hence changed. But I'll wait until close to release, just in case this changes back. The new label with a dot inside looks like an accident.

RobiWahnKnobi, 2022-11-11:
Just fyi: As seen on https://grafana.com/docs/grafana/latest/setup-grafana/installation/debian/ this change won't be reverted, it's moving to a different repository: https://apt.grafana.com stable main

From the Website:
Repository migration (November 8th 2022)
From that date, Grafana packages will be served from a new repository (<packages.grafana.com/deb/{product}> -> <apt.grafana.com>). The new repository serves, from a single APT configuration, all Grafana OSS products, as well as Grafana Enterprise.

The old URLs will still work, serving the content from the new repository, but you may encounter warnings about some repository attributes changing (e.g. Origin and Label).

MichaIng, 2022-11-11:
Thanks for the info, so we should use the new repo and patch it for our users on DietPi update as well.

MichaIng, 2022-11-13:
Migration is now done via pre-patch: https://github.com/MichaIng/DietPi/commit/8e34d5c

closed 2022-11-18 by MichaIng