MichaIng/DietPi #5869: Enable cachedb for unbound

jeremiah-miller, 2022-11-08:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
- I'm trying to set up redis as a persistent cache for unbound (https://unbound.docs.nlnetlabs.nl/en/latest/manpages/unbound.conf.html?highlight=cache#cache-db-module-options) but apparently the cachedb module isn't compiled in for dietpi so it's not possible to do that.

#### Describe the solution you'd like:

- compile unbound with the `--with-libhiredis` option so that redis can be used
- Maybe add an option to automatically install and configure persistent cache with redis when install unbound via dietpi-software

#### Describe alternatives you've considered:

- As far as I know there are no other cachedb backends

jeremiah-miller, 2022-11-09:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1014456 looks like there's an upstream bug too

Joulinar, 2022-11-09:
An interesting feature but it would need to be implemented upstream. For us this is out of scope.

MichaIng, 2022-11-09:
For my understanding: This only works if Redis runs on a dedicated server, right? Because Redis itself is a non-persistent memory-based database, so with it running on the same server, a reboot wouldn't help Unbound to restore anything, is it?

I suggest to contribute to the Debian "bug report" then, underlining the use case where it is important to have immediate fast DNS after a server or service restart, justifying possible additional dependencies, larger binary or such.

jeremiah-miller, 2022-11-09:
No, i have it set up on my laptop running both unbound and redis on the same machine. Redis is a cache based db but it also writes to disk so the db can be loaded back into memory on boot

MichaIng, 2022-11-09:
> but it also writes to disk

You mean what on Debian by default is written to `/var/lib/redis/dump.rdb` is a full database dump? I was actually never really sure about this. It's written to on a set of X amount of change over Y time rules, but for DNS cache whether the last X queries are stored or not usually doesn't matter much, and it's surely dumped as well on shutdown/service stop. I also read that Unbound will never remove anything from the Redis database, so it accumulates until Redis-side limits are hit.

I guess it is important to make clear to Debian maintainers in which case such persistent DNS cache is important, to satisfy efforts and overhead. For the major Unbound use cases we face (in combination with Pi-hole/AGH), whether the (only) first query for a hostname after a server/service restart takes 0.007 or 3 seconds doesn't really matter. A very minor convenience, which also for the admin isn't really worth the effort of setting up an extra Redis server with added complexity. Tell'em why this is a too narrow view 😉.

closed 2022-11-24 by MichaIng