MichaIng/DietPi #5870: [FAILED] DietPi-Software | systemctl start mariadb

eskimos2000, 2022-11-09:
#### Details:
- Date           | Wed Nov  9 13:17:13 AEDT 2022
- Bug report     | d6084f00-71d7-418f-97fb-271fced0bf52
- DietPi version | v8.10.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Odroid C4/HC4 (aarch64) (ID=16)
- Kernel version | `Linux DietPi 5.10.147-meson64 #22.08.4 SMP PREEMPT Fri Oct 7 16:37:19 UTC 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `systemctl start mariadb`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. I wanted to install IceCast and o!mpd via dietpi-software
2. MPD was already installed.
#### Expected behaviour:

- Installed IceCast and o!mpd
#### Actual behaviour:

- Above error and software was not installed
#### Extra details:

- ...
#### Additional logs:
```

Joulinar, 2022-11-09:
Can you share following

```
systemctl restart mariadb.service
journalctl -u mariadb
cat /var/log/mysql/error.log
readlink /var/lib/mysql
readlink -f /var/lib/mysql
```

eskimos2000, 2022-11-09:
1. Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
2. -- Journal begins at Wed 2022-11-09 12:17:02 AEDT, ends at Thu 2022-11-10 09:57:10 AEDT. --
Nov 09 19:58:28 DietPi systemd[1]: Starting MariaDB 10.5.15 database server...
Nov 09 19:58:29 DietPi mariadbd[1963]: 2022-11-09 19:58:29 0 [Note] /usr/sbin/mariadbd (mysqld 10.5.15-MariaDB-0+deb11u1) starting as process 1963 ...
Nov 09 19:58:29 DietPi mariadbd[1963]: 2022-11-09 19:58:29 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Nov 09 19:58:29 DietPi mariadbd[1963]: [92B blob data]
Nov 09 19:58:29 DietPi mariadbd[1963]: 2022-11-09 19:58:29 0 [ERROR] Aborting
Nov 09 19:58:29 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Nov 09 19:58:29 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Nov 09 19:58:29 DietPi systemd[1]: Failed to start MariaDB 10.5.15 database server.
Nov 09 19:58:31 DietPi systemd[1]: Starting MariaDB 10.5.15 database server...
Nov 09 19:58:31 DietPi mariadbd[2178]: 2022-11-09 19:58:31 0 [Note] /usr/sbin/mariadbd (mysqld 10.5.15-MariaDB-0+deb11u1) starting as process 2178 ...
Nov 09 19:58:31 DietPi mariadbd[2178]: 2022-11-09 19:58:31 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Nov 09 19:58:31 DietPi mariadbd[2178]: [92B blob data]
Nov 09 19:58:31 DietPi mariadbd[2178]: 2022-11-09 19:58:31 0 [ERROR] Aborting
Nov 09 19:58:31 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Nov 09 19:58:31 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Nov 09 19:58:31 DietPi systemd[1]: Failed to start MariaDB 10.5.15 database server.
Nov 10 09:57:10 DietPi systemd[1]: Starting MariaDB 10.5.15 database server...
Nov 10 09:57:10 DietPi mariadbd[18052]: 2022-11-10  9:57:10 0 [Note] /usr/sbin/mariadbd (mysqld 10.5.15-MariaDB-0+deb11u1) starting as process 18052 ...
Nov 10 09:57:10 DietPi mariadbd[18052]: 2022-11-10  9:57:10 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Nov 10 09:57:10 DietPi mariadbd[18052]: [92B blob data]
Nov 10 09:57:10 DietPi mariadbd[18052]: 2022-11-10  9:57:10 0 [ERROR] Aborting
Nov 10 09:57:10 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Nov 10 09:57:10 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Nov 10 09:57:10 DietPi systemd[1]: Failed to start MariaDB 10.5.15 database server.
3. cat: /var/log/mysql/error.log: No such file or directory
4. /mnt/dietpi_userdata/mysql
5. /mnt/ssd1/dietpi_userdata/mysql

Hope that will help you to fix it, thanks.

Joulinar, 2022-11-09:
> Nov 09 19:58:31 DietPi mariadbd[2178]: 2022-11-09 19:58:31 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test

Somehow an issue to write test file

can you perform following test

```
ls -la /mnt/dietpi_userdata/mysql
sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
```

eskimos2000, 2022-11-09:
1. touch: cannot touch '/var/lib/mysql/demo.file': Permission denied
2. `total 110604
drwxr-xr-x  2 mysql  mysql       4096 Nov  9 13:14 .
drwxrwxr-x 11 dietpi dietpi      4096 Nov  9 13:16 ..
-rw-r--r--  1 mysql  mysql          0 Nov  9 13:14 debian-10.5.flag
-rw-rw----  1 mysql  mysql   12582912 Nov  9 13:14 ibdata1
-rw-rw----  1 mysql  mysql  100663296 Nov  9 13:14 ib_logfile0`

I have to go so be back late evening. Thanks mate for quick respond, awesome.

Joulinar, 2022-11-09:
hmm permission seems to fit but database user is not able to write to the directory 🤔 
As well, there seems to be something missing on that directory. There should be more inside. Can you share following as well.

```
ls -la /var/lib/mysql/
```

eskimos2000, 2022-11-10:
Here you have it:

`total 110604
drwxr-xr-x  2 mysql  mysql       4096 Nov  9 13:14 .
drwxrwxr-x 11 dietpi dietpi      4096 Nov  9 13:16 ..
-rw-r--r--  1 mysql  mysql          0 Nov  9 13:14 debian-10.5.flag
-rw-rw----  1 mysql  mysql   12582912 Nov  9 13:14 ibdata1
-rw-rw----  1 mysql  mysql  100663296 Nov  9 13:14 ib_logfile0`

Joulinar, 2022-11-10:
ok that's the same content. What is still strange, that there are just 3 files inside and it seems not writable. Do you have a backup of your system? Btw, can you check for kernel error messages.

```
dmesg -l err,crit,alert,emerg
```

MichaIng, 2022-11-10:
Most likely filesystem error and hence auto R/O remount. Can be also checked via:
```sh
mount | grep '/mnt/ssd1'
```
This again probably due to SSD being powered via USB only, without dedicated PSU? Odroids may have less limits than RPi, but I still would not rely on this, depending on what else is attached via USB.

And finally, which PCB revision is it, and which port did you use? https://forum.odroid.com/viewtopic.php?t=44967

eskimos2000, 2022-11-11:
Hi guys,

Here you have: dmesg

`[    1.215018] reg-fixed-voltage regulator-flash_1v8: Failed to register regulator: -517
[    1.215856] reg-fixed-voltage regulator-vcc_1v8: Failed to register regulator: -517
[    1.215962] reg-fixed-voltage regulator-vcc_3v3: Failed to register regulator: -517
[    1.216182] reg-fixed-voltage regulator-vddao_1v8: Failed to register regulator: -517
[    2.003852] thermal thermal_zone0: binding zone cpu-thermal with cdev thermal-cpufreq-0 failed:-17
[    2.040747] reg-fixed-voltage regulator-flash_1v8: Failed to register regulator: -517
[    2.043309] reg-fixed-voltage regulator-vcc_1v8: Failed to register regulator: -517
[    4.798930] panfrost ffe40000.gpu: dev_pm_opp_set_regulators: no regulator (mali) found: -19`

and mount:

`systemd-1 on /mnt/ssd1 type autofs (rw,relatime,fd=48,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=585)
/dev/sda1 on /mnt/ssd1 type ext4 (rw,noatime,lazytime,x-systemd.automount)`

MichaIng, 2022-11-12:
Those outputs look good, no R/O flag. This still fails?
```sh
sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
sudo -u mysql -g mysql touch /mnt/ssd1/dietpi_userdata/mysql/demo.file
```

eskimos2000, 2022-11-14:
I read about that USB port problem and actually, I had my SSD connected to that right bottom one which people reported issues with! I swapped it to the left top so are readings look good and I reinstalled that soft all is working.  So all good thanks, guys.

MichaIng, 2022-11-14:
Okay great, thanks for reporting back. I'll mark this issue as closed then.

closed 2022-11-14 by MichaIng