MichaIng/DietPi #5871: A weird problem: when compiling the arm64 image device use #make-your-own-distribution

hifitime, 2022-11-10:
Buster with Armbian Linux ttyMV0
SBC device | Generic Device (aarch64)
Power supply used | 5V 2A RAVpower
SDcard used | SanDisk ultra)

when compiling the arm64 image device use #make-your-own-distribution
https://dietpi.com/docs/hardware/#make-your-own-distribution

After I successfully compiled step by step, I pretended to be dead when I entered the system interface after restarting. There was no response. The computer could not be found on the network. I could not enter the system in any way. There was no response!

**This is the compilation process information（Because it is too long, I generated a file link~）:**

[http://open.kanshangjie.com/build%20for%20arm%2064.txt](url)

**The following is the output information of the first startup. It seems that everything is normal. When the last step is about to enter the system, there is no response...（Because it is too long, I generated a file link~）**

[fitst boot .txt](url)

Joulinar, 2022-11-10:
Looks like your system is not getting an IP address assigned via Ethernet

```
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 3
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 5
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 8
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 14
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 18
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 6
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
```

Joulinar, 2022-11-10:
yes because it does not get an IP address assigned. You could try to use STATIC IP instead of DHCP.

hifitime, 2022-11-10:
But I access the device through a serial connection, not through the network
Use a serial connection to gain access to the console

Now the serial console also crashes, and there is no response~

hifitime, 2022-11-10:
Normally, even if there is no network card or the network card is not started, the serial console access will not be affected

hifitime, 2022-11-10:
> Normally, even if there is no network card or the network card is not started, the serial console access will not be affected

Thank you very much for your reply, but I still haven't found the problem... Maybe I'm too stupid...: (

hifitime, 2022-11-10:
Is it possible that the CPU frequency setting problem caused the system to crash when entering the system?

hifitime, 2022-11-10:
I repeated the test three times according to the following build link address, and the compiled system failed to enter and crash each time...: (

hifitime, 2022-11-10:
I have repeatedly tested for 3 times by referring to the build link. Unfortunately, the compiled system cannot enter and crash each time...: (

MichaIng, 2022-11-10:
Which SBC is it, actually? There are some (Amlogic especially) having boot issues with recent Armbian kernel upgrade. Please also try this on the Armbian image before applying the Debian installer, to assure it is not an Armbian issue:
```sh
apt update
apt full-upgrade
reboot
```

> Is it possible that the CPU frequency setting problem caused the system to crash when entering the system?

Rare but possible. Try to change it from `schedutil` to `ondemand` in `/boot/dietpi.txt` right after you applied the installer and before reboot.

> Now the serial console also crashes, and there is no response~

Which serial console device does it use (like `/dev/ttyS0`)? And which baud rate does it use (can be usually seen in `/boot/boot.cmd`)?

hifitime, 2022-11-10:
The sbc is customized by us. 

On the Armbian image（Armbian 22.08 Bullseye）, everything is normal. It can be started normally and enter the system interface

MichaIng, 2022-11-10:
> The sbc is customized by us.

Which Armbian image do you use then, or which target platform, if you used the Armbian image builder?

Please also go through the other points mentioned above.

hifitime, 2022-11-10:
Armbian  Espressobin  It is a reference platform for us

We use this for reference to build

hifitime, 2022-11-10:
Thank you for all your help. I really like the dietpi system~

hifitime, 2022-11-10:
> > The sbc is customized by us.
> 
> Which Armbian image do you use then, or which target platform, if you used the Armbian image builder?
> 
> Please also go through the other points mentioned above.

Thank you for all your help. I really like the dietpi system~

hifitime, 2022-11-24:
`
 ─────────────────────────────────────────────────────
 DietPi v8.11.2 : 20:32 - Thu 11/24/22
 ─────────────────────────────────────────────────────
 - LAN IP : fe80::251:82ff:fe11:2200 (eth0)

 Default Login:
 Username = root
 Password = dietpi (or custom dietpi.txt entry)

 Please hit <return> to login`

Every time I missed the last step of success,,

So sad

Joulinar, 2022-11-24:
looks like you don't have a valid IP address.

hifitime, 2022-11-24:
> But I access the device through a serial connection, not through the network Use a serial connection to gain access to the console
> 
> Now the serial console also crashes, and there is no response~

But I access the device through a serial connection, not through the network
Use a serial connection to gain access to the console

Now the serial console also crashes, and there is no response~

hifitime, 2022-11-24:
Output information of reboot:
[](http://open.kanshangjie.com/secondrestart.txt)

MichaIng, 2022-11-24:
Did you actually hit `return` key as stated in the pre-login banner?

chips023, 2022-11-25:
Sure, but the keyboard has no response. I'm sure it's down and the dietpi host can't be found on the router.

MichaIng, 2022-11-27:
I just see that it failed to set CPU governor settings, which alone is strange on a physical system as well 🤔. However, this rules out that `schedutil` is the issue.

getty on `tty1` and `ttyS0` are both started, so there should be a login prompt on both. The UART works with default baud rate 115200 or there is an override in `/etc/systemd/system/serial-getty@ttyS0.service.d`? However, this isn't touched on generic devices by the installer.

Not sure. If you used a fully functional Armbian image as basis, the installer shouldn't break it since kernel, bootloader, boot config, firmware and all such are not touched, or the Espressobin image is composed very differently, e.g. ships with other essential packages.

chips023, 2022-11-27:
The UART works with rate 115200
then 
I mount the sd card on the ubuntu, and then see /etc/systemd/system/ :

```
root@wu-Ub:/etc/systemd/system# ls
 bluetooth.target.wants
 cloud-final.service.wants
 dbus-fi.w1.wpa_supplicant1.service
 dbus-org.bluez.service
 dbus-org.freedesktop.Avahi.service
 dbus-org.freedesktop.ModemManager1.service
 dbus-org.freedesktop.nm-dispatcher.service
 dbus-org.freedesktop.oom1.service
 dbus-org.freedesktop.resolve1.service
 dbus-org.freedesktop.thermald.service
 dbus-org.freedesktop.timesync1.service
 default.target.wants
 display-manager.service
 display-manager.service.wants
 emergency.target.wants
 final.target.wants
 getty.target.wants
 graphical.target.wants
 multi-user.target.wants
 network-online.target.wants
 nfs-client.target.wants
 oem-config.service.wants
 paths.target.wants
 printer.target.wants
 prltoolsd.service
 prltools-reconfig.service
 remote-fs.target.wants
 rescue.target.wants
 sleep.target.wants
 snap-bare-5.mount
 snap-core20-1587.mount
 snap-core20-1695.mount
 snap-firefox-1635.mount
 snap-firefox-2088.mount
'snap-gnome\x2d3\x2d38\x2d2004-112.mount'
'snap-gnome\x2d3\x2d38\x2d2004-119.mount'
'snap-gtk\x2dcommon\x2dthemes-1535.mount'
 snap-snapd-16292.mount
 snap-snapd-17576.mount
'snap-snapd\x2ddesktop\x2dintegration-14.mount'
'snap-snap\x2dstore-582.mount'
'snap-snap\x2dstore-599.mount'
 sockets.target.wants
 sudo.service
 sysinit.target.wants
 syslog.service
 timers.target.wants
'var-snap-firefox-common-host\x2dhunspell.mount'
root@wu-Ub:/etc/systemd/system# 
```

MichaIng, 2022-11-27:
I was reading through the notes of Armbian on the image: https://www.armbian.com/espressobin/

There seem to be different models with different (stable) CPU frequency and it states that flashing a matching U-Boot to SPI flash is mandatory. Our installer runs the command to reflash U-Boot, although the one to flash to the image, not to the SPI flash. Not sure what it does exactly on this system where booting without SPI seems to be not possible? I'll check that when I'm home.

chips023, 2022-11-27:
Thank you. Look forward to your good news

Cheers!

chips023, 2022-12-02:
hi MichaIng：
is the  any news?

thank u

MichaIng, 2022-12-02:
I just checked the script, and as long it is not a specific board variant, flashing U-Boot is a no-op, so the `dietpi-installer` does not change anything that end:
```sh
DIR=/usr/lib/linux-u-boot-current-espressobin_22.11.1_arm64
write_uboot_platform () 
{ 
    if [[ $BOARD = macchiatobin-doubleshot ]]; then
        dd if=$1/flash-image.bin of=$2 bs=512 seek=1 status=noxfer > /dev/null 2>&1;
    else
        /bin/true;
    fi
}
```

However, I think I found it. Is the serial console device you access with on Armbian `ttyMV0`? This serial device name pattern is not part of the serial devices, enabled in `dietpi-installer`. You can check the used console device via:
```sh
tty
```

chips023, 2022-12-03:
hi~

root@wu:~# tty
/dev/ttyMV0

MichaIng, 2022-12-03:
Seems so. Just to assure this really is the issue. After `dietpi-installer` has completed, run:
```sh
systemctl enable serial-getty@ttyMV0
```
and see whether you can now login as expected. If so, I'll add this serial device naming scheme to our scripts.

chips023, 2022-12-03:
Yes, so I can enter the system. Thank you,
please add this serial device naming scheme to scripts.

chips023, 2022-12-03:
cheers

chips023, 2022-12-03:
The next problem is the network link. It seems that the IP address can not be obtained correctly

MichaIng, 2022-12-03:
Indeed, no DHCP offer has been received on the `eth0` interface. Can you show the output of:
```sh
ip a
```

chips023, 2022-12-03:
> Indeed, no DHCP offer has been received on the `eth0` interface. Can you show the output of:
> 
> ```shell
> ip a
> ```

```
root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1024
    link/ether f0:ad:4e:08:00:00 brd ff:ff:ff:ff:ff:ff
```

MichaIng, 2022-12-03:
Does it work to apply a static IP? And I see you have IPv6 disabled, right? Does it change something when you enable it?

Weird is this output:
```
[  OK  ] DietPi-Config | Checking IPv4 network connectivity
```
This is not supposed to be printed by this script, especially not at this place. [Here is the related code](https://github.com/MichaIng/DietPi/blob/8b71d7c/dietpi/dietpi-config#L2032-L2034):
```sh
		# Restart network
		G_DIETPI-NOTIFY 2 'Restarting network connections, please wait...'
		G_EXEC systemctl daemon-reload
```
So you should instead see:
```
[  OK  ] DietPi-Config | systemctl daemon-reload
```
And of course the IPv4 connection check (which is a `dietpi-globals` function, called by `dietpi-update` and `dietpi-software`, but not by `dietpi-config`) shouldn't succeed as long as no IPv4 address has been assigned.

Can you also check carrier/link status, please:
```sh
cat /sys/class/net/eth0/carrier
ethtool eth0
```

chips023, 2022-12-03:
hi～ @MichaIng 

Does not work to apply a static IP， yes，i  have IPv6 disabled, But nothing works

root@DietPi:~# cat /sys/class/net/eth0/carrier
1
root@DietPi:~# ethtool eth0
Settings for eth0:
        Supported ports: [ MII ]
        Supported link modes:   1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: No
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Supports Wake-on: d
        Wake-on: d
        Link detected: yes

MichaIng, 2022-12-03:
Hmm, I'm a little out-questioned then. The `dietpi-installer` keeps the Armbian kernel and firmware installed, otherwise does really nothing special about the network interfaces. AFAIK Armbian uses NetworkManager by default, but `ifupdown` and `isc-dhcp-server` are the Debian default (lower level) network tools which should just work the same way. The Ethernet link/carrier signal are there, it's just that the SBC does not receive an answer from the router on its DHCP requests.

Are there any kernel errors shown?
```sh
dmesg -l emerg,alert,crit,err
```

Ah, and just to rule out an issue with the recent kernel release, can you test to upgrade everything on the Armbian image and see whether Ethernet still works?
```sh
apt update
apt full-upgrade
reboot
```

chips023, 2022-12-03:
hi～ MichaIng，There is no error message：
root@DietPi:~# dmesg -l emerg,alert,crit,err

I tested it right away on Armbian. Please give me a few minutes and come back soon...

chips023, 2022-12-03:
When I upgrade, the network does not start automatically, I need to execute:
ifconfig eth0 up
dhclient wan

After execution, I can get the LAN IP, but I can't resolve the domain name, and then I add dns to /etc/resolv.conf to resolve the domain name normally

chips023, 2022-12-03:
The Armbian Forum confirmed the existence of the problem by:
systemctl unmask systemd-networkd; systemctl start systemd-networkd

But I executed the above code on the dietpi system, and the reboot system still did not run normally...

chips023, 2022-12-03:
root@DietPi:~# systemctl status networking.service
● networking.service - Raise network interfaces
     Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2022-12-03 17:55:40 GMT; 1min 21s ago
       Docs: man:interfaces(5)
    Process: 779 ExecStart=/sbin/ifup -a --read-environment (code=exited, status=1/FAILURE)
   Main PID: 779 (code=exited, status=1/FAILURE)
        CPU: 6ms

Dec 03 17:55:40 DietPi systemd[1]: Starting Raise network interfaces...
Dec 03 17:55:40 DietPi ifup[779]: ifup: /etc/network/interfaces:11: option with empty value
Dec 03 17:55:40 DietPi ifup[779]: ifup: couldn't read interfaces file "/etc/network/interfaces"
Dec 03 17:55:40 DietPi systemd[1]: networking.service: Main process exited, code=exited, status=1/FAILURE
Dec 03 17:55:40 DietPi systemd[1]: networking.service: Failed with result 'exit-code'.
Dec 03 17:55:40 DietPi systemd[1]: Failed to start Raise network interfaces.

chips023, 2022-12-03:
Armbian's configuration information is as follows:
root@wu:~# cat /etc/network/interfaces
source /etc/network/interfaces.d/*
# Network is managed by Network manager
auto lo
iface lo inet loopback

MichaIng, 2022-12-03:
There are a few things mixed up. First of all there are different network stacks used:
- Armbian, AFAIK uses NetworkManager by default, which implies DHCP: `systemctl status NetworkManager`
  Since it has an own config files, it is expected that `/etc/network/interfaces` is basically empty and the `networking.service` is disabled, respectively, when starting, does nothing or cannot succeed.
- `systemd-networkd` is a different network stack/tool, which implies DHCP as well: `systemctl status systemd-networkd`
  It uses own config files below `/etc/systemd/network`, hence similarly `/etc/network/interfaces` is unused and `networking.service` does nothing.
- DietPi uses `ifupdown` in combination with `isc-dhcp-server`/`dhclient`. Here you run `ifup eth0` to bring everything up, and it uses `/etc/network/interfaces`. It however does not use `networking.service` for bringing up anything else than the loopback interface. For actual interfaces, since they are added via `allow-hotplug eth0` to `/etc/network/interfaces`, they are started via `ifup@eth0.service` as fast as the adapter is detected by the kernel, so nothing fails if e.g. a USB adapter is not attached.

And generally `ifconfig` is a legacy tools, not pre-installed on DietPi anymore. Nowadays one uses `ip` instead:
```sh
ip l set dev eth0 up
```
If `iface eth0 inet dhcp` is defined in `/etc/network/interfaces`, this is done together with proper `dhclient` startup via:
```sh
ifup eth0
```

chips023, 2022-12-04:
hi～ MichaIng:
Looks to use
`ifup eth0`
It also failed to succeed

root@DietPi:~# cat /etc/network/interfaces
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
address
netmask
#gateway .1
#dns-nameservers 9.9.9.9

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address
netmask
#gateway .1
#dns-nameservers 9.9.9.9
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

MichaIng, 2022-12-04:
If the Ethernet port shall be configured automatically at boot, the `#allow-hotplug eth0` line needs to be uncommended `allow-hotplug eth0`. However, the issue is the same, no answer on DHCP requests.

But just to be sure, regarding Armbian, since I couldn't find any recent topic on their forum for Espressobin or mvebu64, this broke Ethernet on Armbian as well?
```sh
apt update
apt full-upgrade
reboot
```
Because then we don't need to continue debugging, because it's a kernel bug then.

chips023, 2022-12-05:
ok thanks

MichaIng, 2022-12-05:
Did you solve the network issue?

chips023, 2022-12-08:
> Did you solve the network issue?

No, I really can't find a solution...

MichaIng, 2022-12-10:
As said, DietPi does not ship with a `systemd-networkd` config, but ifupdown and `/etc/network/interfaces` instead. However, it could be tested. For this, disable Ethernet via `dietpi-config`, then:
```sh
echo -e '[Match]\nName=eth0\n\n[Network]\nDHCP=ipv4' > /etc/systemd/network/eth0.network
systemctl enable --now systemd-networkd
```

MichaIng, 2022-12-11:
Is this bridge there intentionally?

What doea the service say?
```sh
journalctl -u systemd-networkd
```

chips023, 2022-12-11:
> 

root@DietPi:~# journalctl -u systemd-networkd
-- Journal begins at Sat 2022-12-03 20:05:23 GMT, ends at Sat 2022-12-03 20:14:26 GMT. --
Dec 03 20:07:26 DietPi systemd[1]: Stopping Network Service...
Dec 03 20:07:26 DietPi systemd-networkd[682]: eth0: DHCPv6 lease lost
Dec 03 20:07:26 DietPi systemd-networkd[682]: br0: DHCPv6 lease lost
Dec 03 20:07:26 DietPi systemd[1]: systemd-networkd.service: Succeeded.
Dec 03 20:07:26 DietPi systemd[1]: Stopped Network Service.
Dec 03 20:07:26 DietPi systemd[1]: Starting Network Service...
Dec 03 20:07:26 DietPi systemd-networkd[24572]: br0: netdev ready
Dec 03 20:07:26 DietPi systemd-networkd[24572]: br0: Gained IPv6LL
Dec 03 20:07:26 DietPi systemd-networkd[24572]: eth0: Gained IPv6LL
Dec 03 20:07:26 DietPi systemd-networkd[24572]: Enumeration completed
Dec 03 20:07:26 DietPi systemd[1]: Started Network Service.
Dec 03 20:07:26 DietPi systemd-networkd[24572]: br0: netdev exists, using existing without changing its parameters

MichaIng, 2022-12-11:
Did you setup this bridge `br0` manually?

MichaIng, 2022-12-11:
There are no kernel defaults for network bridges or network interfaces at all. These are software setup, not related to hardware devices. And it wasn't there in you first attempts. Not sure if systemd-networkd is somehow responsible, we/I do not use it. Since I'm abroad until January, I'll probably not find time to test this (whether systemd-networkd works as expected on DietPi with above config) until then.

MichaIng, 2023-02-12:
Just tested, and commenting out any `allow-hotplug` and `auto` line in `/etc/network/interfaces`, then
```sh
echo -e '[Match]\nName=eth0\n\n[Network]\nDHCP=ipv4' > /etc/systemd/network/eth0.network
systemctl enable --now systemd-networkd
```
leads to a perfectly accessible and Internet capable system here. I think there is still some conflicting network daemon/service on your system which is responsible for e.g. the bridge and breaks other things as well. Can you show:
```sh
find /etc/network/interfaces.d/
find /etc/systemd/network/
systemctl
```

MichaIng, 2023-03-18:
I'll mark this as closed. Feel free to reopen if the issue persists.

chips023, 2024-07-16:
@MichaIng 

`echo -e '[Match]\nName=eth0\n\n[Network]\nDHCP=ipv4' > /etc/systemd/network/eth0.network
systemctl enable --now systemd-networkd`

hi Michalng

After this setting, the network is normal and the IP address can be obtained

chips023, 2024-07-16:
however
After executing apt updatei, the IP address cannot be obtained again

`
root@DietPi:~# ip addr show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1024
    link/ether f0:ad:4e:08:00:00 brd ff:ff:ff:ff:ff:ff
3: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN group default qlen 1000
    link/ether 22:f0:cf:17:bf:6a brd ff:ff:ff:ff:ff:ff
`

`
root@DietPi:~# cat /etc/systemd/network/eth0.network
[Match]
Name=eth0

[Network]
DHCP=ipv4
`

`root@DietPi:~# cat /etc/network/interfaces
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 9.9.9.9

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 9.9.9.9
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
`

chips023, 2024-07-16:
> Oh that issue is old, forgive me that I do not remember why we used `systemd-networkd` here, probably as alternative to `ifupdown` which caused issues in your case?
> 
> See the commands I suggested above to check for possibly conflicting services or `ifupdown` network configs.

Apologies for taking so long to recall this issue
The current problem is still unable to obtain an IP address

chips023, 2024-07-16:
```
root@DietPi:~# find /etc/network/interfaces.d/
/etc/network/interfaces.d/

root@DietPi:~# find /etc/systemd/network/
/etc/systemd/network/
/etc/systemd/network/10-br0.netdev
/etc/systemd/network/10-lan1.network
/etc/systemd/network/10-br0.network
/etc/systemd/network/99-default.link
/etc/systemd/network/eth0.network
/etc/systemd/network/10-wan.network
/etc/systemd/network/10-lan0.network
/etc/systemd/network/10-eth0.network

root@DietPi:~# systemctl
  UNIT                                                                                                                            LOAD   ACTIVE SUB       DESCRIPTION
  proc-sys-fs-binfmt_misc.automount                                                                                               loaded active waiting   Arbitrary Executable File Formats File System Automount Point
  sys-devices-platform-serial8250-tty-ttyS0.device                                                                                loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS0
  sys-devices-platform-serial8250-tty-ttyS1.device                                                                                loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS1
  sys-devices-platform-serial8250-tty-ttyS2.device                                                                                loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS2
  sys-devices-platform-serial8250-tty-ttyS3.device                                                                                loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS3
  sys-devices-platform-soc-soc:internal\x2dregs\x40d0000000-d0010600.spi-spi_master-spi0-spi0.0-mtd-mtd0-mtdblock0.device         loaded active plugged   /sys/devices/platform/soc/soc:internal-regs@d0000000/d0010600.spi/spi_master/spi0/spi0.0/mtd/mtd0/mtdblock0
  sys-devices-platform-soc-soc:internal\x2dregs\x40d0000000-d0010600.spi-spi_master-spi0-spi0.0-mtd-mtd1-mtdblock1.device         loaded active plugged   /sys/devices/platform/soc/soc:internal-regs@d0000000/d0010600.spi/spi_master/spi0/spi0.0/mtd/mtd1/mtdblock1
  sys-devices-platform-soc-soc:internal\x2dregs\x40d0000000-d0012000.serial-tty-ttyMV0.device                                     loaded active plugged   /sys/devices/platform/soc/soc:internal-regs@d0000000/d0012000.serial/tty/ttyMV0
  sys-devices-platform-soc-soc:internal\x2dregs\x40d0000000-d0030000.ethernet-net-eth0.device                                     loaded active plugged   /sys/devices/platform/soc/soc:internal-regs@d0000000/d0030000.ethernet/net/eth0
  sys-devices-platform-soc-soc:internal\x2dregs\x40d0000000-d00d0000.sdhci-mmc_host-mmc0-mmc0:aaaa-block-mmcblk0-mmcblk0p1.device loaded active plugged   /sys/devices/platform/soc/soc:internal-regs@d0000000/d00d0000.sdhci/mmc_host/mmc0/mmc0:aaaa/block/mmcblk0/mmcblk0p1
  sys-devices-platform-soc-soc:internal\x2dregs\x40d0000000-d00d0000.sdhci-mmc_host-mmc0-mmc0:aaaa-block-mmcblk0.device           loaded active plugged   /sys/devices/platform/soc/soc:internal-regs@d0000000/d00d0000.sdhci/mmc_host/mmc0/mmc0:aaaa/block/mmcblk0
  sys-devices-virtual-block-ram0.device                                                                                           loaded active plugged   /sys/devices/virtual/block/ram0
  sys-devices-virtual-net-br0.device                                                                                              loaded active plugged   /sys/devices/virtual/net/br0
  sys-module-configfs.device                                                                                                      loaded active plugged   /sys/module/configfs
  sys-subsystem-net-devices-br0.device                                                                                            loaded active plugged   /sys/subsystem/net/devices/br0
  sys-subsystem-net-devices-eth0.device                                                                                           loaded active plugged   /sys/subsystem/net/devices/eth0                   
  -.mount                                                                                                                         loaded active mounted   Root Mount
  dev-hugepages.mount                                                                                                             loaded active mounted   Huge Pages File System
  dev-mqueue.mount                                                                                                                loaded active mounted   POSIX Message Queue File System
  sys-kernel-config.mount                                                                                                         loaded active mounted   Kernel Configuration File System
  sys-kernel-debug.mount                                                                                                          loaded active mounted   Kernel Debug File System
  tmp.mount                                                                                                                       loaded active mounted   /tmp
  var-log.mount                                                                                                                   loaded active mounted   /var/log                                          
[... cut ...]
  systemd-tmpfiles-clean.timer                                                                                                    loaded active waiting   Daily Cleanup of Temporary Directories

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
92 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.
```

chips023, 2024-07-16:
root@DietPi:~# cat /etc/network/interfaces
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Ethernet
#allow-hotplug eth0
iface eth0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 9.9.9.9

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 9.9.9.9
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

MichaIng, 2024-07-16:
Your `systemctl` output still shows `ifup@eth0.service`. Probably you did not reboot yet for this change to be effective?

[41 more replies]

closed 2025-05-31 by MichaIng