MichaIng/DietPi #5873: Downloading update archive failed.

SargonofAssyria, 2022-11-12:
#### Details:
- Date           | Sat Nov 12 13:45:11 CET 2022
- DietPi version | v8.2.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux HomeAssistant 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `curl -sSfLO https://github.com/MichaIng/DietPi/archive/master.tar.gz`
- Exit code      | 6
- Software title | DietPi-Update
#### Steps to reproduce:

Updated from old version to newversion

#### Expected behaviour:
Should go ok.
#### Actual behaviour:
Download failed because github.com could not be resolved.
No idea what the impact will be.
#### Additional logs:
```
curl: (6) Could not resolve host: github.com
```

Joulinar, 2022-11-12:
Do you use a lokal DNS server like PiHole or AdGuardHome? Maybe as Docker container?

SargonofAssyria, 2022-11-12:
Yes I do, but also tried with Google DNS. 8.8.8.8
It seems that the DNS settings in dietpi-config does not hold over a reboot.

MichaIng, 2022-11-12:
Please show:
```sh
ls -l /etc/resolv.conf
cat /etc/resolv.conf
```
Probably some other software is unintentionally dynamically managing this file.

SargonofAssyria, 2022-11-12:
After a reboot
ls -l /etc/resolv.conf
```
-rw-r--r-- 1 root root 30 Nov 12 14:05 /etc/resolv.conf
```
cat /etc/resolv.conf
```
# Generated by NetworkManager
```
After I set the DNS in dietpi-config the normal setting is
```
nameserver 8.8.8.8
nameserver 8.8.4.4
```
But these settings disappear after every reboot.
The only thing I am running on this Pi is HomeAssistant.

SargonofAssyria, 2022-11-12:
I have an second 'test' DietPi with version v8.9.2 which works fine in the same network.
This DietPi does not lose DNS after a reboot.
The problematic DietPi is upgraded after a second run to version v.8.10.2 but I do not know what I missed with the
```
curl -sSfLO https://github.com/MichaIng/DietPi/archive/master.tar.gz
```

Joulinar, 2022-11-12:
```
cat /etc/resolv.conf
# Generated by NetworkManager
```

You are using `NetworkManager` in your system. This is conflicting application for DietPi as we use `ifupdown`. I guess you installe HA Supervision?

SargonofAssyria, 2022-11-12:
Yes I did.

Joulinar, 2022-11-12:
> Yes I did.

This means you are not able to use `dietpi-config` for network settings anymore. Have a look to your `NetworkManager` and set correct upstream DNS server.

SargonofAssyria, 2022-11-12:
Ok, will do.

SargonofAssyria, 2022-11-12:
Tnx, for the assistance. Found the right setting in the Supervisor.
Hope that I did not too much from the **master.tar.gz** file.

closed 2022-11-12 by SargonofAssyria

MichaIng, 2022-11-12:
> Hope that I did miss not too much from the **master.tar.gz** file.

What do you mean by that? It should be downloaded and installed, now isn't it?

SargonofAssyria, 2022-11-12:
How can I check if it is installed, because it gave a dns error with the first 'attempt'

MichaIng, 2022-11-12:
If on second attempt the `dietpi-update` went through, then everything got installed. There is nothing skipped on a second attempt, if a first attempt failed somewhere in the middle.

SargonofAssyria, 2022-11-12:
Oke, tnx for the assistance.