MichaIng/DietPi #5876: urbackup CLIENT

kerryland, 2022-11-13:
Given that DietPi already supports the [UrBackup Server](https://dietpi.com/docs/software/cloud/#urbackup) it should also support the client, so that I can actually backup files that are on the dietpi machine too.

The [official client](https://www.urbackup.org/download.html#linux_all_binary) does not run on my 64bit PI4, DietPi 8.8.1, bullseye, Linux blackbox 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

The official installation process:
```
TF=$(mktemp) && wget "https://hndl.urbackup.org/Client/2.5.21/UrBackup%20Client%20Linux%202.5.21.sh" -O $TF && sudo sh $TF; rm -f $TF
```
fails, as follows:
```
--2022-11-14 12:02:01--  https://hndl.urbackup.org/Client/2.5.21/UrBackup%20Client%20Linux%202.5.21.sh
Resolving hndl.urbackup.org (hndl.urbackup.org)... 198.58.118.162, 2600:3c00::f03c:91ff:fec8:ecf6
Connecting to hndl.urbackup.org (hndl.urbackup.org)|198.58.118.162|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 25448161 (24M) [text/x-sh]
Saving to: ‘/tmp/tmp.FzIwzmdEjh’

/tmp/tmp.FzIwzmdEjh                             100%[====================================================================================================>]  24.27M  2.30MB/s    in 14s

2022-11-14 12:02:16 (1.79 MB/s) - ‘/tmp/tmp.FzIwzmdEjh’ saved [25448161/25448161]

Verifying archive integrity... All good.
Uncompressing UrBackup Client Installer for Linux  100%
Installation of UrBackup Client 2.5.20 to /usr/local ... Proceed ? [Y/n]
y
Uncompressing install data...
Detected Debian \(derivative\) system
Detected systemd
Detected architecture aarch64-linux-android
Error running executable on this system \(aarch64\). Stopping installation.
```

Installing from [source](https://www.urbackup.org/client_debian_ubuntu_install.html) seems easy enough, although I cheated and did a headless build because I'm not running any UI and didn't want to deal with any more libraries than I needed to, so YMMV.

MichaIng, 2022-11-14:
I think there is a bug in this installer/binary combi file. Please try the following:
```sh
cd /tmp
curl -f 'https://hndl.urbackup.org/Client/2.5.21/UrBackup%20Client%20Linux%202.5.21.sh' -o installer.sh
sed -i '0,\|^install -c "\$TARGET/urbackupclientbackend" "\$PREFIX/sbin"$|s|^install -c "\$TARGET/urbackupclientbackend" "\$PREFIX/sbin"$|install -c "$TARGET/urbackupclientctl" "$PREFIX/bin"|' installer.sh
```
It has this code block:
```sh
test -e "$PREFIX/sbin" || install -c -m 755 -d "$PREFIX/sbin"
test -e "$PREFIX/bin" || install -c -m 755 -d "$PREFIX/bin"
install -c "$TARGET/urbackupclientbackend" "$PREFIX/sbin"
install -c "$TARGET/urbackupclientbackend" "$PREFIX/sbin"
```
Hence installs `urbackupclientbackend` while missing `urbackupclientctl`, which is tried to be executed. In case of `x86_64` and 32-bit ARM, in case of this failure, it installs the Android binary, which should work on Linux as well, but on `aarch64` and `i386` it is doomed to fail, since no Android fallback is in place. The above `sed` replaces the first match of the doubled `urbackupclientbackend` with `urbackupclientctl`, so it looks like:
```sh
test -e "$PREFIX/sbin" || install -c -m 755 -d "$PREFIX/sbin"
test -e "$PREFIX/bin" || install -c -m 755 -d "$PREFIX/bin"
install -c "$TARGET/urbackupclientctl" "$PREFIX/bin"
install -c "$TARGET/urbackupclientbackend" "$PREFIX/sbin"
```
which should then work. To be compared with the fallback block `[ $TARGET != $ORIG_TARGET ]` some lines below.

MichaIng, 2022-11-14:
Interesting, here it is correct: https://github.com/uroni/urbackup_backend/blob/b9d04e3/install_client_linux.sh#L150-L153
Since they don't have issues enabled on their repo, not sure how to report this divergence between source code and downloadable installer.

Do you have a UrBackup forum account? If so, you may want to report this bug + solution there: https://forums.urbackup.org/

kerryland, 2022-11-14:
I'll sign up and let them know. Thanks for your excellent work.

MichaIng, 2022-11-19:
Does it actually work well with the fix?

kerryland, 2022-11-19:
I didn't actually test it as I had already built it from source.

The problem is differently related to a recent patch, which they have promised to fix properly :-)

Here's where I reported the issue: https://forums.urbackup.org/t/urbackclient-not-installing-on-aarch64-and-i386-with-fix/12002

kerryland, 2023-05-25:
This has since been fixed upstream. Nothing for dietpi to do :-)

MichaIng, 2023-05-27:
Let's keep this open, since having the UrBackup client in our software catalogue still makes sense 🙂.

myB0B, 2025-02-03:
hi, any update ? same problem on a rpi5

Joulinar, 2025-02-03:
As you can see above, this is a problem with the installation program outside of DietPi. As far as I can see, you posted already within Urbackup forum, which would be correct place.

myB0B, 2025-02-04:
oups...