MichaIng/DietPi #5877: Dietpi seems not to support creating virtual wireless interfaces with iw dev interface add

louigi600, 2022-11-14:
### Creating a feature request
I have a  RPi4 that is capable of  interface combinations, and I have used them on RPiOS, can't do the same on dietpi.

#### Is your feature request related to a problem? Please describe:
not sure if this is related to some firmware,kernel compile time options or iw compile time options but I'm guessing it needs to be one of these as I can't think of any other reason why it will not work on dietpi.

#### Describe the solution you'd like:
I'd like to be able to be able to do this on my RPi4 running dietpi
iw wlan0 interface add vap1 type <some supported type like managed or __ap>

MichaIng, 2022-11-14:
Many thanks for your report.

This should work, kernel and package repository are identical to RPi OS. Did you use our ARMv6, ARMv7 or ARMv8 image?

louigi600, 2022-11-14:
I get this error:
  root@wrap:~# iw phy0 info | grep 'valid interface combinations:' -A 4
        valid interface combinations:
                 * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
                   total <= 3, #channels <= 2
                 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
                   total <= 4, #channels <= 1
  root@wrap:~# ip link
  1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
  2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether dc:a6:32:e3:68:fd brd ff:ff:ff:ff:ff:ff
  3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 1000
    link/ether dc:a6:32:e3:68:fe brd ff:ff:ff:ff:ff:ff
  5: wlan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DORMANT group default qlen 1000
    link/ether 00:0c:f6:86:8c:50 brd ff:ff:ff:ff:ff:ff
  root@wrap:~# iw phy phy0 interface add vw1 type managed
  command failed: Operation not supported (-95)
  root@wrap:~#`

It does not matter if the device is in use you should be able to do it anyway.
I can do it on the wlan1 that is the one I'm using to remotely access the RPi4:

  root@wrap:~# iw phy phy1 interface add vw2 type managed
  root@wrap:~# ip link
  1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
  2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether dc:a6:32:e3:68:fd brd ff:ff:ff:ff:ff:ff
  3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 1000
    link/ether dc:a6:32:e3:68:fe brd ff:ff:ff:ff:ff:ff
  5: wlan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DORMANT group default qlen 1000
    link/ether 00:0c:f6:86:8c:50 brd ff:ff:ff:ff:ff:ff
  7: vw2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:0c:f6:86:8c:51 brd ff:ff:ff:ff:ff:ff
  root@wrap:~#

It is the 64 bit version so I have the ARMv8 image
root@wrap:~# uname -a
  Linux wrap 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
  root@wrap:~# cat /etc/debian_version
  11.3
  root@wrap:~#

There must be some difference between Dietpi and RPIos because on the latter it works normally also on the integrated wifi.

MichaIng, 2022-11-14:
Did you test in on Raspberry Pi OS 64-bit? And did you
```sh
apt update
apt full-upgrade
```
on RPi OS, before testing it, so that `uname -a` and `cat /etc/debian_version` are identical to the output on DietPi?

Also please compare the `ip l` and `ip a` states of the interface. I've not much experience with VLANs, so not sure whether they need to be up, have an IP assigned or any such.

It cannot make a difference, but on DietPi, we use the classic interface names `ethX`/`wlanX` while on RPi OS the "predictable" ones are used.

louigi600, 2022-11-15:
I tested on PIOS 64 not version but not with the latest updates ... I will try that later.
It does not matter if the interfaces remain with the kernel assigned names or they get the predictable but totally crazy names ... it should work non the less.
Are the firmwares identical too?

Joulinar, 2022-11-15:
Our images use the original firmware as wel as kernel. We don't do own development on this area.

louigi600, 2022-11-15:
well this is odd.
both the dietpi and the pios are not updated with the latest packages but are the same version (11.3) but likely very similar in age as for the updates.
I'm currently doing an dapt update , apt upgrade.
I will then check if anything brakes on pios.

louigi600, 2022-11-15:
Interesting: with the latest updates on pios I get the same erratic behavior.
RPi foundation broke it with some updates.
I forgot to take note of what was updated ... but I'm betting on firmware.

Joulinar, 2022-11-15:
Best would be to report this issue to RPi developer https://github.com/raspberrypi/firmware/issues

louigi600, 2022-11-15:
It could be something specific to the device driver in the kernel.

louigi600, 2022-11-17:
I had a look at the packages that ware installed/upgraded when this feature got broken:
I I found some intresting information there from Nov 15'th 
davide@raspberrypi:/tmp $ grep -E "kernel|firmware|wireless" history_Nov15.txt
 firmware-brcm80211:arm64 (1:20210315-3+rpt5
 firmware-atheros:arm64 (1:20210315-3+rpt5
 wireless-regdb:arm64 (2020.04.29-2
 firmware-realtek:arm64 (1:20210315-3+rpt5
 firmware-libertas:arm64 (1:20210315-3+rpt5
 bluez-firmware:arm64 (1.2-4+rpt8
 firmware-misc-nonfree:arm64 (1:20210315-3+rpt5
 raspberrypi-kernel:arm64 (1:1.20220331-1
davide@raspberrypi:/tmp $

So both the kernel and the onboard wifi firmware were updated just prior to this feature braking on pios.

Joulinar, 2022-11-17:
Theoretically you could try to downgrade kernel to an older version

louigi600, 2022-11-17:
For that I'd need to know either the target release I want to go to or the full target package name and version, neither of which I'm certain about.
I'll see if I can figure it out.
Anyway for the moment I'm more inclined to think it is a firmware issue.
If you guys have not filed a bug report for pios where would be the right place for it ?

MichaIng, 2022-11-17:
What you posted are the old package versions (since newer ones are available), right? So you have your target version there 🙂. Let's try to downgrade the WiFi firmware first, since it is much simpler:
```sh
cd /tmp
curl -f 'https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_20210315-3+rpt5_all.deb' -o package.deb
dpkg -i package.deb
rm package.deb
# reboot required, or manually removing/readding the WiFi kernel module
```

louigi600, 2022-11-17:
Yes I had discovered that but I was unable to download the old package:
root@raspberrypi:/var/log/apt# apt-cache madison  firmware-brcm80211
firmware-brcm80211 | 1:20210315-3+rpt7 | http://archive.raspberrypi.org/debian bullseye/main arm64 Packages
firmware-brcm80211 | 1:20210315-3+rpt7 | http://archive.raspberrypi.org/debian bullseye/main armhf Packages
firmware-brcm80211 | 20210315-3 | http://deb.debian.org/debian bullseye/non-free arm64 Packages
firmware-brcm80211 | 20210315-3 | http://deb.debian.org/debian bullseye/non-free armhf Packages
root@raspberrypi:/var/log/apt#

Not sure how you found it ... but thanks for the hint. I'm proceeding with upgrade from what I was able to get :
firmware-brcm80211 | 20210315-3
to 
firmware-brcm80211_20210315-3+rpt5_all.deb
I will update asap

louigi600, 2022-11-17:
Seems like the firmware-brcm80211 downgrade did not solve the issue. 
Will try to downgrade kernel.

MichaIng, 2022-11-17:
The kernel packages can be found here (at best downgrade `raspberrypi-kernel` together with `raspberrypi-bootloader`): https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/?C=M;O=D
Also `rpi-update` (package needs to be installed first) can be used to switch to a very specific kernel/firmware commit,

louigi600, 2022-11-17:
downgraded also the kernel but its still not fixed.
I give up trying to pinpoint what it is ... the update was quite bib because I hardly ever use the uSD with pios on it Just sort of 250 packages.

Start-Date: 2022-11-15  08:03:13
Commandline: apt upgrade
Install: libnma-common:arm64 (1.8.30-1+rpt1, automatic
 libnma0:arm64 (1.8.30-1+rpt1, automatic
 lxplug-netman:arm64 (0.4, automatic
 lxplug-menu:arm64 (0.3, automatic
 dos2unix:arm64 (7.4.1-1, automatic
 libnm0:arm64 (1.30.6-1+deb11u1, automatic)
Upgrade: dpkg:arm64 (1.20.10, 1.20.12
 libntfs-3g883:arm64 (1:2017.3.23AR.3-4+deb11u1, 1:2017.3.23AR.3-4+deb11u3
 realvnc-vnc-server:arm64 (6.9.1.46706, 6.10.1.47571
 cifs-utils:arm64 (2:6.11-3.1, 2:6.11-3.1+deb11u1
 vlc-plugin-qt:arm64 (3.0.16-1+rpi1+rpt2, 3.0.17.4-0+deb11u1+rpi1+rpt4
 libpoppler-qt5-1:arm64 (20.09.0-3.1, 20.09.0-3.1+deb11u1
 libvlc5:arm64 (3.0.16-1+rpi1+rpt2, 3.0.17.4-0+deb11u1+rpi1+rpt4
 libcurl4:arm64 (7.74.0-1.3+deb11u1, 7.74.0-1.3+deb11u3
 libpulse0:arm64 (14.2-2+rpi1, 14.2-2+rpt1
 libpulse0:armhf (14.2-2+rpi1, 14.2-2+rpt1
 lxplug-ejecter:arm64 (0.18, 0.20
 console-setup:arm64 (1.205, 1.205+rpt1
 libpam-runtime:arm64 (1.4.0-9+deb11u1+rpt1, 1.4.0-9+deb11u1+rpt2
 udev:arm64 (247.3-7, 247.3-7+deb11u1
 krb5-locales:arm64 (1.18.3-6+deb11u1, 1.18.3-6+deb11u2
 libsnmp-base:arm64 (5.9+dfsg-3, 5.9+dfsg-4+deb11u1
 bind9-host:arm64 (1:9.16.27-1~deb11u1, 1:9.16.33-1~deb11u1
 raspberrypi-bootloader:arm64 (1:1.20220331-1, 1:1.20221104-1
 libgssapi-krb5-2:arm64 (1.18.3-6+deb11u1, 1.18.3-6+deb11u2
 libgssapi-krb5-2:armhf (1.18.3-6+deb11u1, 1.18.3-6+deb11u2
 libcurl3-gnutls:arm64 (7.74.0-1.3+deb11u1, 7.74.0-1.3+deb11u3
 liblivemedia77:arm64 (2020.01.19-1, 2020.01.19-1+build1
 openssh-client:arm64 (1:8.4p1-5, 1:8.4p1-5+deb11u1
 libgdk-pixbuf2.0-bin:arm64 (2.42.2+dfsg-1, 2.42.2+dfsg-1+deb11u1
 lxplug-volumepulse:arm64 (0.12, 0.14
 vlc-data:arm64 (3.0.16-1+rpi1+rpt2, 3.0.17.4-0+deb11u1+rpi1+rpt4
 libfreetype6-dev:arm64 (2.10.4+dfsg-1, 2.10.4+dfsg-1+deb11u1
 dbus-user-session:arm64 (1.12.20-2, 1.12.24-0+deb11u1
 libsnmp40:arm64 (5.9+dfsg-3+b1, 5.9+dfsg-4+deb11u1
 lxpanel-data:arm64 (0.10.1-2+rpt15, 0.10.1-2+rpt19
 firmware-brcm80211:arm64 (1:20210315-3+rpt5, 1:20210315-3+rpt7
 gpg:arm64 (2.2.27-2+deb11u1, 2.2.27-2+deb11u2
 systemd-timesyncd:arm64 (247.3-7, 247.3-7+deb11u1
 rpi-eeprom:arm64 (13.14-1, 13.21-1
 libtirpc-common:arm64 (1.3.1-1, 1.3.1-1+deb11u1
 tzdata:arm64 (2021a-1+deb11u3, 2021a-1+deb11u8
 chromium-codecs-ffmpeg-extra:armhf (101.0.4951.57-rpt2, 104.0.5112.105-rpt2
 libldb2:arm64 (2:2.2.3-2~deb11u1, 2:2.2.3-2~deb11u2
 ntfs-3g:arm64 (1:2017.3.23AR.3-4+deb11u1, 1:2017.3.23AR.3-4+deb11u3
 userconf-pi:arm64 (0.2, 0.5
 xserver-xorg-core:arm64 (2:1.20.11-1+rpt1+deb11u1, 2:1.20.11-1+rpt1+deb11u3
 libpam-systemd:arm64 (247.3-7, 247.3-7+deb11u1
 libpulsedsp:arm64 (14.2-2+rpi1, 14.2-2+rpt1
 libmutter-7-0:arm64 (1:3.38.6-2~deb11u1+rpt4, 1:3.38.6-2~deb11u2+rpt6
 libvlccore9:arm64 (3.0.16-1+rpi1+rpt2, 3.0.17.4-0+deb11u1+rpi1+rpt4
 libusageenvironment3:arm64 (2020.01.19-1, 2020.01.19-1+build1
 libpam-chksshpwd:arm64 (1.4.0-9+deb11u1+rpt1, 1.4.0-9+deb11u1+rpt2
 usb.ids:arm64 (2022.02.15-0+deb11u1, 2022.05.20-0+deb11u1
 ssh:arm64 (1:8.4p1-5, 1:8.4p1-5+deb11u1
 pulseaudio:arm64 (14.2-2+rpi1, 14.2-2+rpt1
 libavahi-common-data:arm64 (0.8-5, 0.8-5+deb11u1
 libavahi-common-data:armhf (0.8-5, 0.8-5+deb11u1
 vlc:arm64 (3.0.16-1+rpi1+rpt2, 3.0.17.4-0+deb11u1+rpi1+rpt4
 raspi-config:arm64 (20220506, 20221018
 libexpat1:arm64 (2.2.10-2+deb11u3, 2.2.10-2+deb11u5
 libexpat1:armhf (2.2.10-2+deb11u3, 2.2.10-2+deb11u5
 gir1.2-gdkpixbuf-2.0:arm64 (2.42.2+dfsg-1, 2.42.2+dfsg-1+deb11u1
 libavahi-core7:arm64 (0.8-5, 0.8-5+deb11u1
 vlc-bin:arm64 (3.0.16-1+rpi1+rpt2, 3.0.17.4-0+deb11u1+rpi1+rpt4
[... 159 lines cut ...]
 gpg-wks-client:arm64 (2.2.27-2+deb11u1, 2.2.27-2+deb11u2
 raspberrypi-kernel:arm64 (1:1.20220331-1, 1:1.20221104-1
 zlib1g:arm64 (1:1.2.11.dfsg-2+deb11u1, 1:1.2.11.dfsg-2+deb11u2
 zlib1g:armhf (1:1.2.11.dfsg-2+deb11u1, 1:1.2.11.dfsg-2+deb11u2
 libavresample4:arm64 (7:4.3.4-0+deb11u1+rpt1, 7:4.3.5-0+deb11u1+rpt2
 firefox-esr:arm64 (91.9.1esr-1~deb11u1, 102.4.0esr-1~deb11u1
 libtirpc3:arm64 (1.3.1-1, 1.3.1-1+deb11u1
 libtirpc3:armhf (1.3.1-1, 1.3.1-1+deb11u1
dbus:arm64 (1.12.20-2, 1.12.24-0+deb11u1
 cmake:arm64 (3.18.4-2+rpt1+rpi1, 3.18.4-2+rpt1+rpi1+deb11u1
 libavahi-client3:arm64 (0.8-5, 0.8-5+deb11u1
 libavahi-client3:armhf (0.8-5, 0.8-5+deb11u1
 gpgconf:arm64 (2.2.27-2+deb11u1, 2.2.27-2+deb11u2
 libtirpc-dev:arm64 (1.3.1-1, 1.3.1-1+deb11u1
 raspberrypi-sys-mods:arm64 (20220224, 20221019
 libvlc-bin:arm64 (3.0.16-1+rpi1+rpt2, 3.0.17.4-0+deb11u1+rpi1+rpt4
 chromium-browser-l10n:arm64 (101.0.4951.57-rpt2, 104.0.5112.105-rpt2
 alacarte:arm64 (3.36.0-1+rpt1, 3.36.0-1+rpt2
 openssl:arm64 (1.1.1n-0+deb11u2+rpt1, 1.1.1n-0+deb11u3+rpt1
 vlc-plugin-base:arm64 (3.0.16-1+rpi1+rpt2, 3.0.17.4-0+deb11u1+rpi1+rpt4
 vlc-plugin-visualization:arm64 (3.0.16-1+rpi1+rpt2, 3.0.17.4-0+deb11u1+rpi1+rpt4
 linux-libc-dev:arm64 (1:1.20220331-1, 1:1.20221104-1
 libavfilter7:arm64 (7:4.3.4-0+deb11u1+rpt1, 7:4.3.5-0+deb11u1+rpt2
 libgdk-pixbuf2.0-common:arm64 (2.42.2+dfsg-1, 2.42.2+dfsg-1+deb11u1)
End-Date: 2022-11-15  08:11:06

MichaIng, 2022-11-17:
Uff, would have been easier to test with RPi OS Lite to skip all the desktop/GUI stuff 😄. Apart of WiFi firmware and kernel/bootloader, I cannot thing of something else which could have broken it, but with this large list of upgraded packages, difficult to identify possibly related ones and test.

Just to rule it out: Before upgrading the packages on RPi OS, but after a reboot, VLANs still work? Just in case some first run setup breaks it.

louigi600, 2022-11-17:
VLAN as in lan tagging I don't use. 
What boroke was the creation of virtual WLAN devices .... kind of like ethernet NIC alias.
That broke after the update ... did not try it before rebooting: I updated rebooted and then tried.
I was not expecting it to brake ... I was hoping it was something limited to dietpi because if it worked on pios it would have been easy for you guys to fix.
Now we are in one of the worst possible situations: we cant pinpoint what broke, it is a feature that is not commonly used so the pi foundation is likely no not give a crap about it as long as the common features are working.
I may have some old lite images to play with not sure if the last one I downloaded will be already broken.
I will check later.

If the dietpi crew have not filed a bug report for pios where would be the right place for it ?

Joulinar, 2022-11-17:
See above https://github.com/MichaIng/DietPi/issues/5877#issuecomment-1315315285

louigi600, 2022-11-17:
But it may not be firmware ... 
I think I need to try and better pinpoint what caused it to brake first.

MichaIng, 2022-11-17:
> What boroke was the creation of virtual WLAN devices .... kind of like ethernet NIC alias.

Thanks for clarification. I didn't ever use any of these features myself.

> we cant pinpoint what broke

As said, flash the RPi OS **Lite** image, assure it works, then reboot, assure it still works, then `apt update` and see which package upgrades are available. Start with firmware upgrade, then kernel upgrade, and always do reboot, then check whether it still works. If it does, based on the remaining smaller list of available package upgrades, we should be able to identify then culprit.

Yes, reporting to RPi devs is an option, but more efficient if we can narrow it down first.

louigi600, 2022-11-17:
I'm doing that hoping the 2022-09-22-raspios-bullseye-arm64-lite will still work.

MichaIng, 2022-11-17:
In attempt to replicate the issue on RPi Zero W, I get:
```console
# iw phy phy0 interface add vw2 type managed
command failed: Device or resource busy (-16)
```
Do you get this as well, if you use the adapter for already, or do you see
```console
command failed: Operation not supported (-95)
```
directly? Otherwise I need to attach HDMI and keyboard to be able to replicate.

louigi600, 2022-11-17:
first let me try get it working again on a Pi
Right now I may not have any debian based image where it works on the internal wlan card

MichaIng, 2022-11-17:
Ah, it was such an older image where it worked, okay that explains the large number of available package upgrades.

louigi600, 2022-11-17:
Actually I will go for a lite version:
2022-01-28-raspios-bullseye-arm64-lite.zip
https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-01-28/2022-01-28-raspios-bullseye-arm64-lite.zip

louigi600, 2022-11-17:
@MichaIng : maybe if you try https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz
while I try the January one

louigi600, 2022-11-17:
bad news ... even the Jan image is not working
I'm not sure how old the image I has was ... but it worked  ... maybe the 2019 image was the one on which it worked

louigi600, 2022-11-17:
what the hell: I went back to 2020 08 and still could not get it to work.
I have no idea what I had when I managed to get it to work ... I'm even beginning to doubt myself it ever worked on the RPi4 onboard wlan.

MichaIng, 2022-11-17:
Strange, on my RPi Zero W `command failed: Device or resource busy (-16)` regardless whether the WiFi interface/connection is up or now, even if I disable it completely and reboot (so that it is not touched at all) 🤔.

MichaIng, 2024-01-22:
Do you still face this issue?

MichaIng, 2026-11-07:
I assume this got solved by recent kernel packages in the meantime.

closed 2026-11-07 by MichaIng