MichaIng/DietPi #5878: Installing Home Assistant fails with "could not build wheels for cryptography"

68040, 2022-11-14:
#### Details:
- Date           | Mon Nov 14 13:47:10 CET 2022
- DietPi version | v8.10.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux homebot 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `sudo -u homeassistant dash -c 
. /home/homeassistant/pyenv-activate.sh
pyenv install 3.9.15
pyenv local 3.9.15
pip3 install --no-cache-dir -U pip setuptools wheel
[ -z '' ] || pip3 install --no-cache-dir 
exec pip3 install --no-cache-dir homeassistant`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Downloading Python-3.9.15.tar.gz...
-> https://www.python.org/ftp/python/3.9.15/Python-3.9.15.tgz
Installing Python-3.9.15...
WARNING: The Python lzma extension was not compiled. Missing the lzma lib?
Installed Python-3.9.15 to /home/homeassistant/.pyenv/versions/3.9.15

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Requirement already satisfied: pip in ./.pyenv/versions/3.9.15/lib/python3.9/site-packages (22.0.4)
Collecting pip
  Downloading pip-22.3.1-py3-none-any.whl (2.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 5.8 MB/s eta 0:00:00
Requirement already satisfied: setuptools in ./.pyenv/versions/3.9.15/lib/python3.9/site-packages (58.1.0)
Collecting setuptools
  Downloading https://www.piwheels.org/simple/setuptools/setuptools-65.5.1-py3-none-any.whl (1.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 3.4 MB/s eta 0:00:00
Collecting wheel
  Downloading https://www.piwheels.org/simple/wheel/wheel-0.38.4-py3-none-any.whl (36 kB)
Installing collected packages: wheel, setuptools, pip
  Attempting uninstall: setuptools
    Found existing installation: setuptools 58.1.0
    Uninstalling setuptools-58.1.0:
      Successfully uninstalled setuptools-58.1.0
  Attempting uninstall: pip
    Found existing installation: pip 22.0.4
    Uninstalling pip-22.0.4:
      Successfully uninstalled pip-22.0.4
Successfully installed pip-22.3.1 setuptools-65.5.1 wheel-0.38.4
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Collecting homeassistant
  Downloading https://www.piwheels.org/simple/homeassistant/homeassistant-2022.11.2-py3-none-any.whl (20.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 20.1/20.1 MB 8.5 MB/s eta 0:00:00
Collecting jinja2==3.1.2
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.1/133.1 kB 1.3 MB/s eta 0:00:00
Collecting pyyaml==6.0
  Downloading https://www.piwheels.org/simple/pyyaml/PyYAML-6.0-cp39-cp39-linux_armv7l.whl (45 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 45.3/45.3 kB 9.3 MB/s eta 0:00:00
Collecting voluptuous==0.13.1
[... 291 lines cut ...]
          setuptools: 65.5.1
          setuptools_rust: 1.5.2
          rustc: n/a
          =============================DEBUG ASSISTANCE=============================
      
      error: can't find Rust compiler
      
      If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
      
      To update pip, run:
      
          pip install --upgrade pip
      
      and then retry package installation.
      
      If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
      
      This package requires Rust >=1.48.0.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects
```

Joulinar, 2022-11-15:
as an alternative, install Rust beforehand https://dietpi.com/forum/t/unable-to-upgrade-home-assistant-to-2021-10-x/5823/4

MichaIng, 2023-02-04:
Ah, solved in the meantime with v8.12 since Rust is now installed on all ARMv6/7 systems. piwheels cannot be used anymore for HA since it shops Python 3.9 wheels while HA now uses/requires Python 3.10.

closed 2023-02-04 by MichaIng

Joulinar, 2023-04-11:
You would need to update to latest DietPi version first.

josh3003, 2023-04-11:
> You would need to update to latest DietPi version first.

Am I not already on the latest version? How do I do that then?

MichaIng, 2023-04-11:
> Running Dietpi v8.12.2

Latest version is 8.16.2:
```sh
dietpi-update
```

josh3003, 2023-04-11:
I typo'd
I am running DietPi v8.16.2

Joulinar, 2023-04-12:
pls open a new issue for your error and fill the bug report with relevant data. Pls add the full install log.

josh3003, 2023-04-12:
> 

Ok I have created #6306