MichaIng/DietPi #5879: How to set IP Adress of wlan0 while using WiFi Hotspot?

DrCWO, 2022-11-14:
### Creating a bug report/issue

#### Required Information
- DietPi version |
-G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | 
bullseye 0

- Kernel version | 
Linux dietPi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

- SBC model | 
RPi 4 Model B (aarch64)

- Power supply used | 15W
- SD card used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | WiFi Hotspot
- fresh
- yes

#### Steps to reproduce

#### Expected behaviour
I can set a static IP address for wlan0 in /etc/network/interfaces
But the file tells me to use dietpi-config to set it.
**I can't find where to set it.** 

#### Actual behaviour

#### Extra details
I want to implement a feature that first runs dietpi as a hotspot.
In the mode the user can connect via a http interface I offer via script.
In this page the user shall be able to enter the AP in his network where the Pi should connect to
After connection the Pi shall reboot and connect to this AP instead of offering an Ap by itself.

**Question: How to disable WiFi Hotspot and reenable mormal WiFi mode so it I can connect via /etc/wpa_supplicant/wpa_supplicant.conf settings?**

MichaIng, 2022-11-14:
You can change the IP in `/etc/network/interfaces` and for the DHCP server in `/etc/dhcp/dhcpd.conf`. Also you need to change the related NAT iptables rule, like:
```sh
sed -i 's/192\.168\.42/192.168.123/g' /etc/network/interfaces /etc/dhcp/dhcpd.conf
iptables -t nat -D POSTROUTING -s 192.168.42.0/24 -o eth0 -j MASQUERADE
iptables -t nat -A POSTROUTING -s 192.168.123.0/24 -o eth0 -j MASQUERADE
iptables-save > /etc/iptables.ipv4.nat
reboot
```
Indeed a way to set that IP on WiFi Hotspot install or via `dietpi-config` is currently missing.

> How to disable WiFi Hotspot and reenable mormal WiFi mode so it I can connect via /etc/wpa_supplicant/wpa_supplicant.conf settings?

```sh
dietpi-software uninstall 60
dietpi-config 8 1 # to setup regular WiFi
```

Joulinar, 2022-11-15:
You could have a look to our install and uninstall steps we do

https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L8362
https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L14601

Maybe this can give an idea

closed 2025-07-18 by MichaIng