MichaIng/DietPi #5880: dietpi-update fails if service systemd-timesyncd is not available

jtmoon79, 2022-11-14:
### Creating a bug report/issue

#### Required Information
- DietPi version |
```text
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `Raspbian GNU/Linux 11 (bullseye)`
- Kernel version | `Linux host1 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux`
- SBC model | `RPi 3 Model B+ (armv7l)`

#### Additional Information (if applicable)
- Software title | `dietpi-update`
- Was the software title installed freshly or updated/migrated? The system had been upgraded from Debian 10
- Can this issue be replicated on a fresh installation of DietPi? Have not attempted.
- Bug report ID | `c563c9b4-0d5d-4eee-b436-00099d558ef1`

#### Steps to reproduce

1. using `apt` uninstall `systemd-timesyncd`
2. using `apt` install `ntp`
3. run `dietpi-update`

#### Expected behaviour
No errors during an update.

#### Actual behaviour
`dietpi-update` fails during time update
```text
[FAILED] DietPi-TimeSync | systemctl start systemd-timesyncd
nl80211 not found.
... *dietpi-bugreport occurred* ...
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found.
...
[FAILED] DietPi-TimeSync | Unable to continue, DietPi-TimeSync will now terminate.
```

#### Extra details
- This system was upgraded from Debian 10. After upgrade, the datetime of the system was not updated. So I uninstalled `systemd-timesyncd` and installed `ntp`. The system had been restarted before this error of `dietpi-update`.
- The process `dietpi-update` did successfuly resume after the error dialog was complete.

MichaIng, 2022-11-14:
This is expected: DietPi scripts expect `systemd-timesyncd` by default. But you can disable it in `dietpi-config 3` switching time sync mode to `0 : Custom`.

jtmoon79, 2022-11-14:
> This is expected: DietPi scripts expect `systemd-timesyncd` by default. But you can disable it in `dietpi-config 3` switching time sync mode to `0 : Custom`.

This worked.
I changed the configuration as recommended. Then I ran `dietpi-update`. It printed
```text
[ INFO ] DietPi-TimeSync | Manual mode detected, skipping time sync
```

Thanks!

closed 2022-11-14 by jtmoon79