MichaIng/DietPi #5881: Docker rootless

x0tester0x, 2022-11-14:
Can you please add the option to install Docker also in rootless mode?
[https://docs.docker.com/engine/security/rootless/](https://docs.docker.com/engine/security/rootless/)

MichaIng, 2022-11-15:
Many thanks for your suggestion. Isn't this much more than simply executing the Docker engine as respective user?

... ah I see some more things required:
- As `root`:
```sh
dietpi-software install 162
systemctl unmask systemd-logind
apt install docker-ce-rootless-extras uidmap slirp4netns
systemctl start dbus systemd-logind
```
- As unprivileged user:
```sh
dockerd-rootless-setuptool.sh install
systemctl --user enable --now docker
docker run hello-world
```
- And to disable the system Docker daemon, again as `root`:
```sh
systemctl disable --now docker
G_CONFIG_INJECT '- docker' '- docker' /boot/dietpi/.dietpi-services_include_exclude
```

Probably as `dietpi.txt` and/or interactive choice to install for one or more selected unprivileged login users and to optionally disable/deselect the system/root Docker service?

x0tester0x, 2022-11-15:
In my opinion, an interactive choice would be better.