MichaIng/DietPi #5882: no access to dns9.quad9.net

bluer1sing, 2022-11-15:
#### Details:
- Date           | Tue Nov 15 10:44:01 CET 2022
- DietPi version | v8.10.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux dpi4cinema 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `getent hosts dns9.quad9.net`
- Exit code      | 2
- Software title | DietPi-Software
#### Steps to reproduce:

1. Software installation through dietpi-software
2. ...
#### Expected behaviour:

- installation of Samba
#### Actual behaviour:

- nothing, installation aborts
#### Extra details:

- ...
#### Additional logs:
```

```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

Joulinar, 2022-11-15:
You are using a local DNS Server like PiHole or AdGuardHome? Maybe running on Docker?

kgc35, 2022-11-15:
I have a similar issue #5884 
Also Quad9

Joulinar, 2022-11-15:
its a completely different issue.

closed 2023-02-12 by MichaIng