MichaIng/DietPi #5883: DietPi-Software | FreshRSS: version is stuck on 1.19.0-dev

pedrom34, 2022-11-15:
#### Required Information
- DietPi version | 8.10.2
- Distro version | bullseye
- Kernel version | 5.15.74-rockship64
- SBC model | Rock64

#### Additional Information (if applicable)
- Software title | FreshRSS
- Was the software title installed freshly or updated/migrated? No.
- Can this issue be replicated on a fresh installation of DietPi? I guess, but I didn't checked 😕 

#### Steps to reproduce
1. Install FreshRSS
2. Try to update FreshRSS to latest version.

#### Expected behaviour
- Version 1.20.1 (out November 14th) should be detected and installed.

#### Actual behaviour
- Version 1.19.0-dev is installed and refuse to update as 1.19.0-dev is considered latest (which is not).
[image] 
(Green text reads "Ugh! No update to apply")

#### Extra details
- Force reinstall doesn't resolve the issue.
- Related to #5767 ?

MichaIng, 2022-11-15:
It seems the auto-updater does not work with a `dev` version: https://update.freshrss.org/?v=1.19.0-dev
Please try this:
```sh
sed -i 's/1\.19\.0-dev/1.19.0/' /opt/FreshRSS/constants.php
systenctl restart php7.4-fpm
```
It should then offer the update to v1.20.0: https://update.freshrss.org/?v=1.19.0
v1.20.1 seems to be not yet implemented into the online updater.

pedrom34, 2022-11-15:
Yes, that was it...
Sorry about that. Your command resolved my issue. Thanks a lot!

closed 2022-11-15 by pedrom34

MichaIng, 2022-11-15:
I guess it was our issue since we shipped archives from `master` branch until DietPi v8.10 instead of (stable) releases. I just became aware of stable releases when the `master` branch got removed and now `edge` branch is used instead, self-indicating that it's not stable 😉. Great that the solution is simple.