MichaIng/DietPi #5884: dietpi-update not working (Command: ping -6nc 1 -W 10 2620:fe::fe fails)

kgc35, 2022-11-15:
I can't run dietpi-update, since the Command: ping -6nc 1 -W 10 2620:fe::fe fails. 
This is Quad9 I believe. 

I updated my bug report
Reference code: dada73e3-3713-4609-8d0e-6e909a0ae99a
#### Details:
- Date           | Tue Nov 15 15:21:46 EST 2022
- DietPi version | v8.9.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi B (armv6l) (ID=1)
- Kernel version | `Linux DietPi 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `ping -6nc 1 -W 10 2620:fe::fe`
- Exit code      | 1
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
PING 2620:fe::fe(2620:fe::fe) 56 data bytes

--- 2620:fe::fe ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```

Joulinar, 2022-11-15:
pls can you try to disable IPv6 from error menue.

kgc35, 2022-11-15:
Yes, if I disable IPV6, then dietpi-update is working. 
However in dietpi-settings, IPV6 test is passing. 

So it looks like the IPV6 test in dietpi-update sequence is having an error.

Joulinar, 2022-11-16:
You mean an IPv6 test in `dietpi-config` network settings? Well we don't have such a dedicated test there.

MichaIng, 2022-11-16:
When you enable IPv6, is this the only host you cannot ping, or is it true for every IPv6 connection? 
```sh
ping 2620:fe::fe
ping -6 dietpi.com
```

kgc35, 2022-11-16:
[image]
Yes here. 

Good point about that,` ping -6 dietpi.com` also doesn't work. 
What is the resolution for this? I am using opnsense as my router.
Not sure if this is a dietpi or opnsense issue.

Joulinar, 2022-11-16:
Your network seems not correctly configured to support IPv6. Looks like a local issue inside your network/router config.

kgc35, 2022-11-16:
[image]
Thanks, when I do this ping from OPNsense interface, it looks like it works. 
But it isn't working from dietpi device or kubuntu desktop. 
Sounds like I will ask on OPNsense forum.

Joulinar, 2022-11-16:
can you share following

```
ip a
```

closed 2022-11-17 by kgc35

MichaIng, 2022-11-17:
and as well:
```sh
ip r
```
Might be similar to #5891, router not sending out router advertisements timely to update the client's IPv6 address and/or default route.

Or did you solve it already (and therefore closed the issue)?