MichaIng/DietPi #5886: Kernel does not include br_netfilter on soquartz breaking k3s

brucedevcom, 2022-11-16:
On the soquartz (possibly others? I have no other devices to test with), both in the 8.10 and 8.11.1 beta the br_netfilter module is not included with the kernel which prevents k3s from starting.

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8 G_DIETPI_VERSION_SUB=11 G_DIETPI_VERSION_RC=1 G_GITBRANCH='beta' G_GITOWNER='MichaIng'`
- Distro version | `bullseye`
- Kernel version | `Linux DietPi 6.1.0-rc1 #1 SMP PREEMPT Wed Nov  9 01:08:14 UTC 2022 aarch64 GNU/Linux`
- SBC model | `Quartz64 (aarch64)` (actually soquartz)
- Power supply used | 12v supply to CM4 IO Board
- SD card used | Sandisk Ultra 64GB

#### Additional Information (if applicable)
- Software title | k3s
- Was the software title installed freshly or updated/migrated? | Fresh
- Can this issue be replicated on a fresh installation of DietPi? | Yes

#### Steps to reproduce

1. Install k3s, either via diepti-software or the official install script

#### Expected behaviour
- k3s service should be working

#### Actual behaviour

- k3s gets killed soon after start due to failure of flannel network mesh
```
 ● k3s.service - Lightweight Kubernetes
     Loaded: loaded (/etc/systemd/system/k3s.service; disabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/k3s.service.d
             └─dietpi.conf
     Active: failed (Result: exit-code) since Wed 2022-11-16 04:19:17 GMT; 11s ago
       Docs: https://k3s.io
    Process: 1800 ExecStartPre=/bin/sh -xc ! /usr/bin/systemctl is-enabled --quiet nm-cloud-setup.service (code=exited, status=0/SUCCESS)
    Process: 1802 ExecStartPre=/sbin/modprobe br_netfilter (code=exited, status=1/FAILURE)
    Process: 1803 ExecStartPre=/sbin/modprobe overlay (code=exited, status=0/SUCCESS)
    Process: 1805 ExecStart=/usr/local/bin/k3s server (code=exited, status=1/FAILURE)
   Main PID: 1805 (code=exited, status=1/FAILURE)
        CPU: 1min 4.847s

Nov 16 04:19:16 DietPi k3s[1805]: I1116 04:19:16.925011    1805 reconciler.go:357] "operationExecutor.VerifyControllerAttachedVolume started for volume \"tmp-dir\" (UniqueName: \"kubernetes.io/empty-dir/b8cb6ce4-6f95-4ed0-8f50-d5b9c4648351-tmp-dir\") pod \"metrics-server-5c8978b444-vkz25\" (UID: \"b8cb6ce4-6f95-4ed0-8f50-d5b9c4648351\") " pod="kube-system/metrics-server-5c8978b444-vkz25"
Nov 16 04:19:16 DietPi k3s[1805]: I1116 04:19:16.925347    1805 reconciler.go:357] "operationExecutor.VerifyControllerAttachedVolume started for volume \"kube-api-access-tfgvw\" (UniqueName: \"kubernetes.io/projected/429e6b53-6043-44e9-8f55-80a59d04701d-kube-api-access-tfgvw\") pod \"local-path-provisioner-5b5579c644-5qv6x\" (UID: \"429e6b53-6043-44e9-8f55-80a59d04701d\") " pod="kube-system/local-path-provisioner-5b5579c644-5qv6x"
Nov 16 04:19:16 DietPi k3s[1805]: I1116 04:19:16.925501    1805 reconciler.go:357] "operationExecutor.VerifyControllerAttachedVolume started for volume \"config-volume\" (UniqueName: \"kubernetes.io/configmap/1093b71e-cc1d-49be-b03d-fd690d236123-config-volume\") pod \"coredns-75fc8f8fff-mj6hr\" (UID: \"1093b71e-cc1d-49be-b03d-fd690d236123\") " pod="kube-system/coredns-75fc8f8fff-mj6hr"
Nov 16 04:19:16 DietPi k3s[1805]: I1116 04:19:16.925636    1805 reconciler.go:357] "operationExecutor.VerifyControllerAttachedVolume started for volume \"custom-config-volume\" (UniqueName: \"kubernetes.io/configmap/1093b71e-cc1d-49be-b03d-fd690d236123-custom-config-volume\") pod \"coredns-75fc8f8fff-mj6hr\" (UID: \"1093b71e-cc1d-49be-b03d-fd690d236123\") " pod="kube-system/coredns-75fc8f8fff-mj6hr"
Nov 16 04:19:16 DietPi k3s[1805]: I1116 04:19:16.975445    1805 kube.go:127] Node controller sync successful
Nov 16 04:19:16 DietPi k3s[1805]: I1116 04:19:16.975702    1805 vxlan.go:138] VXLAN config: VNI=1 Port=0 GBP=false Learning=false DirectRouting=false
Nov 16 04:19:16 DietPi k3s[1805]: time="2022-11-16T04:19:16Z" level=fatal msg="flannel exited: operation not supported"
Nov 16 04:19:17 DietPi systemd[1]: k3s.service: Main process exited, code=exited, status=1/FAILURE
Nov 16 04:19:17 DietPi systemd[1]: k3s.service: Failed with result 'exit-code'.
Nov 16 04:19:17 DietPi systemd[1]: k3s.service: Consumed 1min 4.847s CPU time.
```

#### Extra details
```
root@DietPi:~# lsmod | grep br_netfilter
root@DietPi:~# /sbin/modprobe br_netfilter
modprobe: FATAL: Module br_netfilter not found in directory /lib/modules/6.1.0-rc1
```

MichaIng, 2022-11-16:
Related forum topic: https://dietpi.com/forum/t/dietpi-soquartz-bullseye-kernel-downgrade/14313

I'll open an MR at Peter Geis' repository. The same was for for WireGuard already, but sadly no reaction so far.

MichaIng, 2022-11-17:
MR up: https://gitlab.com/pgwipeout/quartz64_ci/-/merge_requests/6
Crossing fingers 🤞.

MichaIng, 2023-02-11:
FYI: #6151

MichaIng, 2023-02-19:
Solved with new images: https://dietpi.com/downloads/images/testing/
Moving to stable downloads after some more review and testing tomorrow.

The kernel can be upgraded as well with the matching firmware package form here: https://dietpi.com/downloads/binaries/
Will be offered on next DietPi update as well.

closed 2023-02-19 by MichaIng