MichaIng/DietPi #5888: Nextcloud install option is missing in Dietpi-Software on Odroid XU4 Buster

PaulPink, 2022-11-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | buster
- Kernel version | Linux DietPi 4.14.294+ 
- SBC model | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)
- Power supply used | Odroid Cloudshell2 Power Supply
- SD card used | eMMC 16GB

#### Additional Information (if applicable)
- Software title |  Nextcloud
- Was the software title installed freshly or updated/migrated?
Not able to install, there is no option in the dietpi-software menu

#### Expected behaviour

- Like bevor, it simple installs Nextcloud

#### Actual behaviour

- [FAILED] DietPi-Software | Software title (Nextcloud) is not supported on armv7l systems.

Joulinar, 2022-11-16:
Yes, we disabled NC on ARMv6/7 due to following NC issue https://github.com/nextcloud/server/issues/34674

@MichaIng do you know if this will be permanent? At least related issue has been closed.

MichaIng, 2022-11-16:
Yes, it is already broken since NC25 and the update to NC26 is blocked. 32-bit support has gone.

Joulinar, 2022-11-16:
ok, personally I would include this information in our release notes.

PaulPink, 2022-11-16:
ok, i was afraid of something like that. the saying comes true: Never change a running system. 
Thank you very much for the quick reply

PaulPink, 2022-11-16:
> ok, personally I would include this information in our release notes.

That would be great, was searching the "whole" web for the reason.
So I close this for now.

Joulinar, 2022-11-16:
> That would be great, was searching the "whole" web for the reason.

Quite deep within GitHub https://github.com/MichaIng/DietPi/commit/0efb5d22806049f81da33811e89077aae482a1bb  but yes not that visible.

MichaIng, 2022-11-16:
> Never change a running system.

Be careful with this wording when it's about server systems. The safety of your data and privacy relies on that your software is up-to-date, so it is very much mandatory to keep things updated, especially Nextcloud 😉.

Actually on Buster, NC23 is latest supported anyway, due to PHP7.3. We could pull NC24 as latest supported version on ARMv6/7, to increase the time a bit for admins to switch to 64-bit hardware. But there is no proper update blocker for NC24 => 25 (broken already), only 25 => 26. I'll open a PR to re-add 32-bit support to Nextcloud (it's removing 4 characters only) so that users do not run into a broken instance and we can safely provide NC25 for those systems. Still supported for nearly a year: https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule

MichaIng, 2022-11-16:
Ah done already: https://github.com/nextcloud/server/commit/c124456

PaulPink, 2022-11-16:
Of course I always install all updates and keep my system up to date, but until Sunday morning everything was running fine on Buster and for some reason I can't quite figure out I went ahead and upgraded to Bullseye. 
After that the fan on my Cloudshield2 stopped running and Nextcloud was gone. After an evening of disappointment, I grabbed the Buster version from the archives, rebuilt everything, and found the above problem. 
So all in all, a reason for: Never change a Running system 😉
If you have an off topic tip how I get the fan running again my future-me would be very much obliged.

MichaIng, 2022-11-18:
Re-enabled Nextcloud for 32-bit systems by installing latest v25 and, until v25.0.2 release, patching v25.0.1 the same way done by Nextcloud devs: https://github.com/MichaIng/DietPi/commit/450c613

closed 2022-11-18 by MichaIng

MichaIng, 2023-02-03:
The decision has been reverted btw. Nextcloud 26 will support 32-bit, it seems 🙂: https://help.nextcloud.com/t/update-on-32-bit-support-decision-to-block-the-update-on-nextcloud-26-reverted/153345

The beta is still blocked for the update, but I think this is a matter of another Nextcloud 25 release, since there it needs to be unblocked.

PaulPink, 2023-02-08:
That's great news, thank you for making it possible and keeping us up to date. Which, of course, sounds like a joke with 32bit... 😉

Joulinar, 2023-02-08:
But even if NC26 will support 32bit systems, you might not be able to upgrade because of PHP7.4 you are running on (using Debian Bullseye)

MichaIng, 2023-02-08:
Yes, on Buster, latest supported Nextcloud version is 23, on Bullseye 25, for NC26 an upgrade to Bookworm is needed.

PaulPink, 2023-02-09:
So it's about time for me to get a 64bit alternative to my XU4. So far, I've only been using Nextcloud in the internal network anyway, so of course it's desirable to be up to date, but not as essential as in the wild.

Joulinar, 2023-02-09:
first step would be to upgrade the existing system to Debian Bullseye ;)

https://dietpi.com/blog/?p=811

PaulPink, 2023-02-09:
I would do that directly, but the last time I tried, I not only ran into problems with Nextcloud, but also the fans of Cloudshell2 no longer wanted to be addressed. I have found information on this, but none that has calmed the heated tempers. But I should address the problem at Hardkernel, unless you have a quick answer for me.

Joulinar, 2023-02-09:
Maybe you can have a look to following forum post https://dietpi.com/forum/t/cloudshell-2-not-working-after-upgrade/15331

PaulPink, 2023-02-09:
Great, if the answer to your problems comes when you have given up the search. 😑
Thanks a lot 
I will give it a try

MichaIng, 2023-02-09:
> So it's about time for me to get a 64bit alternative to my XU4.

For Nextcloud at least, it is not necessary anymore 😉.