MichaIng/DietPi #5891: dietpi run on rockpi4plus is not updating

r0nindesperad0, 2022-11-17:
i am new to the dietpi  os.
ia m facing problem to start dietpi on rockpi4 plus v1.73.
i cant ssh to the pi and need to use monitor and keyboard but it show me warning and error from deb.debian.org
pi is connected to via lan cable to my router,keyboard and a monitor now.
after success login using root/dietpi it started updating but received this message
Err 2 https://deb.debian.org/debian bullseye InRelease
Temporary failure resolving 'deb.debian.org'
and so on....

Joulinar, 2022-11-17:
Are you sure you have a valid network connection? Can you show following

```
ip a
```

r0nindesperad0, 2022-11-17:
I have a valid internet connection..now i had my rockpi4 running on
bullseye from radxa and it works fine..i was trying to opt out to using a
lighter os and seems like dietpi jad support on rockpi4.plus sbc...i'll
make new installation tomorrow using dietpi os just to be sure what is the
cause because it register to my router and it shows that connected to
certain ip but return an error or package not available..i'll let u know
when i run the dietpi os back on rockpi4.
Thanks

Joulinar, 2022-11-17:
> Temporary failure resolving 'deb.debian.org'

Your issue is/was incorrect DNS resolution. Therefore check IP address information assigned if they are correct.

r0nindesperad0, 2022-11-17:
This is the ip status of my rockpi.

r0nindesperad0, 2022-11-17:
[image]

Joulinar, 2022-11-17:
There is nothing shared. I guess you tried to send a picture via email. best is to login to GitHub directly. Information would be required from DietPi system.

MichaIng, 2022-11-17:
> because dietpi doesnt use the /etc/network/interfaces like other debian.

DietPi does use `/etc/network/interfaces`. But on first boot, settings from `/boot/dietpi.txt` and in case `/boot/dietpi-wifi.txt` are applied, overriding `/etc/network/interfaces`.

What I'm confused about is that Internet access seems to be there, including general name resolution, as the related checks are passed. But only `deb.debian.org` cannot be resolved, and you cannot SSH into the system? Can you test:
```sh
cat /etc/resolv.conf
ping -4c1 google.com
ping -6c1 google.com
ping -4c1 deb.debian.org
ping -6c1 deb.debian.org
ip -6 r
```
Since I see only an LLA (local-link address) IPv6 address in your screenshot, but no GUA (global unicast address), the system shouldn't be able to connect to any remote host via IPv6, but via IPv4 only. Not sure why it connects to `google.com` via IPv4 by default, usually IPv6 should always be used, or `ping` has an internal fallback.

But in this case, the IPv6 connection check should have failed first, before APT updates are even attempted 🤔. GUAs when received via router advertisements (RA), have an expiry time, although it is usually very long (compared to the default route, which expires usually in 30 minutes). So in theory it is possible, but unlikely, that there was a GUA and IPv6 default route when the IPv6 connection check was done, but it expired when the APT update was done. If any of this is the reason, the router does not send out RAs in sufficient frequency to update the device's IPv6 information.

r0nindesperad0, 2022-11-17:
[image]

Joulinar, 2022-11-17:
Could you disable IPv6 on `dietpi-config` and try again to update the system?

MichaIng, 2022-11-17:
The default route indeed is missing (as well). Are other systems on your network able to use IPv6 reliably (e.g. `ping -6 google.com` can be used from Windows on macOS terminal as well)? And are there IPv6 related settings available in your router?

Disabling IPv6 is a workaround, of course.

r0nindesperad0, 2022-11-17:
i dont use ipv6 setting in my router for any of my devices.i had disable ipv6 on dietpi-config and reboot the rockpi and still cant update the dietpi.
i have a number bpi m64 running debian os from avafinger as my iot gateway and is running normally.
suppose if ipv6 is not enable it should use ipv4 right?

MichaIng, 2022-11-17:
> i had disable ipv6 on dietpi-config and reboot the rockpi and still cant update the dietpi.

What is the error message then? Should be a different one.

> suppose if ipv6 is not enable it should use ipv4 right?

No, there is no automatic fallback for IPv6 on Linux. If it is enabled, which is by default, all programs will prefer IPv6, if not overwritten by settings explicitly.

On Radxa's Debian Bullseye image the issue should be the same. At least I didn't see any setting which would disable IPv6 or blacklist the module there 🤔.

r0nindesperad0, 2022-11-17:
i have other sd card with radxa debian bullseye image and was running normally.
ssh and update n upgrade is ok.
[image]

MichaIng, 2022-11-17:
Does it work there to ping IPv6 hosts?
```sh
ping -6c1 deb.debian.org
ip -6 a
ip -6 r
```

r0nindesperad0, 2022-11-17:
[image]

MichaIng, 2022-11-17:
So the issue is the same. Only for some reason `httpredir.debian.org` is resolved to an IPv4 address. It is actually only an alias for `debian.map.fastlydns.net`, same as `deb.debian.org`. Can you show:
```sh
getent hosts httpredir.debian.org
getent ahosts httpredir.debian.org
```

r0nindesperad0, 2022-11-17:
rock@rockpi-4b:~$ getent hosts httpredir.debian.org
2a04:4e42:48::644 debian.map.fastlydns.net httpredir.debian.org
rock@rockpi-4b:~$ getent ahosts httpredir.debian.org
199.232.46.132  STREAM debian.map.fastlydns.net
199.232.46.132  DGRAM  
199.232.46.132  RAW    
2a04:4e42:48::644 STREAM 
2a04:4e42:48::644 DGRAM  
2a04:4e42:48::644 RAW    
rock@rockpi-4b:~$

MichaIng, 2022-11-17:
Interesting, based on this the IPv6 address should be used by default, i.e. this should fail as well:
```sh
ping -c1 httpredir.debian.org
```
I'll have another look into Radxa's image whether I can find something that makes APT prefer IPv4. However, as you can see, the issue is the same and you need to generally actively disable IPv6 on Linux systems, if your router does not support it/does not send out router advertisements.

r0nindesperad0, 2022-11-17:
the command does not fail

rock@rockpi-4b:~$ ping -c1 httpredir.debian.org
PING debian.map.fastlydns.net (199.232.46.132) 56(84) bytes of data.
64 bytes from 199.232.46.132 (199.232.46.132): icmp_seq=1 ttl=53 time=15.0 ms

--- debian.map.fastlydns.net ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 14.966/14.966/14.966/0.000 ms
rock@rockpi-4b:~$

eyduh, 2022-12-05:
I just had the same problem on latest dietpi on x86_64 and following this thread solved the issue, spcifically getting rid of all traces of ipv6.

eyduh, 2022-12-05:
```
   self.gen.throw(type, value, traceback)                                                                             
                                             │   File "/tmp/tmpfz0tt1i7/pip.zip/pip/_vendor/urllib3/response.py", line 442, in _error_catcher                         
                                             │     raise ReadTimeoutError(self._pool, None, "Read timed out.")                                                        
                                             │ pip._vendor.urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read    
                                             │ timed out.   
```

MichaIng, 2023-02-12:
Marking as closed. If IPv6 is not reliably handled by the router (assigning a default route via RA regularly), it should be disabled on all clients. There may be fallbacks, but they may not work for all software, at least not reliably.

closed 2023-02-12 by MichaIng