MichaIng/DietPi #5895: Diet Scheduled Reboot.

jdaien4783, 2022-11-20:
What is the easiest / simplest way to tell the DietPi on raspberry pi to reboot daily or weekly?

Joulinar, 2022-11-20:
Basically this is not specific to DietPi. You can schedule a reboot as on every other Linux system. Easiest might be to create a `crontab` entry.

MichaIng, 2022-11-20:
Before you try the obvious way:
```sh
echo -e '#!/bin/dash\nreboot' > /etc/cron.daily/zzz-reboot
chmod +x /etc/cron.daily/zzz-reboot
```
The problem with this is that, AFAIK, those jobs can run concurrently, so that the reboot hard kills other jobs, e.g. backups and such, which should definitely not be aborted. So it's better to create a dedicated `crontab` entry, either via command:
```sh
sudo crontab
```
or more transparently
```sh
echo '59 3 * * * root reboot' > /etc/cron.d/daily-reboot
```
to reboot every day at 03:59 or
```sh
echo '59 3 * * 7 root reboot' > /etc/cron.d/weekly-reboot
```
to reboot every Sunday at 03:59.

jdaien4783, 2022-11-20:
Still a little confused? Is this suppose to be a one step one command setup? If I just copy and paste:

"echo '59 3 * * * root reboot' > /etc/cron.d/daily-reboot"

I get the following error:

[image]

I have tried it with sudo also. I get the same error.

MichaIng, 2022-11-20:
You do not need root permissions for executing `echo`, but for writing to the file, so `sudo echo something > /path/to/file` does generally not work (the `>` redirect is shell syntax and not part of the `sudo echo` command). Instead use:
```sh
echo '59 3 * * * root reboot' | sudo tee /etc/cron.d/daily-reboot
```

MichaIng, 2022-11-24:
Did this work for you?

StephanStS, 2022-11-30:
Could it also be a solution, that on reboot you execute `shutdown -r --no-wall +1440` (daily) resp. `shutdown -r --no-wall +10080` (weekly)? The idea is to trigger the reboot directly at system startup.

MichaIng, 2022-11-30:
Requires `dbus` and probably `systemd-logind`, and not sure what the benefit is?

MichaIng, 2023-03-01:
Marking as closed, as initial question has been answered.

closed 2023-03-01 by MichaIng