MichaIng/DietPi #5896: How to reset rfkill status on boot?

DrCWO, 2022-11-20:
On a Raspberry Pi my application uses `rfkill `to disable WiFi if connected to ethernet.
On fresh power on reboot this is still disabled even is no ethernet is connected.

Question:
How to setup DietPi to enable WiFi on a reboot or power on?

Joulinar, 2022-11-20:
In past we tested with `ifpluged`. Maybe you can have a look https://dietpi.com/forum/t/using-ethernet-wlan-interfaces-not-at-the-same-time-in-v6-30/4251

MichaIng, 2022-11-20:
Btw, instead of blocking WiFi via `rfkill`, just disabling (or not enabling) the WiFi interface via `ifup`/`ifdown` (temporarily) and `dietpi-config`/`/etc/network/interfaces` (persistent) is IMO preferable. Otherwise the system still tries to setup WiFi, failing of course. `rfkill` is to additionally assure that radio connections cannot be established if there are unknown mechanisms trying to set them up or which are difficult to control. But on DietPi such shouldn't be an issue 😉.

DrCWO, 2022-11-21:
I have to `rfkill `as I like to prevent sending radio signals if not needed :-)
After some experiments I fould this solution:
```
rfkill unblock wlan
rfkill unblock bluetooth
systemctl mask systemd-rfkill
```
The first two commands enable Wlan and Bluetooth. The third command prevents that changes done with `rfkill `in my application got recorded in `/var/lib/systemd/rfkill/*`.

After this procedure the **rfkill settings are always unblocked after reboot** no matter what my application does 👍

MichaIng, 2022-11-21:
> I have to `rfkill` as I like to prevent sending radio signals if not needed :-)

As said, `ifdown wlan0` practically should have the same effect on plain DietPi (although it might depend on adapter firmware). I'd at least do this before `rfkill block wlan`, so wpasupplicant and in case the DHCP server are stopped properly. Then when needed, `rfkill unblock wlan; ifup wlan0`.

Ah right, `systemd-rfkill` stores/restores `rfkill` persistently at shutdown/boot 👍.

DrCWO, 2022-11-21:
Good proposal, I will do this 👍

closed 2022-11-24 by MichaIng