MichaIng/DietPi #5899: How to save the system running on emmc as an image? thank you!

hifitime, 2022-11-20:
Hi Dear friends

How to save the dietpi system running on emmc as an image? thank you!

Cheers~

MichaIng, 2022-11-20:
You mean forcing boot from SD card? Which SBC and image?

Joulinar, 2022-11-20:
Yes, best would be to boot of SD card and use our `dietpi-imager` to create `*.img` file from eMMC.

hifitime, 2022-11-20:
> 

That's exactly what I want. Thank you very much,

Cheers!

hifitime, 2022-11-20:
> Yes, best would be to boot of SD card and use our `dietpi-imager` to create `*.img` file from eMMC.

Thank you. That's exactly what I want. Thank you very much,

Cheers!

MichaIng, 2022-11-20:
Ah, I misunderstood 😅. Since it's not (yet) included on DietPi systems, the imager can be called, also from non-DietPi systems (but Debian or Ubuntu Jammy is possible):
```sh
bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-imager')"
```

hifitime, 2022-11-20:
> Thank you. That's exactly what I want. Thank you very much,
> 
> Cheers!

Thank you very much~

Cheers!

hifitime, 2022-11-20:
> Ah, I misunderstood 😅. Since it's not (yet) included on DietPi systems, the imager can be called, also from non-DietPi systems (but Debian or Ubuntu Jammy is possible):
> 
> ```shell
> bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-imager')"
> ```

Thank you very much,

Cheers!

MichaIng, 2022-11-24:
I'll mark this as closed. Feel free to reopen if required.

closed 2022-11-24 by MichaIng

jalsco, 2022-11-24:
Hello, I also want to do similar 😊 Currently I have (running and bootable) a dietpi system on SD card. I would like to clone it onto an NVMe SSD disk on the same system . The second step would be to make the SBC boot from that NVMe SSD.   I am running on RockPi 4A v1.4 -( it has a flashable SPI bootloader).  Is the `dieptpi-imager` suitable for that purpose too ?

MichaIng, 2022-11-24:
I guess the ROCK Pi 4 cannot boot from USB directly (using U-Boot flashed on the USB drive), but needs to load the bootloader from SPI then, right?

Not sure which bootloader is flashed OOTB by Radxa, probably it can do that right away, detecting and using the `/boot/boot.scr` on the USB drive? Otherwise, either the SPI bootloader needs an update (probably provided by Radxa), or Armbian's U-Boot needs to be flashed to the SPI:
```sh
SPI_DEV=$(mawk '/mtd/{print $NF}' /proc/partitions)
if [[ $SPI_DEV ]]
then
    . /usr/lib/u-boot/platform_install.sh
    write_uboot_platform_mtd "$DIR" "$SPI_DEV"
else
    echo 'ERROR: No SPI device found'
fi
```

jalsco, 2022-11-24:
I have successfully flashed the SPI with uboot Bootloader as per :   [https://wiki.radxa.com/Rockpi4/dev/spi-install](url)
The uboot searches for `/extlinux/extlinux.conf` and load` kernel,dtb` on first partition or partition which set bootable flag. 
as stated here: [http://rockchip.wikidot.com/boot-sequence](url)

MichaIng, 2022-11-24:
> as stated here: http://rockchip.wikidot.com/boot-sequence

This is a generic Rockchip Wiki, which isn't necessarily true for the bootloader shipped by Radxa. Usually, U-Boot looks for `boot.scr` first, then falls back to extlinux. But it depends, since vendors can ship highly customised bootloaders.

> My SSD is connected to the M2 connector - not via USB.
> So - the uboot - finds nothing on my system.

Then indeed either Radxa's SPI bootloader does not support `boot.scr`, or the syntax used in it.

> Then I disbale the uboot (in SPI) and just let it boot from the SD card.

The hardware cannot boot from M2 attached SSD directly, when SPI bootloader is disabled?

> But... if the SSD was setup with bootable partition (as above) , I beleive the SPI uboot would then boot from the SSD.

Sure, the question is what it really needs. We could theoretically add an extlinux config, but it doesn't support everything we need, respectively the vendor SPI bootloaders do not support all of the latest extlinux features (like device tree overlays). I hope Armbian's ROCK Pi 4 U-Boot build does support booting from M2, then it would use `/boot/boot.scr`.

jalsco, 2022-11-25:
Hi @MichaIng , thanks so much for replying. 

The "[Armbian's ROCK Pi 4 U-Boot bootloader](https://dl.radxa.com/rockpi/images/loader/spi/)"  - which I have flashed to the SPI on my device - DOES- support booting from M2 SSD devices. I specifically bought M2 NVMe SSD model that was tested and confirmed to work, including booting from it. 

**_Does "Radxa's SPI bootloader support `boot.scr`" ?_**  I think it probably yes. 

_**Can the hardware boot from M2 attached SSD directly, when SPI bootloader is disabled?**_ - Not sure. Finding out the answer, by me testing it, is actually the subject of my question.  In order to do that, I must:
1.  image copy my DietPi system from my SD onto the SSD 
2.  make the SSD partition bootable and then 
3.  try it - with SPI enabled and with SPI disabled.. 
I know how to do steps 2 and 3 - just looking for some advice about doing step 1 (above), 
```
[Thu Nov 24] root@batopi: ~$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0  59.6G  0 disk
└─mmcblk0p1 179:1    0  59.6G  0 part /
nvme0n1     259:0    0 238.5G  0 disk
└─nvme0n1p1 259:1    0 238.5G  0 part
```

```
[Thu Nov 24] root@batopi: ~$ gdisk /dev/nvme0n1
GPT fdisk (gdisk) version 1.0.6

Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with protective MBR; using GPT.

Command (? for help): p
Disk /dev/nvme0n1: 500118192 sectors, 238.5 GiB
Model: GIGABYTE GP-GSM2NE3256GNTD
Sector size (logical/physical): 512/512 bytes
Disk identifier (GUID): 4A4B2744-9F24-403F-8B47-FD6FB5DDF5DE
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 500118158
Partitions will be aligned on 2048-sector boundaries
Total free space is 2669 sectors (1.3 MiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048       500117503   238.5 GiB   8300  primary
```

jalsco, 2022-11-25:
```
# default values
setenv rootdev "/dev/mmcblk0p1"
```
That is in `boot.scr`  is it likely to be a problem ? 
The file says "DO NOT EDIT" - otherwise it would be easy to change to `/dev/nvme0n1p1` after doing an image copy from my SD card (however I will do that copying ?)

Thoughts ?

jalsco, 2022-11-26:
Hi @MichaIng, 
any thoughts you have, or advice,  on how best to clone my dietpi SD onto my NVMe SSD  - 
would be most welcome  and highly appreciated 🙏🏻😊

Joulinar, 2022-11-26:
This one could give a hint https://dietpi.com/blog/?p=1236

jalsco, 2022-11-27:
Thanks so much ❤👍 @Joulinar ! That is EXACTLY what I was looking for 😁🙏🏻🙏🏻