MichaIng/DietPi #5911: No SSH access on fresh Zeropi install

skyweir, 2022-11-22:
Flashed several times with DietPi and cannot access the device. Test flash with Armbian and all is well.

MichaIng, 2022-11-22:
Thanks for reporting.

Does the ZeroPi show up in router as DHCP client and can you ping it?

skyweir, 2022-11-23:
ZeroPi shows up as DHCP client. SSH and ping both reply host is down

MichaIng, 2022-11-23:
It shows up as DHCP client but ping doesn't work? That is confusing as the first implies the second, as long as ICMP packages are not actively filtered/blocked 🤔.

You are sure to used the correct IP address (sorry for asking 😉)?

And you used the Bullseye image dated "2022-11-20 23:35+01:00" right?

skyweir, 2022-11-23:
I was mistaken. There is no DHCP entry for the ZeroPI. I am using the version referenced above. Sorry no UART adapter on hand. 

It was working with a previous DietPI version. Is it possible to obtain an older image ~6 months ago?

Joulinar, 2022-11-23:
We don't keep old images available. Usually the current image should work the very same way.

MichaIng, 2022-11-23:
I moved all images to our build script, so there is a difference. But I reviewed everything and couldn't find an error so far. Bootloader, kernel and boot config should match the one from Armbian 100%, respectively boot config in all relevant aspects.

Although, found something that might affect U-Boot on 32-bit systems. I'll upload a new image shortly.

MichaIng, 2022-11-23:
Okay, I generated some images to test, each with a different change which may or may not fix the boot issue: https://dietpi.com/downloads/images/testing/

@StephanStS 
If your ZeroPi isn't in use otherwise, could you verify that our latest image from download page does not boot? You do not have a UART adapter either, in case, do you?

Nasty that I don't have any Allwinner H3 board here to test myself. For sure, if there was an issue, all of them would be affected.

skyweir, 2022-11-23:
No UART adapter available.

None of the images tested resulted in success. The latest image from the download page failed as well.

Additionally I ran the script to convert my Armbian install to DietPi. Completed w/o incident but failed with the same result upon reboot.

MichaIng, 2022-11-23:
Okay, thanks for the tests, including running `dietpi-installer` on an Armbian image, which rules out partitioning and filesystem feature differences.

Just to assure, the Armbian image remains functional after upgrading to latest kernel and flashing latest U-Boot, is it?
```sh
apt update
apt full-upgrade
. /usr/lib/u-boot/platform_install.sh
write_uboot_platform "$DIR" "write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$(findmnt -Ufnro SOURCE -T /boot)")"
```

StephanStS, 2022-11-23:
> Okay, I generated some images to test, each with a different change which may or may not fix the boot issue: https://dietpi.com/downloads/images/testing/
>
> @StephanStS If your ZeroPi isn't in use [...]

My 2nd ZeroPi is unpowered, I will check all images soon.

StephanStS, 2022-11-23:
Tested images:

- DietPi_ZeroPi-ARMv7-Bullseye.img. Strange display in my router, it should show DietPi on the left instead of PC-192-168-178-134.  
  [image]  
  Also no SSH connection to the board. A ping to the board works.
- DietPi_ZeroPi-ARMv7-Bullseye_devnum.img: Identical behaviour as DietPi_ZeroPi-ARMv7-Bullseye.img
- DietPi_ZeroPi-ARMv7-Bullseye_bootargs.img: Identical behaviour as DietPi_ZeroPi-ARMv7-Bullseye.img
- DietPi_ZeroPi-ARMv7-Bullseye_console.img: Identical behaviour as DietPi_ZeroPi-ARMv7-Bullseye.img
- DietPi_ZeroPi-ARMv7-Bullseye_checksum.img: Identical behaviour as DietPi_ZeroPi-ARMv7-Bullseye.img

MichaIng, 2022-11-23:
This is probably U-Boot looking for network drives, using a different hostname or none at all.

StephanStS, 2022-11-23:
My last Bullseye image (which I generated via PREP on 2021-09-11) boots fine with SSH access.   
:100: %

MichaIng, 2022-11-24:
Found it, the initramfs was built in `arm64` format. New images are currently created: https://github.com/MichaIng/DietPi/actions/workflows/dietpi-build.yml?query=branch%3Adev

MichaIng, 2022-11-24:
Images have been updated.

MichaIng, 2022-11-24:
Great, many thanks for testing 👍.

> the initial (grey) licence acknowledge dialog did not appear

I removed it 🙂. It isn't common on other similar FLOSS projects, users who want to know the license can find it in many places, so one action less to do for end users.

StephanStS, 2022-11-24:
> Great, many thanks for testing +1.
> 
> > the initial (grey) licence acknowledge dialog did not appear
> 
> I removed it slightly_smiling_face. It isn't common on other similar FLOSS projects, users who want to know the license can find it in many places, so one action less to do for end users.

Is there a pull request # or an issue # where I could refer the resulting DietPi-Doc changes to (screenshot and text in https://dietpi.com/docs/install/#4-first-logon-on-dietpi)?

MichaIng, 2022-11-24:
https://github.com/MichaIng/DietPi/commit/267a582

> Also, no CHANGELOG.txt entry is present.

Too small to mention in changelog.

skyweir, 2022-11-24:
Thanks for the fixes my ZeroPi is happier now.

 I hate to mention this, I just reflashed a NEO2 and am experiencing the same issues previously reported.

MichaIng, 2022-11-24:
Great that it works now 🙂. I'll mark it as closed then. Feel free to reopen if required.

closed 2022-11-24 by MichaIng