MichaIng/DietPi #5912: /sbin/shutdown: Failed to connect to bus: No such file or directory

Razerban, 2022-11-22:
#### Required Information
- DietPi version | `v8.11.2`
- Distro version | `bullseye 0`
- Kernel version | `Linux raspberrypi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`

#### Steps to reproduce
Run `/sbin/shutdown`. I executed the following command when I got the error: `/sbin/shutdown --no-wall --reboot 0`

#### Expected behaviour
System reboots.

#### Actual behaviour
Error message: `Failed to connect to bus: No such file or directory`

#### Extra details
This happened on a fresh install of DietPi v8.11.2 from the image: `DietPi_RPi-ARMv8-Bullseye.7z` - checksum: `md5: 2abefb48da214d9ac0a68acecc441c7a`.

MichaIng, 2022-11-22:
Why is everyone using the `shutdown` command for simple shutdowns and reboots? 😄 
Use `poweroff` and `reboot`.

`shutdown` is mainly for scheduled shutdowns in X seconds and such, which requires `dbus` and, AFAIK even `systemd-logind`. If you need this:
```sh
apt install dbus
shutdown +5 # shutdown in 5 minutes
```
If the error still happens:
```sh
systemctl unmask systemd-logind
systemctl start systemd-logind
shutdown +5 # shutdown in 5 minutes
```

Razerban, 2022-11-23:
Bad habit I suppose 😆 

I saw that the there is an option inside /boot/dietpi.txt to have the service `systemd-logind` unmasked during first inital setup/boot:
```shell
# Unmask (enable) systemd-logind service (including dbus), which is masked by default on DietPi
AUTO_UNMASK_LOGIND=0
```

The same option installs `dbus` if it's not installed:
https://github.com/MichaIng/DietPi/blob/98873288ede2c05f4a3b60201759c519da82ed04/dietpi/dietpi-software#L15938-L15950

Thanks for your help @MichaIng 😄

closed 2022-11-23 by Razerban

MichaIng, 2022-11-23:
Yes exactly, since there are some other features depending on this.

frankk74, 2023-03-01:
shutdown usage is from the “olden days” when you had to warn other people (give them some time to save stuff) who were also using the system… it also allowed a message to sent as well, the message goes into the logs..

MichaIng, 2023-03-01:
I do also see use for this nowadays, but for the typical DietPi user very rare.

Actually, since Bookworm, even the `poweroff` and `reboot` commands ping dbus (likely `systemd-logind`) and print an error message (no affecting the poweroff/reboot) if not present. I didn't check yet if it's those commands, or some part of the shutdown sequence, and whether there is a way to omit it. It does not cause any issue, but it is ugly to have those error messages, so I'm thinking whether we finally (need to) pre-install and enable dbus, systemd-logind and libpam-systemd on DietPi with Bookworm.