MichaIng/DietPi #5915: [FAILED] DietPi-Software

ckazi, 2022-11-24:
#### Details:
- Date           | Thu Nov 24 13:28:28 MSK 2022
- DietPi version | v8.10.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux dlna 5.10.0-19-amd64 #1 SMP Debian 5.10.149-2 (2022-10-21) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y --allow-change-held-packages install libatomic1`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
libatomic1 is already the newest version (10.2.1-6).
libatomic1 set to manually installed.
The following package was automatically installed and is no longer required:
  linux-image-5.10.0-18-amd64
Use 'apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up openssh-server (1:8.4p1-5+deb11u1) ...
rescue-ssh.target is a disabled or a static unit, not starting it.
Job for ssh.service failed because the control process exited with error code.
See "systemctl status ssh.service" and "journalctl -xe" for details.
invoke-rc.d: initscript ssh, action "start" failed.

MichaIng, 2022-11-24:
Many thanks for reporting.

The error is unrelated to the software installation you are trying, but the OpenSSH server fails to start. That it is triggered here, on an unrelated package install, indicates that you tried to install it before where it failed already, so the package is in an incomplete state. Let's check this:
```sh
journalctl -u ssh
```

ckazi, 2022-11-24:
I think that is because I use DropBear.
I dont understand why command "apt-get update" try to install OpenBSD Secure Shell server.

journalctl -u ssh
Nov 24 13:59:09 dlna systemd[1]: Starting OpenBSD Secure Shell server...
Nov 24 13:59:10 dlna sshd[35213]: error: Bind to port 22 on 0.0.0.0 failed: Address already in use.
Nov 24 13:59:10 dlna sshd[35213]: error: Bind to port 22 on :: failed: Address already in use.
Nov 24 13:59:10 dlna sshd[35213]: fatal: Cannot bind any address.
Nov 24 13:59:10 dlna systemd[1]: ssh.service: Main process exited, code=exited, status=255/EXCEPTION
Nov 24 13:59:10 dlna systemd[1]: ssh.service: Failed with result 'exit-code'.
Nov 24 13:59:10 dlna systemd[1]: Failed to start OpenBSD Secure Shell server.

MichaIng, 2022-11-24:
Ah of course, Dropbear is an SSH server as well which uses the same port. Didn't you switch to OpenSSH via `dietpi-software` selection? There Dropbear would be uninstalled automatically to avoid such conflict.

MichaIng, 2023-02-12:
Did you solve the issue? If you aim to use Dropbear, you can purge OpenSSH:
```sh
apt autopurge openssh
```

MichaIng, 2023-03-18:
I'll mark this as closed. Feel free to reopen if the issue persists.

closed 2023-03-18 by MichaIng