MichaIng/DietPi #5916: How to credit you?

DrCWO, 2022-11-24:
**Thank you so much for DietPi and your great support**

I finally succeeded migrating my [rooExtend Software](http://diy.rooextend.com) from Raspbian to DietPi.

I like to credit DiePi on the HTML settings page of rooExtend.
What did you like me to write? Here examples what I wrote for others:

Roon
Copyright © 2016 Roon Labs LLC

[node-roon-api](https://github.com/RoonLabs/node-roon-api)
Copyright © 2016 Roon Labs LLC

[UPnPBridge-1.45.3](https://sourceforge.net/projects/lms-plugins-philippe44/files/)
Copyright © [philippe44ca](https://sourceforge.net/u/philippe44ca/profile/)

[rocket-nuimo-node](https://github.com/happycodelucky/rocket-nuimo-node)
Copyright © [Paul Bates](https://github.com/happycodelucky)

Best DrCWO

MichaIng, 2022-11-24:
Interesting software and also interesting platform to sell code/software 🙂.

Please use
> [DietPi](https://github.com/MichaIng/DietPi)
> Copyrigh @ [DietPi contributors](https://github.com/MichaIng/DietPi/graphs/contributors)

DrCWO, 2022-11-24:
Thanks, will be done with next release quite soon 👍

closed 2025-07-18 by MichaIng