MichaIng/DietPi #5918: ASUS Tinker Board images broken

snoopen, 2022-11-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | v8.11.2 (possibly all u-boot versions)
- Distro version | Bullseye
- Kernel version | N/A
- SBC model | ASUS Tinker Board
- Power supply used | 5V 2.1A
- SD card used | SanDisk ultra

#### Steps to reproduce
1. Try to boot DietPi on Tinker Board

#### Expected behaviour
Should boot into DietPi

#### Actual behaviour
Boot error: Ramdisk is corrupt or invalid

#### Extra details
uInitrd image type states AArch64 but I think it should be ARM
the u-boot convert step doesn't `sed` arm64 to arm for Tinker Board and possibly other 32bit boards

MichaIng, 2022-11-25:
This has been fixed a few hours ago. Check whether you tried it with the latest image with today's timestamp.

snoopen, 2022-11-26:
Awesome, it works now. Thank you!

MichaIng, 2022-11-27:
Great, I'll mark this issue as closed then.

closed 2022-11-27 by MichaIng