MichaIng/DietPi #5919: DietPi-Software | myMPD: Service is not running after update to dietpi v8.11.2

sofad, 2022-11-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=11
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- buster 1
- Kernel version | `uname -a`
- Linux ZeroPi2 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- RPi Zero W (armv6l)
- Power supply used | (EG: 5V 1A RAVpower)
- 5V 1A USB
- SD card used | (EG: SanDisk ultra)
- Samsung EVO 

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
mympd
- Was the software title installed freshly or updated/migrated?
updated with dietpi-update to v8.11.2
- Can this issue be replicated on a fresh installation of DietPi?
yes, same on a similar second pizero

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. reboot / boot pi
2. try to open http://local-ip:1333

#### Expected behaviour

- mympd webpage should open in browser

#### Actual behaviour

- "ERR_CONNECTION_REFUSED"

#### Extra details

**journalctl -u mympd**
-- Logs begin at Fri 2022-11-25 14:42:48 CET, end at Fri 2022-11-25 14:44:16 CET. --
Nov 25 14:43:22 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:15: Unknown lvalue 'ProtectClock' in section 'Service', ignoring
Nov 25 14:43:22 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:18: Unknown lvalue 'ProtectHostname' in section 'Service', ignoring
Nov 25 14:43:22 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:19: Unknown lvalue 'ProtectKernelLogs' in section 'Service', ignoring
Nov 25 14:43:22 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:22: Unknown lvalue 'ProtectProc' in section 'Service', ignoring
Nov 25 14:43:23 ZeroPi2 systemd[1]: Started myMPD server daemon.
Nov 25 14:43:24 ZeroPi2 mympd[407]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Nov 25 14:43:24 ZeroPi2 mympd[407]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Nov 25 14:43:24 ZeroPi2 mympd[407]: ERROR    mympd     File exists
Nov 25 14:43:24 ZeroPi2 systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILURE
Nov 25 14:43:24 ZeroPi2 systemd[1]: mympd.service: Failed with result 'exit-code'.

**ss -tulpn | grep LISTEN**
tcp     LISTEN   0        64               0.0.0.0:38255          0.0.0.0:*                                                                                     
tcp     LISTEN   0        128              0.0.0.0:53775          0.0.0.0:*      users:(("rpc.mountd",pid=383,fd=17))                                           
tcp     LISTEN   0        128              0.0.0.0:49967          0.0.0.0:*      users:(("rpc.mountd",pid=383,fd=9))                                            
tcp     LISTEN   0        128              0.0.0.0:111            0.0.0.0:*      users:(("rpcbind",pid=185,fd=4),("systemd",pid=1,fd=31))                       
tcp     LISTEN   0        128              0.0.0.0:22             0.0.0.0:*      users:(("sshd",pid=313,fd=3))                                                  
tcp     LISTEN   0        5                0.0.0.0:631            0.0.0.0:*      users:(("cupsd",pid=410,fd=6))                                                 
tcp     LISTEN   0        255              0.0.0.0:7000           0.0.0.0:*      users:(("shairport-sync",pid=404,fd=6))                                        
tcp     LISTEN   0        50               0.0.0.0:445            0.0.0.0:*      users:(("smbd",pid=369,fd=31))                                                 
tcp     LISTEN   0        64               0.0.0.0:2049           0.0.0.0:*                                                                                     
tcp     LISTEN   0        128              0.0.0.0:48673          0.0.0.0:*      users:(("rpc.mountd",pid=383,fd=13))                                           
tcp     LISTEN   0        5              127.0.0.1:6600           0.0.0.0:*      users:(("mpd",pid=399,fd=11))                                                  
tcp     LISTEN   0        50               0.0.0.0:139            0.0.0.0:*      users:(("smbd",pid=369,fd=32))                                                 
tcp     LISTEN   0        128                 [::]:111               [::]:*      users:(("rpcbind",pid=185,fd=6),("systemd",pid=1,fd=33))                       
tcp     LISTEN   0        128                    *:21                   *:*      users:(("proftpd",pid=362,fd=0))                                               
tcp     LISTEN   0        128                 [::]:22                [::]:*      users:(("sshd",pid=313,fd=4))                                                  
tcp     LISTEN   0        255                 [::]:7000              [::]:*      users:(("shairport-sync",pid=404,fd=7))                                        
tcp     LISTEN   0        64                  [::]:40155             [::]:*                                                                                     
tcp     LISTEN   0        50                  [::]:445               [::]:*      users:(("smbd",pid=369,fd=29))                                                 
tcp     LISTEN   0        128                 [::]:39551             [::]:*      users:(("rpc.mountd",pid=383,fd=11))                                           
tcp     LISTEN   0        64                  [::]:2049              [::]:*                                                                                     
tcp     LISTEN   0        128                 [::]:56229             [::]:*      users:(("rpc.mountd",pid=383,fd=19))                                           
tcp     LISTEN   0        128                 [::]:40773             [::]:*      users:(("rpc.mountd",pid=383,fd=15))                                           
tcp     LISTEN   0        5                  [::1]:6600              [::]:*      users:(("mpd",pid=399,fd=10))                                                  
tcp     LISTEN   0        50                  [::]:139               [::]:*      users:(("smbd",pid=369,fd=30))                                                 

i tried to reinstall mympd
dietpi-software reinstall 148
-> no success

I'm happy to provide additional informations, thank you.

MichaIng, 2022-11-25:
Please try to run:
```sh
G_EXEC mympd -c
systemctl restart mympd
```
myMPD v8.0.0 changed the way/path where configs are changed, but we do neither grant the `mympd` user permissions to grant those new configs itself, nor do we force a generation of these new configs on reinstall (if `/var/lib/mysql` exists already). I'll check how default permissions of the package are, I guess it makes sense to chown `/var/lib/mysql` completely to `mysql` user.

sofad, 2022-11-25:
Hi MichaIng, i followed your suggestion:

**1. sudo mympd -c**
15:16:11 NOTICE   mympd     Work dir: "/var/lib/mympd"
15:16:11 NOTICE   mympd     Config dir: "/var/lib/mympd/config"
15:16:11 NOTICE   mympd     Cache dir: "/var/cache/mympd"
Created myMPD config and exit
Exiting gracefully, thank you for using myMPD

**2. sudo systemctl restart mympd**

**3. Browser: http://local-ip:1333**
-> ERR_CONNECTION_REFUSED

MichaIng, 2022-11-25:
```sh
journalctl -u mympd
```

sofad, 2022-11-25:
**journalctl -u mympd**
-- Logs begin at Fri 2022-11-25 15:13:03 CET, end at Fri 2022-11-25 15:20:46 CET. --
Nov 25 15:13:37 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:15: Unknown lvalue
 'ProtectClock' in section 'Service', ignoring
Nov 25 15:13:37 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:18: Unknown lvalue
 'ProtectHostname' in section 'Service', ignoring
Nov 25 15:13:37 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:19: Unknown lvalue
 'ProtectKernelLogs' in section 'Service', ignoring
Nov 25 15:13:37 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:22: Unknown lvalue
 'ProtectProc' in section 'Service', ignoring
Nov 25 15:13:37 ZeroPi2 systemd[1]: Started myMPD server daemon.
Nov 25 15:13:39 ZeroPi2 mympd[406]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Nov 25 15:13:39 ZeroPi2 mympd[406]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Nov 25 15:13:39 ZeroPi2 mympd[406]: ERROR    mympd     File exists
Nov 25 15:13:39 ZeroPi2 systemd[1]: mympd.service: Main process exited, code=exited, sta
tus=1/FAILURE
Nov 25 15:13:39 ZeroPi2 systemd[1]: mympd.service: Failed with result 'exit-code'.
Nov 25 15:20:45 ZeroPi2 systemd[1]: Started myMPD server daemon.
Nov 25 15:20:46 ZeroPi2 mympd[768]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Nov 25 15:20:46 ZeroPi2 mympd[768]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Nov 25 15:20:46 ZeroPi2 mympd[768]: ERROR    mympd     File exists
Nov 25 15:20:46 ZeroPi2 systemd[1]: mympd.service: Main process exited, code=exited, sta
tus=1/FAILURE
Nov 25 15:20:46 ZeroPi2 systemd[1]: mympd.service: Failed with result 'exit-code'.

MichaIng, 2022-11-25:
Above it shows it being created, below it shows it's missing, strange. Probably it must be owned by `mympd` now:
```sh
ls -al /var/lib/mympd
chown -R mympd /var/lib/mympd
systemctl restart mympd
```

sofad, 2022-11-25:
ls -al /var/lib/mympd
lrwxrwxrwx 1 root root 13 Nov 25 12:52 /var/lib/mympd -> private/mympd

chown -R mympd /var/lib/mympd
systemctl restart mympd

=
**journalctl -u mympd**
-- Logs begin at Fri 2022-11-25 15:13:03 CET, end at Fri 2022-11-25 15:25:12 CET. --
Nov 25 15:13:37 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:15: Unknown lvalue 'ProtectCl
ock' in section 'Service', ignoring
Nov 25 15:13:37 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:18: Unknown lvalue 'ProtectHo
stname' in section 'Service', ignoring
Nov 25 15:13:37 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:19: Unknown lvalue 'ProtectKe
rnelLogs' in section 'Service', ignoring
Nov 25 15:13:37 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:22: Unknown lvalue 'ProtectPr
oc' in section 'Service', ignoring
Nov 25 15:13:37 ZeroPi2 systemd[1]: Started myMPD server daemon.
Nov 25 15:13:39 ZeroPi2 mympd[406]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Nov 25 15:13:39 ZeroPi2 mympd[406]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Nov 25 15:13:39 ZeroPi2 mympd[406]: ERROR    mympd     File exists
Nov 25 15:13:39 ZeroPi2 systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILU
RE
Nov 25 15:13:39 ZeroPi2 systemd[1]: mympd.service: Failed with result 'exit-code'.
Nov 25 15:20:45 ZeroPi2 systemd[1]: Started myMPD server daemon.
Nov 25 15:20:46 ZeroPi2 mympd[768]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Nov 25 15:20:46 ZeroPi2 mympd[768]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Nov 25 15:20:46 ZeroPi2 mympd[768]: ERROR    mympd     File exists
Nov 25 15:20:46 ZeroPi2 systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILU
RE
Nov 25 15:20:46 ZeroPi2 systemd[1]: mympd.service: Failed with result 'exit-code'.
Nov 25 15:24:32 ZeroPi2 systemd[1]: Started myMPD server daemon.
Nov 25 15:24:32 ZeroPi2 mympd[778]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Nov 25 15:24:32 ZeroPi2 mympd[778]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Nov 25 15:24:32 ZeroPi2 mympd[778]: ERROR    mympd     File exists
Nov 25 15:24:32 ZeroPi2 systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILU
RE
Nov 25 15:24:32 ZeroPi2 systemd[1]: mympd.service: Failed with result 'exit-code'.
Nov 25 15:25:12 ZeroPi2 systemd[1]: Started myMPD server daemon.
Nov 25 15:25:12 ZeroPi2 mympd[783]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Nov 25 15:25:12 ZeroPi2 mympd[783]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Nov 25 15:25:12 ZeroPi2 mympd[783]: ERROR    mympd     File exists
Nov 25 15:25:12 ZeroPi2 systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILU
RE
Nov 25 15:25:12 ZeroPi2 systemd[1]: mympd.service: Failed with result 'exit-code'.
root@ZeroPi2:~#

MichaIng, 2022-11-25:
```
/var/lib/mympd -> private/mympd
```
A symlink, this seems to be the issue:
```sh
rm /var/lib/mympd
mv /var/lib/private/mympd /var/lib/mympd
systemctl restart mympd
```
Why ever this was created, but it makes sense that only root had access to the `private` dir. Probaly now done automatically when running `mysqld -c` as root.

sofad, 2022-11-25:
root@ZeroPi2:~# rm /var/lib/mympd
root@ZeroPi2:~# mv /var/lib/private/mympd /var/lib/mympd
root@ZeroPi2:~# systemctl restart mympd
_________
root@ZeroPi2:~# journalctl -u mympd
-- Logs begin at Thu 2019-02-14 11:12:01 CET, end at Fri 2022-11-25 15:42:00 CET. --
Nov 25 15:37:10 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:15: Unknown lvalue 'ProtectCl
ock' in section 'Service', ignoring
Nov 25 15:37:10 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:18: Unknown lvalue 'ProtectHo
stname' in section 'Service', ignoring
Nov 25 15:37:10 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:19: Unknown lvalue 'ProtectKe
rnelLogs' in section 'Service', ignoring
Nov 25 15:37:10 ZeroPi2 systemd[1]: /lib/systemd/system/mympd.service:22: Unknown lvalue 'ProtectPr
oc' in section 'Service', ignoring
Nov 25 15:37:11 ZeroPi2 systemd[1]: Started myMPD server daemon.
Nov 25 15:37:12 ZeroPi2 mympd[444]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Nov 25 15:37:12 ZeroPi2 mympd[444]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Nov 25 15:37:12 ZeroPi2 mympd[444]: ERROR    mympd     File exists
Nov 25 15:37:12 ZeroPi2 systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILU
RE
Nov 25 15:37:12 ZeroPi2 systemd[1]: mympd.service: Failed with result 'exit-code'.
Nov 25 15:38:26 ZeroPi2 systemd[1]: Started myMPD server daemon.
Nov 25 15:38:26 ZeroPi2 mympd[676]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Nov 25 15:38:26 ZeroPi2 mympd[676]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Nov 25 15:38:26 ZeroPi2 mympd[676]: ERROR    mympd     File exists
Nov 25 15:38:26 ZeroPi2 systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILU
RE
Nov 25 15:38:26 ZeroPi2 systemd[1]: mympd.service: Failed with result 'exit-code'.
Nov 25 15:42:00 ZeroPi2 systemd[1]: Started myMPD server daemon.
Nov 25 15:42:00 ZeroPi2 mympd[692]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Nov 25 15:42:00 ZeroPi2 mympd[692]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Nov 25 15:42:00 ZeroPi2 mympd[692]: ERROR    mympd     File exists
Nov 25 15:42:00 ZeroPi2 systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILU
RE
Nov 25 15:42:00 ZeroPi2 systemd[1]: mympd.service: Failed with result 'exit-code'.
root@ZeroPi2:~# 
____________

ls -al /var/lib/mympd
lrwxrwxrwx 1 root root 13 Nov 25 15:38 /var/lib/mympd -> private/mympd

MichaIng, 2022-11-25:
The symlink was recreated? Last attempt:
```
chown -R mympd /var/lib/private/mympd

sofad, 2022-11-25:
```
root@ZeroPi2:~# chown -R mympd /var/lib/private/mympd
root@ZeroPi2:~# rm /var/lib/mympd
root@ZeroPi2:~# mv /var/lib/private/mympd /var/lib/mympd
root@ZeroPi2:~# systemctl restart mympd

```

brilliant! it's working!

Thank you very much

MichaIng, 2022-11-25:
Okay great, so most important I think we need to own the whole data dir to `mympd` user. No idea how/why the dir is moved to private. Probably systemd does this automatically for root owned dirs in /var/lib. I'll test and fix this on Sunday.

MichaIng, 2022-11-25:
Ah, this recent change most likely is the reason: https://github.com/jcorporation/myMPD/commit/fcf667cbe9945ac4ec91d04129d30bcb466733be

Previously the service started as root (and forked of as mympd user, I guess) and now it starts as mympd user directly + set as "dynamic" user: https://www.freedesktop.org/software/systemd/man/systemd.exec.html#DynamicUser=
Shouldn't have an effect since the user/group exists already, created during package postinst. This however might have to do with the private dir move.

MichaIng, 2022-12-02:
PR up to solve this issue: #5944
On next DietPi update, the `mympd` user is removed, in which case systemd automatically `chown`s the whole `/var/lib/mympd` directory to the dynamic user.

closed 2022-12-04 by MichaIng