MichaIng/DietPi #5920: Rolling Release-esque functionality

mrbluecoat, 2022-11-25:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
Always have Linux up-to-date without manual intervention

#### Describe the solution you'd like:
An automated unattended process to update DietPi daily without triggering prompts

#### Describe alternatives you've considered:
1. Set Armbian apt sources to `testing` and configure `unattended-upgrades`
2. Cron job to run `dietpi-upgrade` daily

#### Additional context
Although the first option works, it's a bit of a hack and comes with the bloat of unnecessary Debian/Armbian packages. DietPi is beautifully lean but the upgrades often spawn blocking prompts. For use cases, like display kiosks, where very limited functionality is needed, is there a way to have a "rolling release" type experience with DietPi?

Joulinar, 2022-11-25:
`apt` packages can be set to automatic updates already. For `dietpi-update` we don't offer such feature by purpose.

ravenclaw900, 2022-11-25:
Quoting @MichaIng from https://github.com/MichaIng/DietPi/pull/3899#issuecomment-725630693
> Automatic DietPi updates is not something I would do anymore. We do dietpi-software reinstalls and inform users about important changes which both should be recognised interactively, some reinstalls even require interactive input to not hang.

mrbluecoat, 2022-11-25:
Got it. Thanks for the quick replies!

closed 2022-11-25 by mrbluecoat

MichaIng, 2022-11-25:
For automatic APT upgrades, check `/boot/dietpi.txt` where you can enable them, which includes kernel upgrades. You can use our Bookworm images at https://dietpi.com/downloads/images/ which is currently Debian testing. When you then change the sources to testing, it won't result in any additional upgrade but switch to Debian 13/Trixie automatically next summer, when Bookworm becomes stable release, as sort of rolling release character.

The DietPi scripts themselves have not really much effect on normal operation but are for interactive system configuration and software installs. So it wouldn't give you much benefit anyway if updating those was possible non-interactively. You can run the DietPi update before you do `dietpi-software` installs or other stuff on console.