MichaIng/DietPi #5921: Chromium doesn't fill the screen anymore

DrCWO, 2022-11-26:
### Creating a bug report/issue

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version
bullseye 0

- Kernel version 
Linux rooExtend 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

- SBC model 
RPi 4 Model B (aarch64)

- Power supply used
5V 15W

- SD card used
SanDisk ultra

#### Additional Information (if applicable)
TV is connected to PI via HDMI
I run chromium as I did before on Raspbian. 

Here the start script of openbox
```
xset s off
xset s noblank
xset -dpms
setxkbmap -option terminate:ctrl_alt_bksp
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' ~/.config/chromium/'Local State'
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/; s/"exit_type":"[^"]+"/"exit_type":"Normal"/' ~/.config/chromium/Default/Preferences
chromium-browser --user-agent="develop (develop) develop (develop) Chromium/74.0.3729.157" --disable-infobars --enable-gpu-rasterization --enable-native-gpu-memory-buffers --enable-oop-rasterization --enable-zero-copy --ignore-gpu-blocklis --enable-accelerated-2d-canvas --no-sandbox --incognito --start-maximized --kiosk http://192.168.0.247:9330/display/
```

Here output of 'ps' after started: 

```
root        1216       1  0 Nov25 ?        00:00:00 /bin/sh /usr/bin/startx -- -nocursor
root        1272    1216  0 Nov25 ?        00:00:00 xinit /etc/X11/xinit/xinitrc -- /usr/bin/X :0 -nocursor -auth /tmp/serverauth.qGBK61PJSX
root        1273    1272  0 Nov25 tty2     00:00:02 /usr/lib/xorg/Xorg :0 -nocursor -auth /tmp/serverauth.qGBK61PJSX

root        1389    1272  0 Nov25 ?        00:00:00 /usr/bin/openbox --startup /usr/lib/aarch64-linux-gnu/openbox-autostart OPENBOX
root        1406    1284  0 Nov25 ?        00:02:18 RAATServer --debug --gc=sgen --server RAATServer.exe
root        1434    1389  0 Nov25 ?        00:00:00 /usr/bin/ssh-agent x-session-manager
root        1440    1389  0 Nov25 ?        00:00:00 /bin/sh /usr/lib/aarch64-linux-gnu/openbox-autostart OPENBOX
root        1444    1440  0 Nov25 ?        00:00:00 sh /etc/xdg/openbox/autostart
root        1455    1444  0 Nov25 ?        00:01:13 /usr/lib/chromium-browser/chromium-browser --enable-pinch --user-agent=develop (develop) develop (develop) Chromium/74.0.3729.157 --disable-infobars --enable-gpu-rasterization --enable-native-gpu-memory-buffers --enable-oop-rasterization --enable-zero-copy --ignore-gpu-blocklis --enable-accelerated-2d-canvas --no-sandbox --incognito --start-maximized --kiosk http://192.168.0.247:9330/display/
root        1492       1  0 Nov25 ?        00:00:00 /usr/lib/chromium-browser/chrome_crashpad_handler --monitor-self --monitor-self-annotation=ptype=crashpad-handler --database=/root/.config/chromium/Crash Reports --annotation=channel=Built on Debian , running on   --annotation=lsb-release=Debian GNU/Linux 11 (bullseye) --annotation=plat=Linux --annotation=prod=Chrome_Linux --annotation=ver=104.0.5112.105 --initial-client-fd=5 --shared-client-connection
root        1495       1  0 Nov25 ?        00:00:00 /usr/lib/chromium-browser/chrome_crashpad_handler --no-periodic-tasks --monitor-self-annotation=ptype=crashpad-handler --database=/root/.config/chromium/Crash Reports --annotation=channel=Built on Debian , running on   --annotation=lsb-release=Debian GNU/Linux 11 (bullseye) --annotation=plat=Linux --annotation=prod=Chrome_Linux --annotation=ver=104.0.5112.105 --initial-client-fd=4 --shared-client-connection
root        1497    1455  0 Nov25 ?        00:00:00 /usr/lib/chromium-browser/chromium-browser --type=zygote --no-zygote-sandbox --no-sandbox --enable-crashpad --crashpad-handler-pid=0 --enable-crash-reporter=,Built on Debian , running on --user-agent=develop (develop) develop (develop) Chromium/74.0.3729.157 --change-stack-guard-on-fork=enable --enable-crashpad
root        1498    1455  0 Nov25 ?        00:00:00 /usr/lib/chromium-browser/chromium-browser --type=zygote --no-sandbox --enable-crashpad --crashpad-handler-pid=0 --enable-crash-reporter=,Built on Debian , running on --user-agent=develop (develop) develop (develop) Chromium/74.0.3729.157 --change-stack-guard-on-fork=enable --enable-crashpad
root        1551    1455  0 Nov25 ?        00:01:44 /usr/lib/chromium-browser/chromium-browser --type=utility --utility-sub-type=network.mojom.NetworkService --lang=en-US --service-sandbox-type=none --no-sandbox --enable-crashpad --crashpad-handler-pid=0 --enable-crash-reporter=,Built on Debian , running on --user-agent=develop (develop) develop (develop) Chromium/74.0.3729.157 --change-stack-guard-on-fork=enable --shared-files=v8_context_snapshot_data:100 --field-trial-handle=0,i,10073283477392839010,1457462670877906476,131072 --disable-features=EventPath --enable-crashpad
root        1563    1498  0 Nov25 ?        00:00:05 /usr/lib/chromium-browser/chromium-browser --type=utility --utility-sub-type=storage.mojom.StorageService --lang=en-US --service-sandbox-type=utility --no-sandbox --enable-crashpad --crashpad-handler-pid=0 --enable-crash-reporter=,Built on Debian , running on --user-agent=develop (develop) develop (develop) Chromium/74.0.3729.157 --change-stack-guard-on-fork=enable --shared-files=v8_context_snapshot_data:100 --field-trial-handle=0,i,10073283477392839010,1457462670877906476,131072 --disable-features=EventPath
root        1599    1498  0 Nov25 ?        00:00:01 /usr/lib/chromium-browser/chromium-browser --type=renderer --enable-crashpad --crashpad-handler-pid=0 --enable-crash-reporter=,Built on Debian , running on --user-agent=develop (develop) develop (develop) Chromium/74.0.3729.157 --display-capture-permissions-policy-allowed --change-stack-guard-on-fork=enable --no-sandbox --disable-databases --lang=en-US --num-raster-threads=2 --enable-zero-copy --enable-main-frame-before-activation --renderer-client-id=6 --launch-time-ticks=42457264 --shared-files=v8_context_snapshot_data:100 --field-trial-handle=0,i,10073283477392839010,1457462670877906476,131072 --disable-features=EventPath
root        1600    1498  0 Nov25 ?        00:03:11 /usr/lib/chromium-browser/chromium-browser --type=renderer --enable-crashpad --crashpad-handler-pid=0 --enable-crash-reporter=,Built on Debian , running on --user-agent=develop (develop) develop (develop) Chromium/74.0.3729.157 --display-capture-permissions-policy-allowed --change-stack-guard-on-fork=enable --no-sandbox --disable-databases --lang=en-US --num-raster-threads=2 --enable-zero-copy --enable-main-frame-before-activation --renderer-client-id=5 --launch-time-ticks=42463578 --shared-files=v8_context_snapshot_data:100 --field-trial-handle=0,i,10073283477392839010,1457462670877906476,131072 --disable-features=EventPath
root        1646    1497  0 Nov25 ?        00:00:24 /usr/lib/chromium-browser/chromium-browser --type=gpu-process --no-sandbox --enable-gpu-rasterization --enable-crashpad --crashpad-handler-pid=0 --enable-crash-reporter=,Built on Debian , running on --user-agent=develop (develop) develop (develop) Chromium/74.0.3729.157 --change-stack-guard-on-fork=enable --gpu-preferences=WAAAAAAAAAAgAAAIAAAAAAAAAAAAAAAAAABgAAAAAAA4AAAAAAAAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAACAAAAAAAAAAIAAAAAAAAAABAAAAAAAAAAgAAAAAAAAACAAAAAAAAAAIAAAAAAAAAA== --use-gl=angle --use-angle=swiftshader-webgl --shared-files --field-trial-handle=0,i,10073283477392839010,1457462670877906476,131072 --disable-features=EventPath
```

**With Raspbian the complete screen was used by Chromium, using DietPi I see black stripes right and left on the screen.**

#### Steps to reproduce

#### Expected behaviour
No stripes right and left on screen

#### Actual behaviour
Stripes right and left on screen

#### Extra details
See attached image
[image]

Joulinar, 2022-11-26:
Did you adjust resolution on `dietpi.txt`?

DrCWO, 2022-11-26:
Made no changes on /boot/dietpi.txt.

Here it is..
```
# IMPORTANT:
# - This is intended for advanced users, unless you know what you are doing, do not edit this file. Please use the DietPi programs instead.
# - Do not remove uncommented lines, as the items are scraped by DietPi programs, on demand.

#------------------------------------------------------------------------------------------------------
##### DietPi-Automation settings, applied on first boot of DietPi only, ONCE! #####
#------------------------------------------------------------------------------------------------------

##### Language/Regional options #####
# Locale: e.g.: "en_GB.UTF-8" / "de_DE.UTF-8" | One entry and UTF-8 ONLY!
AUTO_SETUP_LOCALE=C.UTF-8

# Keyboard layout e.g.: "gb" / "us" / "de" / "fr"
AUTO_SETUP_KEYBOARD_LAYOUT=gb

# Time zone e.g.: "Europe/London" / "America/New_York" | Full list: https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
AUTO_SETUP_TIMEZONE=Europe/London

##### Network options #####
# Enable Ethernet or WiFi adapter: 1=enable | 0=disable
# - If both Ethernet and WiFi are enabled, WiFi will take priority and Ethernet will be disabled.
# - If using WiFi, please edit dietpi-wifi.txt to pre-enter credentials.
AUTO_SETUP_NET_ETHERNET_ENABLED=1
AUTO_SETUP_NET_WIFI_ENABLED=0

# Force Ethernet speeds: 0=automatic speed | 10 = 10 Mbit/s | 100 = 100 Mbit/s etc.
# - Use this when your Ethernet adapter has an unstable 1 Gbit/s link.
AUTO_SETUP_NET_ETH_FORCE_SPEED=0

# WiFi country code: 2 uppercase character value (e.g. GB US DE JP): https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
AUTO_SETUP_NET_WIFI_COUNTRY_CODE=GB

# Enter your static network details below, if applicable.
AUTO_SETUP_NET_USESTATIC=0
AUTO_SETUP_NET_STATIC_IP=192.168.0.100
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.0.1
AUTO_SETUP_NET_STATIC_DNS=9.9.9.9 149.112.112.112

# Set to "1" to convert DHCP leased network settings into static settings automatically on first boot.
AUTO_SETUP_DHCP_TO_STATIC=0

# Hostname
AUTO_SETUP_NET_HOSTNAME=DietPi

# Delay service starts at boot until network is established: 0=disabled | 1=enabled
AUTO_SETUP_BOOT_WAIT_FOR_NETWORK=1

##### Misc options #####
# Swap space size to generate: 0 => disable | 1 => auto | 2 and up => size in MiB
AUTO_SETUP_SWAPFILE_SIZE=0
# Swap space location: "zram" => swap space on /dev/zram0 (auto-size = 50% of RAM size) | /path/to/file => swap file at location (auto-size = 2 GiB minus RAM size)
AUTO_SETUP_SWAPFILE_LOCATION=/var/swap

# Set to "1" to disable HDMI/video output and framebuffers on Raspberry Pi, to reduce power consumption and memory usage: Works on RPi only!
AUTO_SETUP_HEADLESS=0

# Unmask (enable) systemd-logind service (including dbus), which is masked by default on DietPi
AUTO_UNMASK_LOGIND=0

# Custom Script (pre-networking and pre-DietPi install)
# - Allows you to automatically execute a custom script before network is up on first boot.
# - Copy your script to /boot/Automation_Custom_PreScript.sh and it will be executed automatically.
# - Executed script log: /var/tmp/dietpi/logs/dietpi-automation_custom_prescript.log

# Custom Script (post-networking and post-DietPi install)
[... 505 lines cut ...]
SOFTWARE_K3S_EXEC=

# DietPi-Dashboard
# Version to use
# - Stable = Use release version of DietPi-Dashboard.
# - Nightly = Use unstable version DietPi-Dashboard. Might have bugs, but will probably have more features.
SOFTWARE_DIETPI_DASHBOARD_VERSION=Stable
# Whether to only install backend or not
SOFTWARE_DIETPI_DASHBOARD_BACKEND=0

# PiVPN
# - For an unattended install, place a config file named "unattended_pivpn.conf" into the boot partition/directory.
# - For example configs, have a look at: https://github.com/pivpn/pivpn/tree/master/examples

#------------------------------------------------------------------------------------------------------
##### Dev settings #####
#------------------------------------------------------------------------------------------------------
DEV_GITBRANCH=master
DEV_GITOWNER=MichaIng

#------------------------------------------------------------------------------------------------------
##### Settings, automatically added by dietpi-update #####
#------------------------------------------------------------------------------------------------------

```

DrCWO, 2022-11-26:
Chromium wasn't installed via dietpi-software. 
I did this:
```
apt-get install --no-install-recommends xserver-xorg x11-xserver-utils xinit openbox -y
apt-get install --no-install-recommends chromium-browser -y

```

Joulinar, 2022-11-26:
Any specific reason for this? We have a dedicated option inside our configuration file to set correct resolution kiosk mode.

```
# Chromium
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=720
SOFTWARE_CHROMIUM_AUTOSTART_URL=https://dietpi.com/
```

DrCWO, 2022-11-26:
Just for clarification: 
Resolution of the TV is Standard-UHD, 3840 × 2160 pixel

DrCWO, 2022-11-26:
You saw I run with openbox and a special configuration file I made for Raspbian that I just ported.
Do these settings have also effect on my setup not using dietpi-Software?
What happens if I omit them?
```
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=720
```

Joulinar, 2022-11-26:
They are effective on our own start script only.

MichaIng, 2022-11-27:
Did you enable KMS, i.e. is `dtoverlay=vc4-kms-v3d` present in `/boot/config.txt`?

There are also certain flags for defining/maximising window size: https://peter.sh/experiments/chromium-command-line-switches/

DrCWO, 2022-11-28:
Thnak you. I did not yet enable KMS, i.e. is dtoverlay=vc4-kms-v3d in /boot/config.txt?
Would you please tell me what it is for? Al I read about it I did not understand...

I read this comment and asked my customer to test it...

My UHD TV came with a default setting, I think it was something like 19:10. I tried a few settings and one that said 'Just Scan' got rid of the problem with either a black border, or the panel being OFF the screen. None of my messing with config.txt in the week or so before I thought to look there helped at all.

MichaIng, 2022-11-28:
KMS/DRM is the modern Linux graphics driver, deprecating the RPi-only legacy graphics. But actually, it must have been enabled already, since otherwise, AFAIK Chromium wouldn't start in the first place. However, please verify that a line `dtoverlay=vc4-kms-v3d` is present in the `/boot/config.txt` file.

DrCWO, 2022-11-28:
Thx, I'll try but this week I have to travel..

DrCWO, 2022-12-20:
Hi,
late feedback but issue solved 👍 

I added the line below. It is "**f**kms" not "kms"!

`dtoverlay=vc4-fkms-v3d`

to `/boot/config.txt` and UDH works great.

Thanks' again 👍 👍 👍

MichaIng, 2022-12-21:
Ah great. Not sure what "fake KMS" does differently in this regards compared to full KMS, but great that a solution was found and to keep this in mind.

closed 2022-12-21 by MichaIng

DrCWO, 2022-12-21:
Just a quick info.

Using KMS instead of FKMS the command

` tvservice -l -n`

to show the connected device doesn't work any more that's why I used 

`dtoverlay=vc4-fkms-v3d`

MichaIng, 2022-12-22:
Ah, I thought filling the screen did only work with fKMS. Yes, `tvservice` indeed works with legacy driver or fKMS only.