MichaIng/DietPi #5925: Improvement: Keyboard layout vs. password change during initial setup

LeFish1, 2022-11-26:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
- For a first time user with a different keyboard layout than US/GB it can be "dangerous" to be suggested to change the global software and user passwords before going into the initial setup menu.
- I just had to install a second time as I set global software and user passwords as suggested by the installer, before switching to my native keyboard layout in the initial setup menu. Not a big deal, but unnecessary...

#### Describe the solution you'd like:
- (1) I would suggest to just inform the user of the recommended option to change the passwords before switching to the initials setup menu.
- OR better
- (2) It would be even better to be forced into keyboard layout setup before initial change of password just to make sure the charset is as expected.

#### Describe alternatives you've considered:
See two options above. I prefer option (2).

#### Additional context

- The context is the initial setup menu. In my case on a x64 PC-installation.

Thanks for this great OS!

MichaIng, 2022-11-27:
Agreed. At least when login is on ttyX (a local console), selecting the keyboard layout should be a mandatory first step.

MichaIng, 2023-03-27:
Another case: #6276
Giving this priority now for v8.16.

MichaIng, 2023-04-01:
Added for DietPi v8.16 with: https://github.com/MichaIng/DietPi/commit/0423238

closed 2023-04-01 by MichaIng