MichaIng/DietPi #5926: Radxa Zero not booting or working as spected (need help)

k1r0s, 2022-11-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | latest
- Distro version | unknown
- Kernel version | `uname -a`
- SBC model | radxa zero
- Power supply used | usbc likely 5v
- SD card used | emmc

Been following this guide in order to remove radxa emmc and then install the image over there. https://wiki.radxa.com/Zero/install/eMMC_erase

Then im able to detect the device as mass storage. Hence burn the image with dd

The radxa does not have a Gigabyte port but we can connect an ethernet gigabyte cable through a usb-c dock, I've did that, and the port display some lights while the radxa was booting (guessing I got an IP address)

however I was running a port scan in my local network `sudo nmap 192.168.1.0/24 22`

No device was found with port 22 open.

I've modified dietpi.txt settings to have a static IP address but also not found in the local net. 

Im clueless

Any steps that I've missed? ideas? Thx in advance

MichaIng, 2022-11-27:
Thanks for reporting.

Does it work when you boot from SD card?

k1r0s, 2022-11-28:
will reopen and provide more details, thx

closed 2022-11-28 by k1r0s

k1r0s, 2022-11-28:
fyi, the image works from emmc, my usb-c ethernet adapter was not working for some reason

MichaIng, 2022-11-28:
Thanks for the feedback.