MichaIng/DietPi #5927: NanoPi R1 not booting

eadmaster, 2022-11-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | unknown
- Distro version | Bookworm
- Kernel version | unknown
- SBC model | NanoPi R1 (512MB RAM+SD card ver.)
- Power supply used | Anker Power Port AK-A2056311
- SD card used | unbranded 8GB class 4

#### Additional Information (if applicable)

#### Steps to reproduce

1. downloaded [DietPi_NanoPiR1-ARMv7-Bookworm image](https://dietpi.com/downloads/images/DietPi_NanoPiR1-ARMv7-Bookworm.7z)
2. flashed with [USBImager](https://gitlab.com/bztsrc/usbimager/)
3. inserted the microSD into the NanoPi and turned on
4. the power/red led stay on for a few seconds and then turn off with the device.

#### Expected behaviour

- should stay powered on and connect via  ethernet 

#### Actual behaviour

- the NanoPi just turns off after a few seconds (failed boot?)

#### Extra details

I've noticed the official supported images have a different 3 partition  scheme instead of just 1.

Also tried with [DietPi_NanoPiR1-ARMv7-Bullseye](https://dietpi.com/downloads/images/DietPi_NanoPiR1-ARMv7-Bullseye.7z), but the archive seems corrupt:

ELDI-Steinar, 2022-11-28:
Tested https://dietpi.com/downloads/images/DietPi_NanoPiNEO-ARMv7-Bullseye.7z
FILE:   DietPi_NanoPiNEO-ARMv7-Bullseye.img
DATE:   Thu Nov 24 12:12:23 UTC 2022

NanoPi NEO - LTS will not boot from the SD card made with this img file. Older img file works ok.

MichaIng, 2022-11-28:
Thanks for reporting. Not sure how the archive could have been broken. No error in build logs. However, I triggered new builds, probably this fixes the Bookworm image as well.

MichaIng, 2022-11-28:
Both images updated. The Bullseye archive is now valid.

eadmaster, 2022-11-28:
> The Bullseye archive is now valid.

Just tried again, same results of the original Bookworm image: device does not boot, turns off after a few seconds.

MichaIng, 2022-11-28:
The original card was 8 GiB, right? 2 GiB is too small, at least swap file generation will fail on the 512 MiB RAM variant. And you did try with the Bullseye image, or the Bookworm image only?

And which Ethernet port did you use? Network will currently only work with one of them, since first run setup forcefully sets up the `eth0` interface, not the `eth1` interface. However, this wouldn't make it power off. Also I never faced any kind of boot failure which makes it power off, but only being stuck at some point, or not showing up on network 🤔.

You do not have a UART adapter to view serial console output, do you?

And could you try the Armbian Bullseye image? https://www.armbian.com/nanopi-r1/
That one uses same partitioning, (mainline) kernel and bootloader than ours.

eadmaster, 2022-11-28:
> And you did try with the Bullseye image, or the Bookworm image only?

Tried [this](https://dietpi.com/downloads/images/DietPi_NanoPiR1-ARMv7-Bullseye.7z).

> And which Ethernet port did you use? 

the one labelled as "WAN" is attached to the main router.

> You do not have a UART adapter to view serial console output, do you?

Yes, i have one, i may check the serial output if everything else fails.

> And could you try the Armbian Bullseye image? [armbian.com/nanopi-r1](https://www.armbian.com/nanopi-r1/)

Ok, just tried, it boot fine and i can ssh to it.

MichaIng, 2022-11-28:
Just to rule it out, could you also test to upgrade the Armbian kernel and flash new U-Boot and check whether it still boots fine?
```sh
apt update
apt full-upgrade
source /usr/lib/u-boot/platform_install.sh
setup_write_uboot_platform
write_uboot_platform "$DIR" "$DEVICE"
reboot
```

Otherwise, two things to test on our image:
1. In `/boot/dietpiEnv.txt` change `overlays=usbhost1 usbhost2 uart1` to `overlays=usbhost0 usbhost1 uart1`.
2. In `/boot/boot.cmd` remove `coherent_pool=2M` from the line which starts with `setenv bootargs`. But then also the `/boot/boot.scr` needs to be recompiled, which requires the `u-boot-tools` package on Debian/Ubuntu:
    ```sh
    sed -i 's/coherent_pool=2M //' /path/to/boot/boot.cmd
    apt install u-boot-tools
    mkimage -C none -A arm -T script -d /path/to/boot/boot.cmd /path/to/boot/boot.scr
    ```

eadmaster, 2022-11-28:
>>>>>
apt update
apt full-upgrade
source /usr/lib/u-boot/platform_install.sh
setup_write_uboot_platform
write_uboot_platform "$DIR" "$DEVICE"
reboot
>>>>

Seems very slow while running these commands, not sure if it is my class 4 microSD or the SoC itself, but Openwrt was running WAY faster.

I think i'm going to try this again with a faster SD card.

MichaIng, 2022-11-28:
You cannot compare with OpenWRT like this. None of these commands are available on OpenWRT, different package manager (if one at all?) etc. There was a kernel upgrade applied, which is a lot of data and files, so takes a while 😉.

eadmaster, 2022-11-29:
ok, i managed to complete the full-upgrade, flashed the new uboot, and now the system is no longer booting like before. So i guess there must be something wrong with the armbian upstream... maybe i could try an older Dietpi image that is still working?

MichaIng, 2022-11-29:
Nasty. Could you try it with this U-Boot (should match the one on Armbian image before it was upgraded): 
[u-boot-sunxi-with-spl.bin.gz](https://github.com/MichaIng/DietPi/files/10109176/u-boot-sunxi-with-spl.bin.gz)

Can be flashed form an external system, like that if the drive is `/dev/sdb`:
```sh
gzip -d u-boot-sunxi-with-spl.bin.gz
dd if=/dev/zero of=/dev/sdb bs=1k count=1023 seek=1 status=noxfer
dd if=u-boot-sunxi-with-spl.bin of=/dev/sdb bs=1024 seek=8 status=noxfer
```

ELDI-Steinar, 2022-11-29:
Tested https://dietpi.com/downloads/images/DietPi_NanoPiNEO-ARMv7-Bullseye.7z
FILE:   DietPi_NanoPiNEO-ARMv7-Bullseye.img
DATE:   Thu Nov 24 12:12:23 UTC 2022
MD5:    f786e130da28da167fec3581d2a5ed85

On my new NanoPi NEO - LTS with 8GB SD-card.
No problem booting Dietpi 8.11.2

eadmaster, 2022-11-30:
> Could you try it with this U-Boot (should match the one on Armbian image before it was upgraded):
[u-boot-sunxi-with-spl.bin.gz](https://github.com/MichaIng/DietPi/files/10109176/u-boot-sunxi-with-spl.bin.gz)

Yep, this did the trick, it's booting correctly now!

MichaIng, 2022-11-30:
Not sure whether this one is responsible or the solution: https://github.com/armbian/build/commit/39a083e
It's a future Armbian version, but not sure how the build script work. Interestingly, this previous `nanopi_r1_defconfig` U-Boot config never existed in upstream U-Boot. Not sure whether some patch has been dropped, which added this config.

However, for now I'll flash this older U-Boot as a workaround. We can test again when new U-Boot packages have been released.

MichaIng, 2022-11-30:
Images have been updated.

closed 2022-12-04 by MichaIng

ELDI-Steinar, 2025-07-21:
> @eadmaster<https://github.com/eadmaster> in case you still have the NanoPi R1 and find time, could you test whether it works with the latest U-Boot build I just pushed to our APT server?
> Warning: In case it does not work, you would need to flash the fixed binary from an external Linux system.
>
> apt update
>
> apt install linux-u-boot-nanopi-r1-current
>
> /boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
>
> reboot

***@***.***:~$ sudo apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://dietpi.com/apt bookworm InRelease
Hit:6 https://dietpi.com/apt all InRelease
Get:7 https://pkgs.tailscale.com/stable/debian bookworm InRelease
Fetched 6581 B in 6s (1043 B/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

***@***.***:~$ sudo apt install linux-u-boot-nanopi-r1-current
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-u-boot-nanopi-r1-current

***@***.***:~$ dpkg -l  | grep armbian-firmware
ii  armbian-firmware               25.08.0-trunk-dietpi1          all          Armbian - Linux firmware

***@***.***:~$ uname -r
6.6.44-current-sunxi

Best regards,
Steinar

ELDI-Steinar, 2025-07-23:
> @ELDI-Steinar<https://github.com/ELDI-Steinar>
> Hmm, somehow the wrong APT component seems to be used in your case, armbian-firmware being latest, but NanoPi R1 specific kernel and bootloader not available. Can you show this:
>
> cat /etc/apt/sources.list.d/dietpi.list
>
> echo "$G_HW_MODEL"
>
> cat /etc/.dietpi_hw_model_identifier
>
> @eadmaster<https://github.com/eadmaster>
>
> Can i just test the u-boot without updating all the system packages?
>
> Yes, the commands I provided update the bootloader package only, nothing else. But in general, I highly recommend to keep all system packages updated, as they contain security patches as well.

Hi,

I have 16 pcs. NanoPI NEO that run 32bit DietPi.

This is the device I checked last time (originally installed with DietPi and bookworm):
─────────────────────────────────────────────────────
DietPi v9.14.2 : 13:56 - Wed 07/23/25
─────────────────────────────────────────────────────
- Device model : NanoPi NEO (armv7l)
- CPU temp : 21 °C / 69 °F : Who put me in the freezer!

***@***.***:~$ dpkg -l  | grep armbian-firmware
ii  armbian-firmware               25.08.0-trunk-dietpi1          all          Armbian - Linux firmware

***@***.***:~$ uname -r
6.6.44-current-sunxi

***@***.***:~$ cat /etc/apt/sources.list.d/dietpi.list
deb https://dietpi.com/apt bookworm main
deb https://dietpi.com/apt all nanopineo

***@***.***:~$ echo "$G_HW_MODEL"
60

***@***.***:~$ cat /etc/.dietpi_hw_model_identifier
60

This is another device ((originally installed with DietPi and bookworm, but got a kernal upgrade from apt upgrade or dietpi-update at some point):
─────────────────────────────────────────────────────
DietPi v9.14.2 : 13:47 - Wed 07/23/25
─────────────────────────────────────────────────────
- Device model : NanoPi NEO (armv7l)
- CPU temp : 27 °C / 80 °F : Who put me in the freezer!

***@***.***:~$ dpkg -l  | grep armbian-firmware
ii  armbian-firmware               25.08.0-trunk-dietpi1          all          Armbian - Linux firmware

***@***.***:~$ uname -r
6.12.23-current-sunxi

***@***.***:~$ cat /etc/apt/sources.list.d/dietpi.list
deb https://dietpi.com/apt bookworm main
deb https://dietpi.com/apt all nanopineo

***@***.***:~$ echo "$G_HW_MODEL"
60

***@***.***:~$ cat /etc/.dietpi_hw_model_identifier
60

Best regards,
Steinar