MichaIng/DietPi #5931: No storage device is avaliable when using g_mass_storage in a Radxa Zero

kazuoteramoto, 2022-11-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=11
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
bullseye
```
- Kernel version | `uname -a`
```
Linux rhea 5.10.147-meson64 #22.08.4 SMP PREEMPT Fri Oct 7 16:37:19 UTC 2022 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
Radxa Zero (aarch64)
```
- Power supply used | (EG: 5V 1A RAVpower)
USB
- SD card used | (EG: SanDisk ultra)
None/emmc

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
not applicable
- Was the software title installed freshly or updated/migrated?
not applicable
- Can this issue be replicated on a fresh installation of DietPi?
Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
Fresh install of dietpi on a radxa zero, 4gb with 16gb of emmc. Usb cable connect to the OTG usb port

Follow this documentation https://wiki.radxa.com/Zero/dev/usb-mass-storage (adapting the paths)

#### Expected behaviour
I should see a new usb device appears on the computer connected to the usb

#### Actual behaviour
No device appears on the computer connected to the usb

#### Extra details
I got no error and the following is show in the `dmesg`

```
[  340.103701] Mass Storage Function, version: 2009/09/11
[  340.103717] LUN: removable file: (no medium)
[  340.104615] LUN: removable file: /home/dietpi/share
[  340.104620] Number of LUNs=1
[  340.104762] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[  340.104767] g_mass_storage gadget: userspace failed to provide iSerialNumber
[  340.104770] g_mass_storage gadget: g_mass_storage ready
[  340.107912] dwc2 ff400000.usb: bound driver g_mass_storage
```

There is zero messages on dmesg of the connected computer and nothing listed under lsblk or lsusb. 

I tried the same in a fresh installation using this image `radxa-zero-ubuntu-focal-server-arm64-20211120-1315-mbr.img.gz`  from https://github.com/radxa/radxa-zero-images-released/releases and by following the linked wiki the device appears on the computer connected to the usb as expected.

For archival purpose the main part of the linked wiki:
```
Enable USB mass storage on Radxa Zero

Step 0: Make sure that kernel version is at least version 5.10.69-6-amlogic. Otherwise, you can upgrade the kernel with the following commands:

rock@zero:~$ sudo apt update
rock@zero:~$ sudo apt install -y linux-headers-5.10.69-6-amlogic-g775a46783ad8
rock@zero:~$ sudo apt install -y linux-image-5.10.69-6-amlogic-g775a46783ad8

Step 1: Disable amlogic-adbd service. It will interfere with USB mass storage mode:

rock@zero:~$ sudo systemctl disable amlogic-adbd

Step 2: Reboot the device:

rock@zero:~$ sudo reboot

Step 3: Create a file backing for mass storage, and format it in FAT32:

rock@zero:~$ sudo dd if=/dev/zero of=/home/rock/share bs=1M count=512 status=progress # Create a 512M file as /home/rock/share
rock@zero:~$ sudo apt-get install dosfstools # dosfstools contains mkfs.fat
rock@zero:~$ sudo mkfs.fat /home/rock/share -F 32 -I

Step 4: Enable USB mass storage on our backing file and create test.txt file:

rock@zero:~$ sudo modprobe g_mass_storage file=/home/rock/share stall=0 removable=1
rock@zero:~$ sudo mount /home/rock/share /mnt/
rock@zero:~$ sudo touch  /mnt/test.txt
rock@zero:~$ echo "hello radxa" | sudo tee /mnt/test.txt
hello radxa
rock@zero:~$ cat /mnt/test.txt
hello radxa

```

MichaIng, 2022-12-02:
Thanks for reporting.

Not sure whether this is a limitation with the Armbian kernel. Did you try with both USB ports? Not sure how the kernel module determines the USB port to switch to gadget mode. There are further parameters, but I do not see one which would define the USB port:
```
parm:           idVendor:USB Vendor ID (ushort)
parm:           idProduct:USB Product ID (ushort)
parm:           bcdDevice:USB Device version (BCD) (ushort)
parm:           iSerialNumber:SerialNumber string (charp)
parm:           iManufacturer:USB Manufacturer string (charp)
parm:           iProduct:USB Product string (charp)
parm:           file:names of backing files or devices (array of charp)
parm:           ro:true to force read-only (array of bool)
parm:           removable:true to simulate removable media (array of bool)
parm:           cdrom:true to simulate CD-ROM instead of disk (array of bool)
parm:           nofua:true to ignore SCSI WRITE(10,12) FUA bit (array of bool)
parm:           luns:number of LUNs (uint)
parm:           stall:false to prevent bulk stalls (bool)
```
It is the same in your case?
```sh
modinfo g_mass_storage
```

kazuoteramoto, 2022-12-03:
From what I understand there is only one port that works with OTG, its the one closest to the corner.

Reading the forum I found that this as fixed by radxa around oct/21 (https://forum.radxa.com/t/how-to-enable-usb-dwc2-function/7551/4)

I found this commit on the radxa kernel tree, but I'm now sure if this is already applied on the kernel shipped on dietpi

https://github.com/radxa/kernel/commit/775a46783ad8e087d833c88756bfbba54020ffd1

MichaIng, 2022-12-03:
> From what I understand there is only one port that works with OTG, its the one closest to the corner.

It might generally work on every USB port, if configured in device tree accordingly, but not sure. The question is whether the kernel (driver) knows which port to apply gadget mode to. I'm not sure how this is generally done with gadget mode kernel modules.

We do not use the Radxa kernel, but the Armbian kernel. So if Radxa has not provided this patch upstream to be merged into mainline kernel, and Armbian if has not this applied as it as out-of-tree vendor patch, it is missing.

However, as it is a simple device tree change, we should be able to create a device tree overlay for this. Can you check this:
```sh
cat /proc/device-tree/__symbols__/usb; echo
cat "/proc/device-tree$(</proc/device-tree/__symbols__/usb)/dr_mode"; echo
```
The first should show a device tree path, if the same symbol exists that is used in Radxa's device tree. The first should then show the USB mode, if it can be toggled the same way as in Radxa's kernel. There will be a null byte error which you can ignore.

kazuoteramoto, 2022-12-03:
This is the value on dietpi
```
root@rhea:~# cat /proc/device-tree/__symbols__/usb; echo
/soc/usb@ffe09000
root@rhea:~# cat "/proc/device-tree$(</proc/device-tree/__symbols__/usb)/dr_mode"; echo
-bash: warning: command substitution: ignored null byte in input
host
```

and the modinfo on my device:
```
root@rhea:~# modinfo g_mass_storage
filename:       /lib/modules/5.10.147-meson64/kernel/drivers/usb/gadget/legacy/g_mass_storage.ko
license:        GPL
author:         Michal Nazarewicz
description:    Mass Storage Gadget
depends:        usb_f_mass_storage,libcomposite
intree:         Y
name:           g_mass_storage
vermagic:       5.10.147-meson64 SMP preempt mod_unload aarch64
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        18:3E:C4:55:3C:C5:95:74:D1:0B:13:06:AF:5C:89:DF:0D:EB:D5:99
sig_hashalgo:   sha1
signature:      7A:5E:DB:D1:F6:05:F1:F4:E2:39:6F:4E:18:2D:1F:8E:A0:A9:A2:B1:
                E3:5F:45:D6:73:F4:72:81:0B:25:50:FC:8E:01:2C:6C:24:9B:3D:05:
                E5:DA:14:2C:15:AE:34:78:D5:BC:C8:DF:9E:62:8A:88:4D:C8:F3:15:
                3C:DF:6B:FF:11:E6:01:4F:26:80:0E:70:4F:5C:C9:B6:3F:F6:DA:59:
                E2:25:E7:26:D3:18:2B:A3:FF:9C:7F:57:9E:B8:06:B3:64:58:77:EF:
                06:EF:33:90:3F:F0:15:89:DB:A5:E2:F1:62:12:E3:09:EE:90:41:E1:
                B6:A6:FC:41:3E:DF:AC:13:03:F9:79:20:D2:19:38:11:DF:A1:37:12:
                57:03:1D:AB:1C:13:DE:C6:85:0D:FA:79:E0:DA:A6:82:F8:6E:C9:15:
                3D:92:92:CA:9A:76:06:56:CF:D4:EC:18:DF:10:CC:FD:04:34:A3:B3:
                85:F1:BE:97:45:A3:A3:07:2A:D6:22:92:B3:E1:4D:B6:49:46:65:E5:
                8A:8A:EA:F8:54:3E:98:4F:14:31:E6:AF:82:60:AC:CF:2A:D3:6B:E4:
                D1:80:3B:BD:89:5D:5A:55:6B:3A:A5:99:F8:AD:44:08:9D:A6:A7:34:
                74:26:1E:FA:55:5A:73:5A:5B:D6:BB:29:F9:71:13:6D:56:1F:34:40:
                84:34:E2:87:25:1A:BC:5E:12:97:82:45:DA:87:1B:BF:0B:C1:2E:61:
                6E:44:15:C8:D3:5C:5F:9F:5E:48:51:23:C8:6B:94:9A:7E:4A:D8:02:
                C7:65:9B:58:DA:77:45:91:90:E2:38:64:B3:D3:9E:9D:33:4C:73:F4:
                D3:92:F2:0B:C8:5A:10:14:A9:02:AF:49:11:46:8B:41:98:D0:3F:F6:
                16:28:38:07:24:52:79:E6:4E:C2:A6:C8:E2:F3:08:7E:5E:FC:64:9B:
                1A:02:AB:01:38:C8:82:0D:FD:A3:70:A2:35:58:61:C6:71:46:93:84:
                18:F2:98:EE:A4:15:E1:D8:6D:1C:E7:9F:56:96:95:3C:1E:51:3F:2F:
                AA:05:7B:9F:D3:0E:4A:30:53:83:42:E4:E4:28:EE:49:61:FD:45:86:
                B0:74:A7:64:F5:8B:31:51:D2:13:E5:95:FF:F2:5C:61:F8:CE:80:0E:
                99:67:29:DD:52:46:E0:27:CE:9B:87:28:48:07:80:94:C0:45:FC:0B:
                2A:63:93:57:1E:C3:37:19:65:67:C9:16:5D:BC:47:AD:B1:67:9D:7F:
                1C:66:C2:A3:D0:D3:F4:95:B5:D4:C4:2A:71:D9:78:DB:7D:C3:BA:44:
                43:71:EA:8F:8B:9E:56:F7:48:C6:C0:9D
parm:           idVendor:USB Vendor ID (ushort)
parm:           idProduct:USB Product ID (ushort)
parm:           bcdDevice:USB Device version (BCD) (ushort)
parm:           iSerialNumber:SerialNumber string (charp)
parm:           iManufacturer:USB Manufacturer string (charp)
parm:           iProduct:USB Product string (charp)
parm:           file:names of backing files or devices (array of charp)
parm:           ro:true to force read-only (array of bool)
parm:           removable:true to simulate removable media (array of bool)
parm:           cdrom:true to simulate CD-ROM instead of disk (array of bool)
parm:           nofua:true to ignore SCSI WRITE(10,12) FUA bit (array of bool)
parm:           luns:number of LUNs (uint)
parm:           stall:false to prevent bulk stalls (bool)
```

kazuoteramoto, 2022-12-03:
Same command on the official image (ubuntu based, 5.10.69-7-amlogic-gfd159ba07d5c )

```
rock@radxa-zero:~$ cat /proc/device-tree/__symbols__/usb; echo
/soc/usb@ffe09000
rock@radxa-zero:~$ cat "/proc/device-tree$(</proc/device-tree/__symbols__/usb)/dr_mode"; echo
-bash: warning: command substitution: ignored null byte in input
otg
```

```
rock@radxa-zero:~$ modinfo g_mass_storage
filename:       /lib/modules/5.10.69-7-amlogic-gfd159ba07d5c/kernel/drivers/usb/gadget/legacy/g_mass_storage.ko
license:        GPL
author:         Michal Nazarewicz
description:    Mass Storage Gadget
depends:        usb_f_mass_storage,libcomposite
intree:         Y
name:           g_mass_storage
vermagic:       5.10.69-7-amlogic-gfd159ba07d5c SMP preempt mod_unload aarch64
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        DC:16:AF:1C:ED:67:6D:97
sig_hashalgo:   sha1
signature:      7D:13:39:CC:90:8E:EE:B8:C4:18:79:A9:71:E8:3E:74:B5:9E:12:8F:
                C3:2B:49:7B:70:60:3C:97:54:D0:76:D8:B2:92:DF:D1:5D:F0:80:8F:
                2C:33:FA:7A:28:A7:50:BC:BB:CD:78:29:40:CC:87:E9:09:FC:97:DA:
                7D:94:9A:EB:3C:6B:C0:1D:C5:6C:2C:7D:63:DF:1E:E7:26:FD:C4:6F:
                3E:51:BA:DB:C0:C3:C2:7C:36:B9:97:3F:EC:F0:4A:E1:71:CD:A5:8C:
                84:0E:39:DC:AB:8D:E8:E4:09:13:E0:FB:29:88:4A:F5:3C:D2:67:79:
                F9:1C:0D:89:6B:CE:97:47:84:85:9B:55:8B:65:48:24:B7:B2:1B:77:
                27:C4:A9:D9:3C:0B:9C:39:36:06:22:0D:E2:3D:FB:E5:E2:3A:00:03:
                27:19:EF:55:D6:41:39:AC:72:47:46:3E:1E:D8:A1:17:68:5E:22:5D:
                CA:47:50:F0:61:1D:E0:65:B4:8A:E9:88:7F:09:26:A0:21:C3:C7:F9:
                D4:6B:90:3D:9A:22:65:16:68:E2:AE:2F:A0:58:61:EE:3A:BC:C4:2B:
                B4:6B:BB:90:6B:69:12:A3:B3:95:45:3E:8E:E6:D1:84:6D:C5:13:0D:
                18:43:32:94:CA:E4:68:3F:99:58:72:A6:4C:CD:84:AC:9B:7B:CD:E9:
                62:DE:EF:BB:B9:12:F4:AA:C5:64:0A:F1:41:30:30:87:8C:5F:EE:DC:
                0E:F2:0A:40:9A:32:F1:FB:5B:16:7E:A7:AB:6B:C4:7F:7B:0F:53:A0:
                99:FE:A2:2F:DC:EA:28:59:A1:B6:B2:E9:EA:98:0B:EB:71:CF:52:EF:
                B6:BB:50:2A:3B:08:20:D3:81:89:38:B2:3D:C2:C3:C0:7D:90:58:80:
                26:AE:77:99:71:11:05:BE:C5:8C:D8:EE:89:CD:14:44:87:B4:6B:49:
                FC:26:4B:4C:82:92:B8:16:F5:01:00:98:FD:C0:BD:DD:00:EB:F8:7A:
                1E:59:24:C8:E1:5F:61:67:66:2B:23:93:39:21:EE:AF:97:95:C2:E0:
                DB:B1:A3:7B:94:48:16:1C:73:92:39:1B:9E:B3:22:B1:82:B6:2A:E8:
                B4:AA:D8:7D:7E:59:73:00:D8:D3:22:43:69:EB:DA:68:B3:6A:C0:2C:
                D8:8E:E5:D2:F2:3E:4B:D0:9F:5A:80:57:B7:21:6B:C8:9F:E0:FB:EC:
                72:F5:15:EB:13:3B:32:7C:30:F9:5C:AE:1C:60:83:FB:00:17:51:0C:
                1F:AF:58:D0:76:21:06:DF:47:FB:C4:80:B1:09:9C:0E:EE:42:E9:64:
                AD:64:3F:C2:9B:C8:C6:6D:22:93:38:20
parm:           idVendor:USB Vendor ID (ushort)
parm:           idProduct:USB Product ID (ushort)
parm:           bcdDevice:USB Device version (BCD) (ushort)
parm:           iSerialNumber:SerialNumber string (charp)
parm:           iManufacturer:USB Manufacturer string (charp)
parm:           iProduct:USB Product string (charp)
parm:           file:names of backing files or devices (array of charp)
parm:           ro:true to force read-only (array of bool)
parm:           removable:true to simulate removable media (array of bool)
parm:           cdrom:true to simulate CD-ROM instead of disk (array of bool)
parm:           nofua:true to ignore SCSI WRITE(10,12) FUA bit (array of bool)
parm:           luns:number of LUNs (uint)
parm:           stall:false to prevent bulk stalls (bool)
```

MichaIng, 2022-12-03:
So the very same device tree in this regards, but without the path applied. Let's try to generate a device tree overlay:
```sh
mkdir -p /boot/overlay-user
cat << '_EOF_' > /boot/overlay-user/usb-otg.dts
/dts-v1/;
/plugin/;
/ {
	compatible = "radxa,zero", "amlogic,g12a";
	fragment@0 {
		target = <&usb>;
		__overlay__ {
			dr_mode = "otg";
		};
	};
};
_EOF_
dtc -I dts -O dtb -o /boot/overlay-user/usb-gadget.dtbo /boot/overlay-user/usb-gadget.dts
G_CONFIG_INJECT 'user_overlays=' 'user_overlays=usb-gadget' /boot/dietpiEnv.txt
```
This sets OTG mode which should generally allow the USB port to be switched between host and peripheral mode, which is then hopefully done as intended by loading the `g_mass_storage` module.

kazuoteramoto, 2022-12-03:
Worked!

The value now matches the one on the official image
```
root@rhea:~# cat "/proc/device-tree$(</proc/device-tree/__symbols__/usb)/dr_mode"; echo
-bash: warning: command substitution: ignored null byte in input
otg
```

And I can see the device as a usb storage on my computer and access the files on it!

Thanks you very much for helping me!

MichaIng, 2022-12-03:
Okay great, thanks for testing.

MichaIng, 2022-12-06:
Implemented now OOTB: https://github.com/MichaIng/DietPi/commit/c2d1bae

After next DietPi update, you can remove `usb-gadget` from `/boot/dietpiEnv.txt` and `rm -f /boot/overlay-user/usb-otg.*` to avoid redundancy, as the overlay is named `dietpi-usb-otg`, to indicate that it's from us.

closed 2022-12-09 by MichaIng