MichaIng/DietPi #5935: dietpi-software: stop grafana-server before uninstalling else userdel fails

enviousjag, 2022-11-30:
### Creating a bug report/issue
uninstalling grafana fails due to grafana-server not shutting down and the script cannot continue uninstallation due to userdel not working.  

```
[FAILED] DietPi-Software | userdel grafana
[ INFO ] DietPi-Software | Executing alternative command: userdel grafana
userdel: user grafana is currently used by process 15917
```

Suggested fix: 
[image]

Joulinar, 2022-11-30:
Usually, all services should be stopped automatically at the beginning of every software install/reinstall/uninstall. Except you have excluded them manually.

MichaIng, 2022-11-30:
Nope, software uninstalls do not stop services. The service should be instead stopped as part of the `grafana` package removal. But indeed, there is not `prerm` script part of the package which would do this, an upstream bug, I'd say.

enviousjag, 2022-11-30:
I was able to replicate the issue with a new install & uninstall. 
Attached is the preceding console output. 

```
sudo dietpi-software uninstall 77
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated uninstall

[  OK  ] DietPi-Software | Uninstalling Grafana: platform for analytics and monitoring

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Uninstalling Grafana: platform for analytics and monitoring

[ INFO ] DietPi-Software | APT purge for: grafana, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following packages will be REMOVED:
  grafana*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
                                                              After this operation, 290 MB disk space will be freed.
(Reading database ... 52457 files and directories currently installed.)
Removing grafana (9.3.0) ...
(Reading database ... 45021 files and directories currently installed.)
Purging configuration files for grafana (9.3.0) ...
dpkg: warning: while removing grafana, directory '/etc/grafana' not empty so not removed
dpkg: warning: while removing grafana, directory '/usr/lib/systemd/system' not empty so not removed
[  OK  ] DietPi-Software | APT purge for: grafana
[  OK  ] DietPi-Software | rm /etc/apt/sources.list.d/grafana.list
[  OK  ] DietPi-Software | rm /etc/apt/trusted.gpg.d/dietpi-grafana.gpg
[FAILED] DietPi-Software | userdel grafana
[FAILED] DietPi-Software | userdel grafana
[FAILED] DietPi-Software | userdel grafana
```

Joulinar, 2022-11-30:
ahh right we changed this @MichaIng correct? It was the case in past? Or I'm completely wrong 🙄

MichaIng, 2022-11-30:
Yes it was, but already a long time ago. We also added error handling consequently on user removal, to catch exactly such issues.

MichaIng, 2022-11-30:
Opened a PR upstream: #59580
And fixed our end, along with some other enhancements: https://github.com/MichaIng/DietPi/commit/622ea3b

closed 2022-12-04 by MichaIng